verilator/verilator #2803: When using a bit-slice to assign to bits which don't exist, the assigned value erroneously gets stored

davidturner-argon, 2021-02-22:
If have a 2-bit signal `foo` and use a slice based on a variable to assign to bits 1-2, I expect the upper bit of my assignment to get thrown away. What actually happens in verilator is the whole assignment gets stored and the signal takes an erroneously large value given its width. Also, if you later downshift that signal using `>>` then the bits which should have been thrown away end up in the actual bits of the signal.

Verilator defends against this when slicing using a literal offset (i.e. `foo[1'd1 +: 2'd2]`) but you can bypass this by doing `foo[offset +: 2'd2]` where `offset = 1'd1`.

I've attached a simple example of this issue in the test_regress format. This displays `foo = 4, downshifted = 2`, where I would expect it to display `foo = 0, downshifted = 0`.

[t_assign_slice_overflow.pl.txt](https://github.com/verilator/verilator/files/6023417/t_assign_slice_overflow.pl.txt)
[t_assign_slice_overflow.v.txt](https://github.com/verilator/verilator/files/6023418/t_assign_slice_overflow.v.txt)

I've observed this with verilator `48986ccd` (current master) as well as `50c28fa9d3d1`

I haven't tried this exact regression test with other simulators but in a similar case I've confirmed that Questa, VCS, and Xcelium all discard the overhanging bits and do not reinstate them when downshifting.

wsnyder, 2021-02-22:
Verilator should be always cleaning the values so the extra bits are never observed.  Given these results there is a bug.

Might you be able to take a look, there should have been probably an AND masking operation by V3Clean, if not done earlier, making another value which is masked correctly will probably help find where this should have been done.  Otherwise I will take a look later.

davidturner-argon, 2021-02-22:
I've done some digging.  The generated C++ for the problematic statement looks like:

`vlTOPp->oob__DOT__foo = (4U | (1U & (IData)(vlTOPp->oob__DOT__foo)));`

So the RHS of the assignment definitely hasn't been cleaned for 2-bit width.  At the V3Clean stage, the AST for this statement looks like:

```
    1:2:3: ASSIGN 0x56429c488420 <e608#> {c9az} @dt=0x56429c4801b0@(G/wu32/2)
    1:2:3:1: CONST 0x56429c4884e0 <e600#> {c9bb} @dt=0x56429c4801b0@(G/wu32/2)  2'h2
    1:2:3:2: SEL 0x56429c488650 <e607#> {c9ai} @dt=0x56429c4801b0@(G/wu32/2) decl[1:0]]
    1:2:3:2:1: VARREF 0x56429c488720 <e601#> {c9af} @dt=0x56429c4801b0@(G/wu32/2)  oob__DOT__foo [LV] => VAR 0x56429c47a390 <e321> {c2an} @dt=0x56429c470780@(G/w2)  oob__DOT__foo [VSTATIC]  VAR
    1:2:3:2:2: CONST 0x56429c488840 <e605#> {c8ao} @dt=0x56429c480ae0@(G/wu32/1)  1'h1
    1:2:3:2:3: CONST 0x56429c4889b0 <e606#> {c9at} @dt=0x56429c4801b0@(G/wu32/2)  2'h2
```

I can edit `V3Clean.cpp:visit(AstNodeAssign* nodep)` and insert a clean after the assign but that produces incorrect C++ output:

`(3U & vlTOPp->t__DOT__foo = (4U | (1U & (IData)(vlTOPp->t__DOT__foo)));`

I think the problem is that when we run the cleaning stage the bit-select is considered a LHS operation, but a later stage is transforming the assignment so that the selection logic goes on the RHS of the assignment, and we can't easily express this clean until that second transformation is done.

I'm not sure how to proceed from here so I'd appreciate if you could take a look.

wsnyder, 2021-02-24:
In V3Expand there's expandLhs(AstNodeAssign* nodep, AstSel* lhsp) which should be handling the bad case.

But if you run with -Ox this expansion isn't called and instead the verilated.h routines are used, so we'll need the test to try that option, that is we also need the test attached:

[t_assign_slice_overflow_ox.pl.txt](https://github.com/verilator/verilator/files/6032745/t_assign_slice_overflow_ox.pl.txt)

There are several cases for different widths (<32, 33-64, 64+) and differing offsets, probably all have similar issues and need tests to check.

Looking at just one of those several likely broken cases I think this would fix one of them:

```
diff --git a/include/verilated.h b/include/verilated.h
--- a/include/verilated.h
+++ b/include/verilated.h
@@ -1752,8 +1752,10 @@ QData VL_POWSS_QQW(int obits, int, int rbits, QData lhs, WDataInP rwp, bool lsig
 // INTERNAL: Stuff LHS bit 0++ into OUTPUT at specified offset
 // ld may be "dirty", output is clean
 static inline void _VL_INSERT_II(int, CData& lhsr, IData ld, int hbit, int lbit) VL_PURE {
+    IData mask = VL_MASK_I(hbit);
     IData insmask = (VL_MASK_I(hbit - lbit + 1)) << lbit;
-    lhsr = (lhsr & ~insmask) | ((ld << lbit) & insmask);
+    // Written so if lbit and hbit is constant, the masking becomes constant
+    lhsr = (lhsr & ~insmask) | ((ld << lbit) & (insmask & mask));
 }
 static inline void _VL_INSERT_II(int, SData& lhsr, IData ld, int hbit, int lbit) VL_PURE {
     IData insmask = (VL_MASK_I(hbit - lbit + 1)) << lbit;
```

davidturner-argon, 2021-02-24:
Thanks for the pointers, that all makes a lot more sense now.  I'll have a go at enumerating and fixing all the bug cases and rustle up a pull request.

PR #2811 (2021-03-01): Fix slice-assign overflow bug (Issue 2803)
When assigning to a bit-select (array slice), there is nothing stopping
the bit-select referring to bits beyond the width of the signal we're
actually assigning to.  Because the expansion of this logic (V3Expand
and verilated.h) happens after V3Clean, this bypasses the normal
cleaning process and can result in a signal containing a value whose
width exceeds the signal width.

A lot of codepaths cover the same logic depending on the widths
involved, which things are constant, and whether the logic is done by
verilated.h or expanded by V3Expand.  This c [... cut]
closed 2021-03-01 by wsnyder