verilator/verilator #2807: Improve gated clock support in hierarhical verilation

yTakatsukasa, 2021-02-23:
If a hierarchical block creates a gated clock inside it, the simulation result may be wrong. (It took several days to figure out).

Thanks to #2172, just marking the original clock as `clocker` is enough.

Here are some random ideas to improve.
- Warn if clock is gated inside a hierarchical block and advise to add `--clk` option
- Set `--clk` option when Verilating a hierarchical block if `--clk` is set to the top module
  - When Verilating a hierarchical block `sub` as below,  `--clk subclk` must be set, `--clk clk` is not correct.

```systemverilog
module top(input clk /*verilator clocker*/);
  wire subclk = clk;
  sub i_sub(.subclk(subclk));
endmodule

module sub(input subclk); /* verilator hier_block*/
  wire gated_clk = subclk | some_gate_ctrl_signal;
  always_ff @(posedge gated_clk) ....
endmodule
```

yTakatsukasa, 2021-05-22:
I recently found that a simple test case shows circular logic error. I am trying to understand the difference from t_prot_lib_clk_gated.pl.

### secret.v
``` systemverilog
module secret(input wire clk /*verilator clocker*/, input wire [7:0] in, output wire [7:0] out);
   logic [7:0] tmp;
   always_ff @(posedge clk) tmp <= in;
   assign out = tmp;
endmodule
```

### t.v
```systemverilog
module t (
    input wire clk /*verilator clocker*/,
    input wire clk_en
);

   logic clk_en_latch /*verilator clock_enable*/;
   /*verilator lint_off LATCH*/
   always_comb if(!clk) clk_en_latch <= clk_en;
   /*verilator lint_on LATCH*/

   wire gated_clk = clk & clk_en_latch;

   logic [7:0] v;
   logic [7:0] exp;
   wire [7:0] out;

   always_ff @(posedge gated_clk) begin
       v <= v + 1;
       exp <= v;
   end

    secret i_secret(.clk(gated_clk), .in(v), .out(out));

    always_ff @(posedge gated_clk) begin
        if (exp != out) $stop;
        if (v == 8'd16) $finish;
    end
endmodule
```

```bash
verilator --cc --protect-lib secret --top-module secret src/secret.v --no-skip-identical
verilator --cc --top-module t src/t.v obj_dir/secret.sv --build --no-skip-identical
```

> %Error: Circular logic when ordering code (non-cutable edge loop)
        src/t.v:11:9:      Example path: t.gated_clk
        obj_dir/secret.sv:76:5:      Example path: ACTIVE
        obj_dir/secret.sv:76:5:      Example path: ALWAYS
        src/t.v:11:9:      Example path: t.gated_clk
        src/t.v:11:19:      Example path: ASSIGNW
        src/t.v:11:9:      Example path: t.gated_clk
        src/t.v:11:19:      Example path: ASSIGNW
        src/t.v:11:9:      Example path: t.gated_clk
        src/t.v:17:4:      Example path: ACTIVE
        src/t.v:24:5:      Example path: ALWAYS
        obj_dir/secret.sv:55:10:      Example path: t.i_secret.last_seq_seqnum__V
        obj_dir/secret.sv:81:9:      Example path: ASSIGNPRE
        obj_dir/secret.sv:55:10:      Example path: __Vdly__t.i_secret.last_seq_seqnum__V
        obj_dir/secret.sv:76:5:      Example path: ALWAYS
        src/t.v:11:9:      Example path: t.gated_clk
        src/t.v:11:19:      Example path: ASSIGNW
%Error: Internal Error: ../V3Graph.cpp:267: Loops detected in graph:   VERTEX=0x55d4b4b101e0\n ASSIGNW r3 c5

yTakatsukasa, 2021-12-11:
#3175 looks related.
A protect lib wrapper uses DPI-C to hide the detail of the secret circuit. context DPI-C functions have dependency to any signals and cause the loop dependency.

wsnyder, 2022-05-15:
Note with #3278 by @gezalore this now gives:

```
%Warning-COMBDLY: t.v:8:38: Non-blocking assignment '<=' in combinational logic process
                          : ... This will be executed as a blocking assignment '='!
    8 |    always_comb if(!clk) clk_en_latch <= clk_en;
%Error: Circular logic when ordering code (non-cutable edge loop)
```

petergrossmann21, 2025-02-18:
I recently reproduced this issue in my own work.  I initially discovered it on Verilator 5.030, and then wrote a testcase and reproduced the issue on `Verilator 5.033 devel rev v5.032-78-gfe1517164`.

I assembled a simplified test case for my own understanding and possible submission.  I believe that it can be adapted readily into the test_regress/t/t_EXAMPLE.v form and have made a best-effort initial attempt at this adaptation.  I will need feedback to confirm that I am compliant, particularly if I am expected to submit a Python wrapper with the HDL code.  To that end I have created a preliminary t_<test_name>.v file for review and included it with the original test case content for review.

I see that this issue was last posted to nearly three years ago, has there been any development since then?

[hierarchical_build_debug.tar.gz](https://github.com/user-attachments/files/18855353/hierarchical_build_debug.tar.gz)

jcteng, 2025-04-07:
any update for this issue ? @wsnyder

gigeresk, 2025-07-02:
I also have this issue.  Any update?