verilator/verilator #2810: Weird 'stuck' problem with cocotb on arm64 (apple m1)

markusdd, 2021-02-26:
Hi again, with another cocotb/verilator story :)

I have now switched to an MacBook Air M1 and now I have a weird issue with my codebase where verilator does not seem to answer cocotb anymore, at least that is how it seems

versions are in the log below, but it is the same behaviour for verilator 108 as well.

```
0.00ns INFO     Running on Verilator version 4.111 devel
     0.00ns INFO     Running tests with cocotb v1.5.0.dev307+g9fe131d9 from /opt/homebrew/Cellar/python@3.9/3.9.2_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb
     0.00ns INFO     Seeding Python random module with 1614331972
     0.00ns INFO     Found test svb_clk_div_tb.test_clk_div
     0.00ns INFO     Found test svb_clk_div_tb.test_clk_div_param_assert
     0.00ns INFO     Running test 1/2: test_clk_div
     0.00ns DEBUG    Trigger fired: <NullTrigger for Start <Test test_clk_div> at 0x106230ec0>
     0.00ns DEBUG    1 pending coroutines for event <NullTrigger for Start <Test test_clk_div> at 0x106230ec0>
                     	test_clk_div
     0.00ns DEBUG    Scheduling coroutine test_clk_div
     0.00ns DEBUG    Scheduling with Value(None)
     0.00ns INFO     Starting test: "test_clk_div"
                     Description:  Test that clock is divided according to param DIV_BY.
     0.00ns DEBUG    Adding new coroutine Clock.start
     0.00ns DEBUG    Scheduling with Value(None)
     0.00ns DEBUG    Adding new coroutine Scheduler._do_writes
     0.00ns DEBUG    Scheduling with Value(None)
     0.00ns DEBUG    Coroutine Scheduler._do_writes yielded <<Event at 0x1054a30a0>.wait() at 0x105c82d40> (mode 1)
     0.00ns DEBUG    Coroutine Clock.start yielded <Timer of 5000.00ps at 0x105db5380> (mode 1)
     0.00ns INFO     Reset dividers.
     0.00ns DEBUG    Coroutine test_clk_div yielded FallingEdge(ModifiableObject(svb_clk_div_tb.clk_i)) (mode 1)
     0.00ns DEBUG    Scheduled coroutine test_clk_div
     0.00ns DEBUG    1 pending coroutines for event <<Event at 0x1054a30a0>.wait() at 0x105c82d40>
                     	Scheduler._do_writes
     0.00ns DEBUG    Scheduling coroutine Scheduler._do_writes
     0.00ns DEBUG    Scheduling with Value(<<Event at 0x1054a30a0>.wait() at 0x105c82d40>)
     0.00ns DEBUG    Coroutine Scheduler._do_writes yielded ReadWrite() (mode 1)
     0.00ns DEBUG    Scheduled coroutine Scheduler._do_writes
     0.00ns DEBUG    All coroutines scheduled, handing control back to simulator
     0.00ns DEBUG    Trigger fired: ReadWrite()
     0.00ns DEBUG    1 pending coroutines for event ReadWrite()
                     	Scheduler._do_writes
     0.00ns DEBUG    Scheduling coroutine Scheduler._do_writes
     0.00ns DEBUG    Scheduling with Value(ReadWrite())
     0.00ns DEBUG    Coroutine Scheduler._do_writes yielded <<Event at 0x1054a30a0>.wait() at 0x106230880> (mode 1)
     0.00ns DEBUG    Scheduled coroutine Scheduler._do_writes
     0.00ns DEBUG    All coroutines scheduled, handing control back to simulator
```

it gets completely stuck after the last handover form cocotb. I cannot even ctrl+c it, I must send a TERM signal from another shell.

Is there any way to debug this further on verilators end? From the log it does not look like a cocotb problem, adding @ktbarrett anyway for notification.

I have not observed this on my Intel macMini nor in my Linux or WSL environments.

It could be totally unrelated to the arm architecture, but it currently somewhat feels like it.

markusdd, 2021-02-26:
I have tried to attach with lldb and it is not standing at one position but it seems to to cycle somewhere:

```
❯ lldb
(lldb) attach svb_clk_div_tb
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
    frame #0: 0x00000001a5a2b480 libsystem_platform.dylib`_platform_memcmp + 112
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b480 <+112>: subs   x2, x2, #0x1              ; =0x1
    0x1a5a2b484 <+116>: b.hs   0x1a5a2b470               ; <+96>
    0x1a5a2b488 <+120>: mov    x0, x3
    0x1a5a2b48c <+124>: ret
Target 0: (svb_clk_div_tb) stopped.

Executable module set to "/Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/sim_build/svb_clk_div_tb".
Architecture set to: arm64e-apple-macosx-.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001a5a2b484 libsystem_platform.dylib`_platform_memcmp + 116
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b484 <+116>: b.hs   0x1a5a2b470               ; <+96>
    0x1a5a2b488 <+120>: mov    x0, x3
    0x1a5a2b48c <+124>: ret

libsystem_platform.dylib`OSAtomicDequeue:
    0x1a5a2b490 <+0>:   tst    x0, #0xf
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001a5a2b470 libsystem_platform.dylib`_platform_memcmp + 96
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b470 <+96>:  ldrb   w4, [x0], #0x1
    0x1a5a2b474 <+100>: ldrb   w5, [x1], #0x1
    0x1a5a2b478 <+104>: subs   x3, x4, x5
    0x1a5a2b47c <+108>: b.ne   0x1a5a2b488               ; <+120>
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001a5a2b474 libsystem_platform.dylib`_platform_memcmp + 100
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b474 <+100>: ldrb   w5, [x1], #0x1
    0x1a5a2b478 <+104>: subs   x3, x4, x5
    0x1a5a2b47c <+108>: b.ne   0x1a5a2b488               ; <+120>
    0x1a5a2b480 <+112>: subs   x2, x2, #0x1              ; =0x1
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001a5a2b478 libsystem_platform.dylib`_platform_memcmp + 104
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b478 <+104>: subs   x3, x4, x5
    0x1a5a2b47c <+108>: b.ne   0x1a5a2b488               ; <+120>
    0x1a5a2b480 <+112>: subs   x2, x2, #0x1              ; =0x1
    0x1a5a2b484 <+116>: b.hs   0x1a5a2b470               ; <+96>
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001a5a2b47c libsystem_platform.dylib`_platform_memcmp + 108
libsystem_platform.dylib`_platform_memcmp:
->  0x1a5a2b47c <+108>: b.ne   0x1a5a2b488               ; <+120>
    0x1a5a2b480 <+112>: subs   x2, x2, #0x1              ; =0x1
    0x1a5a2b484 <+116>: b.hs   0x1a5a2b470               ; <+96>
    0x1a5a2b488 <+120>: mov    x0, x3
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
[... 406 lines cut ...]
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001008798a0 svb_clk_div_tb`VerilatedVpiImp::callCbs(unsigned int) + 4
svb_clk_div_tb`VerilatedVpiImp::callCbs:
->  0x1008798a0 <+4>:  stp    x20, x19, [sp, #0x10]
    0x1008798a4 <+8>:  stp    x29, x30, [sp, #0x20]
    0x1008798a8 <+12>: add    x29, sp, #0x20            ; =0x20
    0x1008798ac <+16>: adr    x8, #0x2b2dc              ; VerilatedVpiImp::s_s
Target 0: (svb_clk_div_tb) stopped.
(lldb) s
Process 28298 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
    frame #0: 0x00000001008798a4 svb_clk_div_tb`VerilatedVpiImp::callCbs(unsigned int) + 8
svb_clk_div_tb`VerilatedVpiImp::callCbs:
->  0x1008798a4 <+8>:  stp    x29, x30, [sp, #0x20]
    0x1008798a8 <+12>: add    x29, sp, #0x20            ; =0x20
    0x1008798ac <+16>: adr    x8, #0x2b2dc              ; VerilatedVpiImp::s_s
    0x1008798b0 <+20>: nop
Target 0: (svb_clk_div_tb) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = instruction step into
  * frame #0: 0x00000001008798a4 svb_clk_div_tb`VerilatedVpiImp::callCbs(unsigned int) + 8
    frame #1: 0x000000010086a2e0 svb_clk_div_tb`main + 220
    frame #2: 0x00000001a5a01f34 libdyld.dylib`start + 4
```

wsnyder, 2021-02-26:
Verilator seems to be sending signal changes into cocotb.  Try compiling with "-DVL_DEBUG", then either calling Verilator::debug(9) from the debugger at setup or put it in the code..

markusdd, 2021-02-26:
Thanks for the quick response.

I did that now. Should this produce more stdout?
I do not see anything.

wsnyder, 2021-02-26:
You'll get messages like
`+    Vt_alw_combdly::_initial__TOP__2\n
`
if you're not debug isn't on due to VL_DEBUG not being defined or the Verilator::debug not being set.

markusdd, 2021-02-26:
Mmmmh weird:

My verilator line:
`verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module svb_clk_div_tb --vpi --public-flat-rw --prefix Vtop -o svb_clk_div_tb -LDFLAGS "-Wl,-rpath,/opt/homebrew/lib/python3.9/site-packages/cocotb/libs -L/opt/homebrew/lib/python3.9/site-packages/cocotb/libs -lcocotbvpi_verilator" --trace-fst --trace-structs --debug -CFLAGS -DTRACE_FST -DVL_DEBUG -DVERILATOR_SIM_DEBUG --timescale 1ns/1ps /Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/../../svb_clk_lib/svb_clk_div.sv /Users/mkrause/projects/sv_base_lib/test_lib/svb_clk_div_tb/svb_clk_div_tb.sv /opt/homebrew/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp`

And the cocotb top of maintop is like this:

```
int main(int argc, char** argv) {
    Verilated::commandArgs(argc, argv);
#ifdef VERILATOR_SIM_DEBUG
    Verilated::debug(99);
#endif
    std::unique_ptr<Vtop> top(new Vtop(""));
    Verilated::fatalOnVpiError(false); // otherwise it will fail on systemtf

#ifdef VERILATOR_SIM_DEBUG
    Verilated::internalsDump();
#endif

    vlog_startup_routines_bootstrap();
    VerilatedVpi::callCbs(cbStartOfSimulation);

```

But no further output :(

ktbarrett, 2021-02-26:
@marlonjames you were seeing a few bugs with Verilator that affected cocotb. Is this one of them?

wsnyder, 2021-02-26:
You are setting the verilog VL_DEBUG define. Use

`-CFLAGS -DTRACE_FST        -CFLAGS -DVL_DEBUG          -CFLAGS -DVERILATOR_SIM_DEBUG `

markusdd, 2021-02-26:
* facepalm *

yes. sorry about that. will check in a minute.

markusdd, 2021-02-26:
> @marlonjames you were seeing a few bugs with Verilator that affected cocotb. Is this one of them?

I guess it should not be directly related to this, if you remember we actually had ironed everything out to a point where it was working reliably. There was still the intention to tweak the maintop for better performance though.

markusdd, 2021-02-26:
That is interesting. It looks like it is stuck when running the first eval cycle with the clock. (from t0,170)

please find the log attached.
I will also attach the cocotb test bench and the Verilog files involved in the .zip.

[testcase.zip](https://github.com/verilator/verilator/files/6051885/testcase.zip)

[log.txt](https://github.com/verilator/verilator/files/6051871/log.txt)

ktbarrett, 2021-02-26:
> we actually had ironed everything out to a point where it was working reliably

Yes, for 4.106. But there was a new bug introduced in 4.108 :laughing:. Only 4.106 works right now.

markusdd, 2021-02-26:
> > we actually had ironed everything out to a point where it was working reliably
> 
> Yes, for 4.106. But there was a new bug introduced in 4.108 😆. Only 4.106 works right now.

Oh splendid! But a good hint. Let me recompile on 106 and check.

markusdd, 2021-02-26:
Yes, confirmed.

On 106 it runs perfectly! 

So I guess we should rename this ticket to something more appropriate, the attached example is still suitable I guess, although it has nothing to do with ARM or the M1 or MacOS

marlonjames, 2021-03-02:
4.106 works, but there are a few VPI issues. Some of those are fixed in 4.108, but unfortunately other changes make cocotb currently incompatible. https://github.com/verilator/verilator/issues/2778 needs to be fixed, I just haven't found the time to do it yet.

markusdd, 2021-03-02:
thanks for already having dived into this...this is as usual the area where my knowledge is very limited.

wsnyder, 2021-03-23:
Upstream bug closed - presume resolved.

closed 2021-03-23 by wsnyder

ktbarrett, 2021-03-24:
This is reported as #2778.