verilator/verilator #2812: Access with VPI to a multidimensional packed array

Guillaume000000, 2021-02-28:
Hello, 

I'm not sure if it's a bug or just an unimplemented feature: I want to modify with VPI the value of a multidimensional packed array, but the function "vpi_handle_by_name()" return a null value. If I replace the multidimensional array by a simple array, it's ok. I have found information about unpacked arrays but not about packed arrays.
my code:

```
module counter  (
                output logic [4:0] count ,
                input  logic [4:0] data ,
                input        clk  ,
                input        load ,
                input        up ,
                input        rst   
                );

timeunit 1ns;
timeprecision 100ps;

reg [1:0][4:0] temp /*verilator public_flat_rw*/;

always_ff @(posedge clk, posedge rst)
begin
	if (rst == 1)
		count <= 5'b0;
	else if (load == 1)
		count <= data;
	else if (up == 1)
		count <= count + 1;
	else if (up == 0 && temp[0] != 5)
		count <= count - 1;
	temp[0] <= count;
	temp[1] <= count+1;
end
endmodule
```
the c function:
```
void modify_temp(){
	vpiHandle temp_handler = vpi_handle_by_name((PLI_BYTE8*)"TOP.counter.temp", NULL);
	if(!temp_handler) { vl_fatal(__FILE__, __LINE__, "sim_main", "No handler found"); }
	s_vpi_value value;
	value.format = vpiIntVal;
	vpi_get_value(temp_handler, &value);
	value.value.integer ^= 8;
	vpi_put_value(temp_handler, &value, NULL, vpiForceFlag);
}
```
(tested with verilator version 4.110)

Thank you for your help.

wsnyder, 2021-02-28:
Can you please either make this into a standalone runnable test, or better modify test_regress/t/t_vpi_var or another existing Verilator test case to show the problem?  Thanks

Guillaume000000, 2021-02-28:
I have added a complete test and simplify a little the systemverilog module.

wsnyder, 2021-02-28:
Thanks for the update, Verilator does not currently support multi-dimensional unpacked array access through the VPI, with your example you'll see it added comments:

        //UNSUP __Vscope_TOP.varInsert(__Vfinal,"count", 

If you change to a 1D array ("[9:0]") it works.  If you would like to improve Verilator to support this a pull request would be appreciated - most the work will be in writing the necessary tests and improving the varInsert and VPI routines to handle the multi-D arrays through VPI.

Guillaume000000, 2021-03-01:
Thanks for your answer, currently I think I don't enough know the internal operation of Verilator to improve it, but may be in the future.