verilator/verilator #2816: mixed blocking and non-blocking assignment on dfferent slices of vector

zhuzhzh, 2021-03-04:
The mixed blocking and non-blocking assignment is supported. Here is my example. It works.

```verilog
module test(input clk);

reg [31:0] cnt=0;

always @(posedge clk) begin
	cnt <= cnt + 1;
end

reg [15:1] ex;

always @(*) begin
	ex[1] = cnt[0];
end

genvar i;
generate
for (i=2;i<=15;i=i+1)  begin: T1T0_INFO_EN
	always @(posedge clk) begin
		ex[i] <= ex[i-1];
	end
end
endgenerate

endmodule

```

Above code can pass the compilation. ex[1] is assigned in blocking mode, while ex[15:2] is assigned in nonblocking mode.

but in the real design, the situation is complex. Then the similar code will report the error.
I simplify the code as below.

```verilog
module test (
	clk
	, rst_n
	, mul_num_ex4
	, t1_mul_num_ex4
	, mul_num_ex5
	, t0_stall_ex1
	, t1_stall_ex1
);

input               clk;
input               rst_n;

input  [1 -1:0] t0_stall_ex1                  ;
input  [1 -1:0] t1_stall_ex1                  ;
input [7:0]                 mul_num_ex4;
input [7:0]                 t1_mul_num_ex4;
output [7:0]            mul_num_ex5;
reg [5:0]               valu_csr_wsts_ex7;
reg [7:0]               mul_num_ex5;

genvar i;
wire pp_en_ex1;
reg [15:1] valu_en_ex ;
reg [1:0] t0_ex [15:1];

wire first_go_stall_ex1 = t0_stall_ex1 | t1_stall_ex1;

always @ (*) begin
	valu_en_ex[1] = pp_en_ex1 & !first_go_stall_ex1;
end

generate

for (i=2;i<=15;i=i+1)  begin: t0
	always @ (posedge clk or negedge rst_n) begin
		if (!rst_n) begin 
			valu_en_ex[i] <= 1'b0;
		end else begin
			valu_en_ex[i] <= valu_en_ex[i-1] ;
		end
	end
	always @ (posedge clk or negedge rst_n) begin
		if (!rst_n) begin 
			t0_ex[i] <= 2'b0;
		end
		else begin
			if (valu_en_ex[i-1]) begin 
				t0_ex[i] <= t0_ex[i-1];
			end
		end
	end
end
endgenerate
always @ (posedge clk ) begin
	case (t0_ex[4])
		2'd0   : mul_num_ex5            <= 8'b0 ;
		2'd1   : mul_num_ex5            <= mul_num_ex4 ;
		2'd2   : mul_num_ex5            <= t1_mul_num_ex4 ;
		default: mul_num_ex5            <= mul_num_ex4 + t1_mul_num_ex4 ;
	endcase
end
endmodule
```

the error:
```
%Error-BLKANDNBLK: test.sv:26:12: Unsupported: Blocked and non-blocking assignments to same variable: 'test.valu_en_ex'
   26 | reg [15:1] valu_en_ex ;
      |            ^~~~~~~~~~
                   test.sv:32:2: ... Location of blocking assignment
   32 |  valu_en_ex[1] = pp_en_ex1 & !first_go_stall_ex1;
      |  ^~~~~~~~~~
                   test.sv:42:4: ... Location of nonblocking assignment
   42 |    valu_en_ex[i] <= valu_en_ex[i-1] ;
      |    ^~~~~~~~~~
%Error: Exiting due to 1 error(s)
```

The issue is not caused by the mix assignment. Actually, if I remove the t0_ex related code (keep the mixed assignment on valu_en_ex), the issue is gone. 

Please help check what's the real issue. And there is any workaround for me to move on.

wsnyder, 2021-03-04:
Your first example does not show BLKANDNBLK because "ex" is never used so it gets optimized away.

As to working around BLNANDNBLK please see the documentation. https://www.veripool.org/projects/verilator/wiki/Manual-verilator#BLKANDNBLK

closed 2021-03-04 by wsnyder

zhuzhzh, 2021-03-05:
I manually changed the code. And I also tried -Wno-BLKANDNBLK.

About -Wno-BLKANDNBLK, I checked the generated C code. Likes like the whole var is treated as S_Data. May it have the potential functionality issue?

I think it's safe to split the var with /*verilator split_var*/;

wsnyder, 2021-03-05:
SData is just related to width, has nothing to do with scheduling the model.

zhuzhzh, 2021-03-11:
got it, thanks.