verilator/verilator #2818: linking error on Mac os

chonwhite, 2021-03-04:
can't get verilator working on mac os Catalina 10.15.7.

```shell
➜  make_hello_c git:(stable) make
-- Verilator hello-world simple example
-- VERILATE & BUILD --------
verilator -cc --exe --build -j top.v sim_main.cpp
g++    sim_main.o verilated.o Vtop__ALL.a      -o Vtop
ld: warning: ignoring file Vtop__ALL.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
Undefined symbols for architecture x86_64:
  "Vtop::final()", referenced from:
      _main in sim_main.o
  "Vtop::eval_step()", referenced from:
      _main in sim_main.o
  "Vtop::Vtop(char const*)", referenced from:
      _main in sim_main.o
  "Vtop::~Vtop()", referenced from:
      _main in sim_main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vtop] Error 1
%Error: make -C obj_dir -f Vtop.mk -j exited with 2
%Error: Command Failed /usr/local/bin/verilator_bin -cc --exe --build -j top.v sim_main.cpp
make: *** [default] Error 2
```

wsnyder, 2021-03-04:
Seems a common problem on macs. Please google the error message, first hit might help https://www.google.com/search?client=firefox-b-1-d&q=%22for+macOS-x86_64+but+attempting+to+link+with+file+built+for+unknown-unsupported+file+format+%28+0x21+0x3C+0x61+0x72+0x63+0x68+0x3E+0x0A+0x2F+0x20+0x20+0x20+0x20+0x20+0x20+0x20

chonwhite, 2021-03-05:
Solved! Thanks
fixed it by removing Homebrew's Binutils and recompiling verilator.

closed 2021-03-05 by chonwhite