verilator/verilator #2819: buffer overflow found in generated code

mytbk, 2021-03-05:
I found some strange behavior when running a project, which made me suspect there was buffer overflow bugs. Using address sanitizer, I find the following code can let ASAN report a buffer overflow. Tested on verilator 4.108 and git revision v4.110-14-g018d9947.

top.sv:
```
`define uWidth 35
`define vLen 16
`define vecBits (`uWidth*`vLen)

class MyClass;
	task run(input bit [`vecBits-1:0] iReg, input int num);
		int i,knum;
		bit [`uWidth-1:0] t;

		knum = (num>`vLen)?`vLen:num;
		for (i = 0; i < knum; i++) begin
			$display("%d",i);
			t = iReg[`uWidth*i+:`uWidth];
			$display("%x",t);
		end
	endtask
endclass;

module top;
	bit [559:0] v;
	int i;
initial begin
	MyClass o;
	o = new;
	v = 560'b0;
	/* it's strage that using the following will not trigger an overflow
	for (i = 0; i < `vLen; i++) begin
		v[`uWidth*i+:`uWidth] = {i[2:0],32'h0};
	end
	*/
	o.run(v,16);
	$finish();
end
endmodule;
```

main.cc:
```
#include "Vtop.h"      // From Verilating "top.v"
#include <verilated.h> // Defines common routines

Vtop *top; // Instantiation of module

int main(int argc, char **argv) {
  Verilated::commandArgs(argc, argv); // Remember args

  top = new Vtop; // Create instance

  while (!Verilated::gotFinish()) {
    top->eval();              // Evaluate model
  }

  top->final(); // Done simulating
  //    // (Though this example doesn't get here)
  delete top;
}
```

sanitizer report:
```
$ verilator --cc --exe top.sv main.cc 
$ make -C obj_dir/ -f Vtop.mk CXXFLAGS='-fsanitize=address' USER_LDFLAGS='-fsanitize=address' OPT_FAST='-O0 -g' 
$ ./obj_dir/Vtop 
          0
000000000
          1
000000000
          2
000000000
          3
000000000
          4
000000000
          5
000000000
          6
000000000
          7
000000000
          8
000000000
          9
000000000
         10
000000000
         11
000000000
         12
000000000
         13
000000000
         14
000000000
         15
=================================================================
==4609==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffe59c382b8 at pc 0x55cb26a17f73 bp 0x7ffe59c38120 sp 0x7ffe59c38110
READ of size 4 at 0x7ffe59c382b8 thread T0
    #0 0x55cb26a17f72 in Vtop___024unit__03a__03aMyClass::__VnoInFunc_run(Vtop__Syms*, unsigned int*, unsigned int) /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop___024unit__03a__03aMyClass__Vclpkg.cpp:34
    #1 0x55cb26a191a5 in Vtop::_initial__TOP__1(Vtop__Syms*) /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop__Slow.cpp:60
    #2 0x55cb26a19356 in Vtop::_eval_initial(Vtop__Syms*) /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop__Slow.cpp:68
    #3 0x55cb26a17c52 in Vtop::_eval_initial_loop(Vtop__Syms*) /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop.cpp:46
    #4 0x55cb26a17b57 in Vtop::eval_step() /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop.cpp:21
    #5 0x55cb269fbe87 in Vtop::eval() Vtop.h:52
    #6 0x55cb269fbd69 in main ../main.cc:12
    #7 0x7fafebfedb24 in __libc_start_main (/usr/lib/libc.so.6+0x27b24)
    #8 0x55cb269fbafd in _start (/home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop+0x8afd)

Address 0x7ffe59c382b8 is located in stack of thread T0 at offset 248 in frame
    #0 0x55cb26a18ab5 in Vtop::_initial__TOP__1(Vtop__Syms*) /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop__Slow.cpp:35

  This frame has 5 object(s):
    [48, 56) 'vlSymsp' (line 35)
    [80, 96) '<unknown>'
    [112, 128) '<unknown>'
    [144, 160) '<unknown>'
    [176, 248) '__Vtemp1' (line 39) <== Memory access at offset 248 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/irudog/programs/verilog/sv/test_overflow/obj_dir/Vtop___024unit__03a__03aMyClass__Vclpkg.cpp:34 in Vtop___024unit__03a__03aMyClass::__VnoInFunc_run(Vtop__Syms*, unsigned int*, unsigned int)
Shadow bytes around the buggy address:
  0x10004b37f000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f030: 00 00 00 00 00 00 00 00 f1 f1 f1 f1 00 00 00 f2
  0x10004b37f040: f2 f2 f8 f8 f2 f2 00 00 f2 f2 00 00 f2 f2 00 00
=>0x10004b37f050: 00 00 00 00 00 00 00[f3]f3 f3 f3 f3 00 00 00 00
  0x10004b37f060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10004b37f0a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==4609==ABORTING
```

The reason for the buffer overflow is that the generated code has ``iReg[((IData)(2U) + (0x1fU & (VL_MULS_III(10,32,32, (IData)(0x23U), i) >> 5U)))]`` where ``iReg`` is of size 18, and when ``i`` is 15, it'll access ``iReg[18]``, which is an overflow.

wsnyder, 2021-03-05:
Obviously your "(num>`vLen)?`vLen:num" should have been "(num> `vLen-1)?` vLen-1 :num", but that's not the point ;)

Verilator is correctly inserting code to check for Verilog array bounds, why the sanitizer fails is related to how it maps code to C++.

With -Ox this is a false positive, in that the generated code is intentionally reading past the end of the array to form 64 bits. It then later masks off the past-end data with a AND mask.  Clang doesn't track bit validity so it doesn't know this. This is somewhat similar to how some strcmp routines read in 8 bytes chunks which might go past the 0 string end.

There might be other cases where the unknown data can "escape", but this would always be when reading past a Verilog array, and the output bits must be "x" anyhow, which Verilator considers fair game to convert to 0/1 as it feels is most performant.

The sanitization failure goes away with -Ox which turns off the expansion optimization, and uses VL_SEL_QWII which is slower and should be clean.

One Verilator fix would be to add two conditional checks on every select, which is runtime-expensive (making it similar to VL_SEL_QWII and -Ox).  This could potentially be a verilation-time option, but I don't find that attractive.

Another Verilator fix would be to add padding to the end of arrays that need it, but this has several practical implementation problems, and wastes memory (and likewise performance).

It is tempting to turn off the address sanitizer around the "over-reads". While the sanitizer does support masking false-positives, it can only do this at function boundaries.  We could potentially wrap the hazard in a function just to allow this masking.  This doesn't seem great.

So in short many distasteful choices.  Need to think about it.

mytbk, 2021-03-08:
I just found that the fault in my project is caused by my wrong code instead of some write overflow, so I don't think it's an issue now.

wsnyder, 2021-03-08:
Yes, the code is presumably  missing the -1, what I mentioned above, but regardless of input code Verilator shouldn't be throwing bad addresses.

closed 2025-08-29 by wsnyder