verilator/verilator #2822: Survey: Help us improve the documentation

wsnyder, 2021-03-06:
We'd like to understand how you use the various documentation sources. If you could please fill in this survey it would be appreciated:

[Verilator Documentation Survey (Google Form)](https://forms.gle/neqKcNjungJfxB7N8)

This will help us plan our sources for the future. For example, we may change the current manpage-friendly style layout to something more book and read-the-docs/Python-style friendly.

If you want to discuss the documentation coversationally, also feel free to leave comments in this issue.

Thanks.

rfdonnelly, 2021-03-06:
I highly recommend [Antora](https://antora.org/).  It is an AsciiDoc based documentation site generator.  See the [Antora docs site](https://docs.antora.org/antora/2.3/) for an example of the output.

* Verilator is already using a lot of AsciiDoc
* Antora supports generating doc for multiple versions (tags).  Example from the Antora docs: 
[image]
* If man pages were written in AsciiDoc (or if you can [convert the existing POD man pages to AsciiDoc](https://metacpan.org/pod/Pod::AsciiDoctor)) then Antora can include these with your book-style documentation.  This allows you to have true man pages on the CLI but also have HTMLized versions that can be linked to.
* [PDF output](https://docs.antora.org/antora/2.1/project/roadmap/) support is planned (but unscheduled).  However, the source documentation could be organized in such a way as to generate a PDF today using AsciiDoctor (instead of Antora).

For a more thorough overview of all the features see https://docs.antora.org/antora/2.3/features/

mithro, 2021-03-07:
The rest of the open source EDA community is standardizing on Sphinx, this includes;
 - OpenROAD
 - SymbiFlow
 - Verilog to Routing
 - SureLog / UHDM
 - Fomu Workshop
 - Google's SkyWater PDK
 - (in future) Google's Verible
 - many others....

There continues to be a lot of work to make Sphinx an awesome set of tools for documenting both EDA and RTL projects, including;
 - https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams
 - https://github.com/SymbiFlow/sphinx-verilog-domain
 - https://github.com/SymbiFlow/symbolator
 - https://pypi.org/project/sphinxcontrib-wavedrom/
 - many more.

You also benefit from all the work going into closely related projects like Renode, Zephyr, Linux and many others too...

You can seem some older screenshots of this at https://github.com/SymbiFlow/sphinx_symbiflow_theme-old#screenshots

mithro, 2021-03-07:
The Verilog to Routing documentation is a good example too - https://docs.verilogtorouting.org/

It includes both user documentation and auto-generated API documentation (https://docs.verilogtorouting.org/en/latest/api/vpr/netlist/).

wsnyder, 2021-03-12:
Thanks for those whom answered.  Results:

How up to date is your version of Verilator?
* 70% It's often within three months of the most recent release
* 17% It's often more than three months old
* 6% It's not under my/my organization's control
* 6% Other/don't know

Please rate your desired/future use of these forms of Verilator documentation:

        Often Sometimes  Never
        77%     16%       7%    Web Read-the-Docs HTML
        25%     25%      50%    Web PDF
        16%     36%      48%    Web documentation from sources
         5%     36%      59%    Local machine's HTML
        13%     16%      71%     Local machine's PDF
        16%     36%      71%    Local machine's manual page
        46%     50%       4%    Local machine's command line output
        13%     34%      53%    Local repository documentation sources

Any other documentation format not listed above that you would use often? (Similar answers combined)
* Presentations

What is the minimal information "verilator --help" should provide?
* 76% "verilator --help" needs to indicate how to find the documentation in other places.
* 18% Other places and show summary.
* 0%  Other places and full help.
* 6%  Don't use verilator --help"

Any other thoughts or comments on documentation? (Similar answers combined)
* More examples. They can be links to code, scripts, etc in repo.
* Use something like Read-the-Docs

closed 2021-03-12 by wsnyder

mithro, 2021-03-12:
On the code / scripts / etc -- take a look at the SymbiFlow examples repo at https://github.com/SymbiFlow/symbiflow-examples and the Fomu Workshop at https://github.com/im-tomu/fomu-workshop