verilator/verilator #2823: Question: Ignoring %Error in not "generated" Verilog input.

berkakinci, 2021-03-06:
I'm just getting started using Verilator, so sorry if this is too much of a noob question.  BTW, thanks for yet another great tool!

I have "sanity checks" in various places in my Verilog, filling in for lack of SV asserts.  It goes like this:
```
   // Synthesis-time sanity Check
   generate
      if(FREQ_CLK_SAMPLE != 4*FREQ_CLK)
        __error__("DDRx4 Requires Edge Clock 4x the Parallel Clock.");
   endgenerate
```
The intention is to cause a show-stopper when this module is instantiated incorrectly so I don't chase my tail later.
Various synthesis tools and simulators haven't complained when the parameters are correct.  Verilator throws this error:
```
$ verilator --default-language "1364-2001" -F ./modulepathflags.vc --lint-only --Wall input_ddrx4_machxo3.v 
%Error: input_ddrx4_machxo3.v:19:18: syntax error, unexpected '(', expecting IDENTIFIER or randomize
   19 |         __error__("DDRx4 Requires Edge Clock 4x the Parallel Clock.");
      |                  ^
%Error: Exiting due to 1 error(s)
```

I traced the meaning down thanks to test cases for Issue468.  I see a way I could fix them for Verilator:  make a real module __error__ and don't use it anonymously.

Here is the question part: Is there an easier short-cut out of this?  Like telling Verilator "don't dive into unused generate branches?"

berkakinci, 2021-03-06:
Searching again, I do see there was a pull (#2754) recently that seems to indicate this shouldn't happen any more.  But it still does.  My version apparently should be the very version when this pull was first included:
`Verilator 4.110 2021-02-25 rev v4.108-79-g17b75c642`

wsnyder, 2021-03-06:
I'm not sure why the other tools are taking this as I don't see what makes this legal.  You could make it into a module definition but then Verilator will still try to resolve the module. Can you use SV asserts, and ifdef it there are tools you need that don't understand?  That's also better as if it fires you'll get a nice error. 

```
`ifdef VERILATOR  // Or any SV tool
 `define __error__(msg) $error(msg)
`else
 `define __error__(msg) __error__(msg)
`endif

module x;
   parameter X = 0;
   generate
      if (X != X)
        `__error__("DDRx4 Requires Edge Clock 4x the Parallel Clock.");
   endgenerate
endmodule

```

wsnyder, 2021-03-06:
This code isn't legal grammar as far as Verilator is concerned so #2754 doesn't apply.

berkakinci, 2021-03-06:
OK.  Thanks.  I'll try adopting something else with correct grammar across my modules.
I think I've gotten away with it for years because tools must not visit the unreachable generate branches.
Now another question -- excuse my being dense.  I know of two issues with the grammar: 1) __error__ is not a module or primitive that exists. 2) the instance is anonymous.
Is there something else I'm missing?

wsnyder, 2021-03-06:
It isn't a module instantiation as there's no such thing as "anonymous" modules, they must be named. (IEEE 23.3.2). It could be a UDP instance which Verilator doesn't support, but IEEE says they need at least two terminals (IEEE 29.8).  I suspect the other simulators are assuming it's a UDP instance and not checking the multi-terminal rule.  Might also be something that was legal pre-SystemVerilog.

berkakinci, 2021-03-06:
Thanks for answering -- all the way to the spec sections.  I didn't realize the UDP multi-terminal rule.
I'll think about it and fix across the board with valid grammar.  It's served me so far, but clearly it's time for an update.

closed 2021-03-06 by berkakinci