verilator/verilator #2824: Check missing coverage in new bit optimizations

wsnyder, 2021-03-07:
@yTakatsukasa thanks for adding the bit ops in #2751. Can you please look at the internal code coverage for this?  (https://codecov.io/gh/verilator/verilator/src/master/src/V3Const.cpp - you might need to login to see it).

There's a bunch of minor stuff I'm fine ignoring if you are, but currently all tests exit matchOrAndNot here https://github.com/verilator/verilator/blob/master/src/V3Const.cpp#L651 and we are never running the rest of the function. So might be worth adding a test to cover those cases.

Thanks.

wsnyder, 2021-03-07:
P.S. just committed adding exclusions to some of the debug statements not being covered, so ignore those.

yTakatsukasa, 2021-04-07:
During #2878 I found some code is unreachable due to #487
(else is unreachable so far as isTPure() returns true. Test in #2869 should cover the else block once isTPure() is fixed).
https://github.com/verilator/verilator/blob/7242ceb998e20b35feee4396e6df3adfba6cedd5/src/V3Const.cpp#L1109-L1117

closed 2021-05-01 by yTakatsukasa

yTakatsukasa, 2021-05-01:
I admit that the line coverage is not 100% yet. I'll send a PR when I come up with the piece of code to cover such lines or get convinced that a line is unreachable.

PR #2915 (2021-05-01): Tests: Add a case to cover the line using m_frozenNodes.
Closes #2824
#2878 and this PR improves coverage of V3Const.cpp

I will merge after CI passes as this is just adding test.