verilator/verilator #2825: Substitute module by C++ model

jotego, 2021-03-08:
Hello

I have a complex system in which, for a module deep in the hierarchy, I can provide a fast C++ model. I would like to simulate the system with Verilator but I'd like to run that particular module using my simple model.
Can this be done?
Can I provide a Verilog module definition where the body actually calls my C++ code? Can I integrate that with the rest of the system in Verilator?

Thank you

wsnyder, 2021-03-08:
There is no direct support for this, however there's at least these options:

1. Create a Verilog model that calls into DPI C routines at appropriate signal changes, and passes any signals to/from C through those calls.  This should work on any simulator, so recommended.

2. Create a Verilog model that calls $c("...") at appropriate signal changes. The C++ code can read the signals through either data passed to $c or verilator public indications.  This will be faster than above but less portable.

3. There's probably a way to use the partitioning to mark the child to be a separate Verilation run.  You'd then manually replace what would normally be the Verilated child with your code. This will require a lot of knowledge of the internals, but could be fully automated and general with appropriate improvements to Verilator.  You might want to do this to see what an example Verilog file would look like, but is otherwise probably least desirable.

Closing as answered but feel free to follow up and/or let us know what you do.

closed 2021-03-08 by wsnyder

jotego, 2021-03-08:
I will try the DPI approach first. Thank  you!