verilator/verilator #2826: --traces always slows down the simulation

jotego, 2021-03-08:
When I try to use more threads by adding the `--traces` argument to the compilation; I always find that the simulation runs significantly slower. For instance:

--trace | Simulation Wall Time (s)
---------|-------------------
1  | 12 
2 |  28
3 | 71

I do not vary anything in my C++ test bench. I just add `--traces` and run the code. I think I must be missing something important as the threads just don't work as intended.

This is my Verilator version:
`Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1`

Could you help me, please?

wsnyder, 2021-03-08:
Please try the most recent release as there's been many tracing changes and we prefer not to debug old releases, thanks.

jotego, 2021-03-08:
Thanky you for your comment. I have compiled and installed the latest version:

`Verilator 4.199 devel rev v4.110-32-g9483ebefa`
I still see the same behaviour. Do you want me to give you the test case?

wsnyder, 2021-03-08:
Please, ideally use test_regress format (see docs), but whatever you have, thanks.

wsnyder, 2021-03-08:
Also please make sure you are using numactl as described in the docs.

jotego, 2021-03-22:
Using `numactl -m 0 -C 0,1 -- ` in front of the simulation command rendered the same result. I will prepare the test case for you.

wsnyder, 2021-03-22:
It probably does the same but also please try "numactl -m 0 -C 0,2".  On some systems 0 & 1 are SMT on same CPU.

wsnyder, 2021-06-13:
Didn't see any additional updates so assuming you figured this out, feel free to post additional comments if needed.

closed 2021-06-13 by wsnyder

jotego, 2022-06-29:
Sorry for not answering back then, I tried and it didn't work and I just gave up.

A year later (today). I have tried again with a more complex design. Just in case the other one was simpler. I am simulating the whole Pang! arcade machine, including the FM chip, the CPU and the video system. I still find that the `--threads` option is significantly slower. The verilator version I'm using is "Verilator 4.224" and I have `numactl -m 0 -C 0,2` in front of the command.

No threads | threads = 2
-----|----
30s | 1m38

I have never seen Verilator perform faster during all the time I'm using it. It's a feature that really puzzles me.