verilator/verilator #2827: Interface array connection ordering regression in v4.110

deo248, 2021-03-09:
I believe there is a regression related to #2475 between versions 4.108 and 4.110. Specifically, updating to v4.110 exhibited a different behavior in related RTL that originally caused me to report #2475 . I had worked around the previous bug and the workaround code was working with verilator, vivado, and questa up to verilator version 4.108 but changed behavior in v4.110.

The original source code is a crossbar implementation that fans out interfaces from masters to an internal array of interfaces, then fans in from the internal array of interfaces ultimately leading to the crossbar slave ports. v4.110 changed the behavior such that that internal mapping is putting the upstream/master signals onto the wrong array indices, leading them to ultimately go to the wrong slave.

I've attached a fairly minimized testcase in test_regress format. The test passes in 4.108 and fails in 4.110. I _think_ the bad behavior is somehow related to the doubly nested for loop that does the crossbar cross connection assignments.

[t_mod_interface_array5.zip](https://github.com/verilator/verilator/files/6110831/t_mod_interface_array5.zip)

deo248, 2021-03-10:
I had a little extra time today to look into this further and run a bisect. It appears that fd6edd90 was the culprit that bisect identifies based on the above attached test, which is the related commit for #2475.

wsnyder, 2021-03-11:
Two of the great historical mistakes causing bugs: 0 to indicate string endings, and endian differences.  Anyhow think I understand how it broke this, will look at a fix shortly.

wsnyder, 2021-03-12:
Fixed in commit 29ed86606 misattributed to #2753.

Please give it a try on your larger code, thanks.

closed 2021-03-12 by wsnyder

deo248, 2021-03-12:
I checked this out on my larger design and it appears that the fix works. Many thanks for the fix and the quick turnaround!