verilator/verilator #2828: Bits OR not working correctly since v4.110

poemonsense, 2021-03-10:
**Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)**
[bit-reduce-or.tar.gz](https://github.com/verilator/verilator/files/6114169/bit-reduce-or.tar.gz)

Command line arguments: `verilator --cc *.v --top-module MissQueue`

Here is the simpler version:
In MissQueue.v, we use `fwrite` to output `_T_1054` and `_T_1215` in binary and decimal format respectively (search for `earlyValids=`).
`_T_1054` is a bit concatenation and _T_1215 is a bit reduce-OR.
```
1865   wire  _WIRE_36_earlyValid = entries_0_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1866   wire  _WIRE_37_earlyValid = entries_1_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1867   wire  _WIRE_38_earlyValid = entries_2_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1868   wire  _WIRE_39_earlyValid = entries_3_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1869   wire  _WIRE_40_earlyValid = entries_4_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1870   wire  _WIRE_41_earlyValid = entries_5_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1871   wire  _WIRE_42_earlyValid = entries_6_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1872   wire  _WIRE_43_earlyValid = entries_7_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1873   wire  _WIRE_44_earlyValid = entries_8_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1874   wire  _WIRE_45_earlyValid = entries_9_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1875   wire  _WIRE_46_earlyValid = entries_10_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1876   wire  _WIRE_47_earlyValid = entries_11_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1877   wire  _WIRE_48_earlyValid = entries_12_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1878   wire  _WIRE_49_earlyValid = entries_13_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1879   wire  _WIRE_50_earlyValid = entries_14_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]
1880   wire  _WIRE_51_earlyValid = entries_15_io_mem_finish_valid; // @[ReadyValidCancel.scala 71:19 ReadyValidCancel.scala 72:20]

1881   wire [7:0] lo_4 = {_WIRE_43_earlyValid,_WIRE_42_earlyValid,_WIRE_41_earlyValid,_WIRE_40_earlyValid,_WIRE_39_earlyValid
1882     ,_WIRE_38_earlyValid,_WIRE_37_earlyValid,_WIRE_36_earlyValid}; // @[Cat.scala 30:58]
1883   wire [15:0] _T_1054 = {_WIRE_51_earlyValid,_WIRE_50_earlyValid,_WIRE_49_earlyValid,_WIRE_48_earlyValid,
1884     _WIRE_47_earlyValid,_WIRE_46_earlyValid,_WIRE_45_earlyValid,_WIRE_44_earlyValid,lo_4}; // @[Cat.scala 30:58]

1937   wire  _T_1215 = _WIRE_36_earlyValid | _WIRE_37_earlyValid | _WIRE_38_earlyValid | _WIRE_39_earlyValid |
1938     _WIRE_40_earlyValid | _WIRE_41_earlyValid | _WIRE_42_earlyValid | _WIRE_43_earlyValid | _WIRE_44_earlyValid |
1939     _WIRE_45_earlyValid | _WIRE_46_earlyValid | _WIRE_47_earlyValid | _WIRE_48_earlyValid | _WIRE_49_earlyValid |
1940     _WIRE_50_earlyValid | _WIRE_51_earlyValid; // @[Arbiter.scala 107:36]
```
In the generated cpp files, _T_1054 is correct but _T_1215 is transformed to 
```
    vlTOPp->MissQueue__DOT___T_1215 = (IData)((0U !=
                                               (1U
                                                & (IData)(vlSymsp->TOP__MissQueue__DOT__entries_0.io_mem_finish_valid))));
```
That is, it only tests the first operator for _T_1215.

**What 'verilator --version' are you using?  Did you try it with the git master version?**
`Verilator 4.199 devel rev v4.110-32-g9483ebef`
v4.108 works well. But v4.110 and master version do not work as expected. 

**Would you be willing to try to fix Verilator yourself with assistance?**
Sorry but we are not able to locate the bug and fix it. I think the bug is caused by some commit between v4.108 and v4.110. Maybe 99c02419c096086303511fb2973282b3665169c6?

wsnyder, 2021-03-10:
Presumably related to #2751, @yTakatsukasa can you please take a look? Thanks.

wsnyder, 2021-03-10:
Think this is fixed, please try it.

Basically we were treating two vars with different scopes as the same, because this code sometimes runs past varscope removal.

closed 2021-03-10 by wsnyder

poemonsense, 2021-03-11:
It works. Thanks.