verilator/verilator #2831: Windows MSVC++ build issue

dt-snps, 2021-03-10:
I'm running into a compatibility issue when trying to compile a Verilated model using MSCV++ (2019) - using version 4.110.  

Build gives me the following error:

C:\cygwin\tools\verilator\4.110\share\verilator\include\verilated_sym_props.h(97): error C2397: conversion from 'VerilatedVarProps::<unnamed-enum-MAGIC>' to 'const vluint32_t' requires a narrowing conversion

Seems that enums in MSVC are 64-bit?  I can fix by explicitly casting to vluint32_t in the class, but it's a hack.

Running verilator with --compiler msvc doesn't help.

Is there some other way I can resolve this without hacking the .h file?

wsnyder, 2021-03-10:
Try adding a U to the end of the constant on that line, I suspect it really means it's upset about the sign bit.  If this works we'd appreciate a pull request to fix it.

wsnyder, 2021-03-10:
UL that is, editing comment...

dt-snps, 2021-03-10:
I tried that, but no joy. 

Is there a reason for using enum instead of just, eg. const vluint32_t MAGIC = 0xddc4f829; ?

wsnyder, 2021-03-10:
Only legacy that verilator didn't use C++11 originally, please try "constexpr vluint32_t MAGIC = 0xddc4f829UL;"

dt-snps, 2021-03-10:
"constexpr vluint32_t MAGIC = 0xddc4f829UL;" doesn't work - I think because MAGIC is defined inside the class.

Using "const" instead of "constexpr" will work, or defining it as "constexpr" outside of the class also works.

wsnyder, 2021-03-10:
Sorry should be "static constexpr vluint32_t MAGIC = 0xddc4f829UL;"  using just const will upset some compilers as needs declaration in .cpp file.

dt-snps, 2021-03-11:
OK, that worked.  Shall I do a pull request?

wsnyder, 2021-03-11:
Please, and add you name to docs/CONTRIBUTORS if not present, thanks!

wsnyder, 2021-03-11:
Thanks for your first pull!

closed 2021-03-11 by wsnyder