verilator/verilator #2832: enable  waveform dump in protect-lib mode

zhuzhzh, 2021-03-11:
I used protect-lib mode for cosim. 
TB is too complex, so vcs/xcelium could be used to handle that, while DUT is handled by verilator.

Unfortuantely, by default waveform dump is not enabled.

With little change, It actually can.

I write one time.cpp file to define sc_time_stamp() and enable traceEverOn.

```c
#include "svdpi.h"
#include "vpi_user.h"

typedef unsigned long long ullong;

extern "C" {
void traceon() {
     Verilated::traceEverOn(true);
}

ullong edahub_time() {
    static s_vpi_time t = {vpiSimTime, 0, 0, 0.0};
    uint64_t simtime;
    vpi_get_time(0, &t); 
    simtime = (((uint64_t)t.high <<32)|t.low);
    printf("time = %llu\n", simtime);
    return simtime;
}

int edahub_time_precision(){
  return vpi_get(vpiTimePrecision, 0); 
}

}

// Called by $time in Verilog
double sc_time_stamp() {
    return static_cast<double>(edahub_time());
}
```

in original DUT file, I add

```verilog
initial begin
  $display("[%0t] Tracing to logs/vlt_dump.vcd...\n", $time);
  $dumpfile("logs/vlt_dump.vcd");
  $dumpvars();
  $display("[%0t] Model running...\n", $time);
end
```

This solution will cause one re-dump issue.
```
%Warning: previous dump at t=131149, requesting t=131149, dump call ignored
```

The issue is caused by the evals in wrapper file for comb and sequential logic.

Don't know if there is solution to avoid this.

in C code, actually I want to get VerilatedVcd object, then use its member function set_time_resolution to set the time precision (as you see, I already use vip to get time precision (in edahub_time_precision).

I also tried to compile the DUT in non-protect-lib mode. Then I write one wrapper by myself to call eval(). With this method, I could create VerilatedVcd object and do anything I want.

wsnyder, 2021-03-12:
You are right the problem giving the trace message is the multiple eval calls in the multiple _update functions.  Really these _eval calls should be calls to eval_step. Then at the end of the master simulator's time stamp there needs to be a call to eval_end_step().  The hard part is getting that, perhaps use the VPI?  Another possibility is to have the Verilog use some sort of event which is activated on any of the other always statements, and then that event calls a new end_step_update() DPI function, that finally calls Verilator's end_step().

wsnyder, 2021-12-12:
#3200 added support for traceing through --lib-create, which I think replaces the need for most of what's discussed here.

closed 2021-12-12 by wsnyder