verilator/verilator #2834: Make fails when the filelist is too long

poemonsense, 2021-03-13:
When I tried to compile a large project `XSSimSoC` (the soc has two out-of-order superscalar cores), Verilator produces `VXSSimSoCV__[0-1857].cpp`, `VXSSimSoC__[0-1648]__Slow`, `VXSSimSoC__Trace__[0-496]`, `VXSSimSoC__Trace__[0-750]__Slow`. In the `ar -cr` step, the make command raises an error `make: execvp: /bin/sh: Argument list too long`. Mannually changing the command to `ar -cr VXSSimSoC__ALL.a VXSSimSoC*.o` works fine. 

To solve this issue, possibly change `$^` in `verilated.mk` line 211 to `$(VM_PREFIX)*.o` (not tested)?

wsnyder, 2021-03-13:
There might be other files with the prefix so I would prefer not to wildcard. ar supports an "@" option which could read the list - perhaps you could make a pull request to do this?

Alternatively if you don't have ~1000+ CPUs to distribute your compiles to, you may want to consider using a larger --output-split setting.

closed 2021-03-13 by wsnyder

poemonsense, 2021-03-14:
Thanks for your help. I didn't figure out an elegant way to dump the .o filelist to a file.

A larger --output-split should work but we probably need to scale this number up with the number of cores we want to simulate. It's better to let Verilator help us deal with the command line limits.

poemonsense, 2021-03-14:
By the way, when we upgrade to v4.200, the compiled emulator keeps printing `%Warning: $dumpvar ignored as not proceeded by $dumpfile`. However, the verilog source code doesn't contain any `$dumpvar`. In our cpp files, we do include the vcd headers, but we never call `Verilated::traceEverOn` or `dump`(we wrap these calls under a command argument). Our source code is at https://github.com/RISCVERS/XiangShan/blob/master/src/test/csrc/emu.cpp. Search `enable_waveform` for the `tfp` variable.

The Verilator source code is at `include/verilated.cpp:2206`. We just comment out this line to avoid printing. Should I open a new issue? Guess it's simply due to some unnecessary internal calls?

wsnyder, 2021-03-14:
Yes, better to file a new issue when you have a different problem. Anyhow looking at the code I think that warning can be false when using save-restore.  Fixed in git.  Feel free to comment out the warning if you don't wan to upgrade to Verilator head.

zhuzhzh, 2021-03-17:
I hit the "too long" issue as well when the --output-split setting is small.  The workaround is to put the command into one file then execute the file.
The better solution still is to have larger --output-split setting.