verilator/verilator #2835: Feature Request: Black box for modules which allow to lint-just-in-time

ToTamire, 2021-03-14:
There is already --bbox-sys and --bbox-unsup. Is there a chance to create blackbox for modules?

--bbox-modules
Black box any not found module. This option will disable %Warning-DECLFILENAME, "%Error: Cannot find file containing module: 'MODULENAME'" and also all "%Error: Pin not found: 'PINNAME'" errors for that module.

Linting one file is now a standard which allows write code much faster even in huge projects. With this option no one will have to compile whole project to check if there is a missed semicolon. One file linting allow in modern processors to compile whole file after each typed word or even each typed sign in reasonable time. 

I have already linting on Vivado, but linting one file takes 1'400ms. Same module on Verilator under wsl2 takes 150ms. So Verilator is 9 times faster. Verilator also lint code style warnings. But Vivado allow to compile only one file and Verilator not...

wsnyder, 2021-03-14:
I can see how this is useful.  Creating the missing module as a shell is the easy part (so to speak) and Verilator already makes a NotFoundModule internally.

The hard part is resolving the implications of not having that module and suppressing the errors/warnings that would otherwise result.  My understanding is such false warnings/errors would break your desired use case.
* What if there's a "sub_module not_found (.*)". The missing module could be consuming any signal, so all UNUSED warnings are false.  Likewise it may be implicitly creating wires on the upper module, so anything undefined is ok.
* What if the instantiation is really an interface?
* What about dotted references to inside the missing module?  The data type can't be known.
* What if the missing module declares a global function, parameter, or something else visible to the upper module.  

A minor point is DECLFILENAME seems orthogonal to this, if you don't want it turn it off.  I agree though a blacked boxed module should not throw this warning.

ToTamire, 2021-03-15:
When whole project can be linted without errors / warnings, then every file in project should be linted without errors / warnings.

The main problem is that I don't know SystemVerilog well, so I can only speak about Verilog.

In my opinion all signals passed to / from NotFoundModule should not set unused signal warnings - assume that signal is used by module. 
Dotted references to inside the missing module also should not set warning or error if name after dot is valid (not start with number etc).
Root scope declaration in Verilog is forbidden so all globals should be declared in .vh files and then --relative-includes allow to include globals to each module. 

SystemVerilog allow scope declarations, so everything declared outside module is global. So in SystemVerilog all undefined signals should be valid, but this let programmer to make undetectable typo in names... I check in Vivado and it don't show errors only when access to not included global parameters by $root.

ToTamire, 2021-04-29:
I was thinking about it for a while. Learning about [verible](https://github.com/google/verible) was groundbreaking. This tool work as I said before:

> When whole project can be linted without errors / warnings, then every file in project should be linted without errors / warnings.

The main problem is that this tool lint nothing. Maybe not nothing. It shows missed semicolons... So I think that there should be some coding style restrictions that will allow to almost fully lint one file. 
If we want access global scope then global scope should be visible by ```--relative-includes``` (and also inlcude directive in linted file) or access by $root. 
If name after $root follow naming conventions (not start with number etc) then we should tread is as valid (data type too).
```systemverilog
bit [31:0] word;
assign word = $root.data;
```
But when we assign blackbox signal to two different signals (with other data type), then we should have warning. How to implement that? I think that when linting engine parse file then backbox signal should get same datatype as first assignment occurrence. In example below set $root.data data type to ```bit [31:0]``` and then line below we get some data type mismatch warning.
```systemverilog
bit [31:0] word;
bit [15:0] halfword;
assign word = $root.data;
assign halfword = $root.data;
```

> The missing module could be consuming any signal, so all UNUSED warnings are false. Likewise it may be implicitly creating wires on the upper module, so anything undefined is ok.

UNUSED warning should be true and undefined signals should be not ok.