verilator/verilator #2836: Documentation reorganization proposal (comments welcome)

wsnyder, 2021-03-14:
Based on feedback collected in (#2822), we'd like to restructure the documentation.

Here's the thinking - please post your comments.

* Primary documentation will be read-the-docs style (if not literally read-the-docs), probably with multi-version support.
* Verilator.pdf/html will remain as a formatted version of the read-the-docs,  important for local documentation for those installing from packages.
* verilator --help will show only an option summary.

There will be two primary documentation trees, the user documentation and internal documentation.

## User Documentation Proposed Outline

  * Overview [was Description]

  * Examples
      * Quick Start
      * Example C++ Execution [was head1]
      * Example SystemC Execution [was head1]
      * Other Examples [new, describe examples/ dir]

  * Installation
      * mostly ./docs/install.rst
      * Docker Environment
        * Build [./ci/docker/buildenv/README.rst]
        * Run [./ci/docker/run/README.rst]

  * Input Languages
      * Language Standard Support
      * Language Extensions
      * Language Limitations
      * Verilator Configuration Files (.vlt)  [or below under arguments]

  * Verilating
      * Multithreading [was head1]
      * Hierarchical Verilation [was head1]
      * Cross Compilation [was head1]
      * CMake [was head3]

  * Connecting to Verilated Models
      * Connecting to C++ [was head1]
      * Connecting to SystemC [was head1]
      * Direct Programming Interface (DPI)  [was head1]
      * Verification Procedural Interface (VPI)  [was head1]
      * Wrappers and Model Evaluation Loop
      * Verilated and VerilatedContext

  * Simulating (Verilated-Model Runtime)
      * Benchmarking & Optimization
      * Profiling
        * [discuss verilator_gantt]
        * [discuss verilator_profcfunc]
      * Coverage Analysis
        * [discuss verilator_coverage]
  * Files and Environment
      * Files in the Distribution
      * Files Read/Written
      * Environment

  * Errors and Warnings
      * Error And Warning Format
      * List Of Warnings  [generate from data file?]

  * Executable and Argument Reference
      * verilator
        * Argument Summary [extract into bin/verilator]
        * List of Arguments
      * verilator_coverage.1 [description links to Coverage]
        * Arguments etc
      * verilator_gantt.1 [description links to Coverage]
        * Arguments etc
      * verilator_profcfunc.1 [description links to Profiling]
        * Arguments etc
      * Simulation Runtime Arguments

  * Verilated Library API Reference
      * [doxygen of include/verilated*.h]

  * Deprecations

  * FAQ/Frequently Asked Questions [sep file?]

  * Reporting Bugs [was Bugs]
       move under contributions below?

  * History and Contributions
      * Authors
      * Contributors [script include ./docs/CONTRIBUTORS]
      * Contributing [from ./docs/CONTRIBUTING.rst]
      * Historical Origin
      * Revision History [from ./Changes]

  * Copyright [old distribution section]

###  Internals Proposed Outline

  * Mostly from ./docs/internals.rst
  * XML Output Format  ./docs/xml.rst
  * [later project] ./test_regress/driver.pl
  * [later project] doxygen of internals

jalcim, 2021-04-03:
I suggest to adding an section for building !

Actually i don't find on the documentation how to build multiple design, or multiple file design.
I search too an solution for compiling my verilog design on library (.a or .so).

simdream, 2021-04-05:
app notes are perferred as well. several topics as we would like and think useful for all:

how to compile a large design
    some guideline and limitation on hierarchy
    atention needed when manually split the design  (manual hierarchical or partition)

how to write the testbench
    sometimes to update both inout data and clock together could creat race condition in eval. do update_data_input -> eval -> update_clock_input -> eval will clearly double the number of evalation so hurt the runtime performance

thanks

wsnyder, 2021-04-13:
Documentation has been rewritten and is available here:  https://verilator.org/guide/latest/

Thanks all for feedback, if you have additional changes or comments please file a new issue (or ideally a pull request to fix the docs), thanks.

closed 2021-04-13 by wsnyder