verilator/verilator #2837: Verilator Multithreaded not Working on Apple Silicon

BracketMaster, 2021-03-15:
Verilator works mostly on new Apple Silicon, but only with `--threads 1``
Unfortunately, [cpuid_count](https://clang.llvm.org/doxygen/cpuid_8h_source.html) for llvm
was never and probably won't be ported to ARM.

Verilator references `cupid_count` to set some multithreaded settings on
Apple platforms as seen [here](https://github.com/verilator/verilator/blob/master/include/verilated_threads.h#L154)

> Would you be willing to try to fix Verilator yourself with assistance?

Sure. This shouldn't be too hard to fix. I'd just need to know what verilator is
looking for from `cpuid_count`

BracketMaster, 2021-03-15:
I suspect that it uses `cupid_count` for NUMA control?
If so, could probably just remove it entirely for Apple Silicon
and let the OS decide how to pin threads to cores?

wsnyder, 2021-03-16:
You can temporarily comment that out and the only side effect is the gantt profiling won't work correctly.  I'm sure there's some way to get the information so if you could please figure out the appropriate code & ifdef's and make a pull it would be appreciated.

wsnyder, 2023-09-16:
Closing due to age - might even be fixed.  If not, we welcome a pull request for this or any portability enhancements.

closed 2023-09-16 by wsnyder