verilator/verilator #2841: Is there mehtod to reduce the C++ file generation or debug which module causes this?

zhuzhzh, 2021-03-17:
I am trying verilator on two designs. Both of them hit the long compile time issue.

One of them is about 40MGates. (reported by emulator synthesis tool)

With the basic option (no --trace --debug), it took about 20 hours to generate C++ file.

```shell
verilator --output-split 10000 --exe -Wno-WIDTH -Wno-fatal -Wno-MODDUP  -Wno-TIMESCALEMOD -Wno-BLKANDBBLK -Mdir obj_dir -cc -02 --top-module chip_tb -j 10 --timescale-override 1ps/1ps -sv -f dut.flist chip_tb.sv -f ./model.f -f ./define.flist sim_main.cpp
```

I want to check if any module has complex structure, which makes verilator slow.
but I don't know if any method is provided except enabling --debug option to output large log file.
With log file, actually it's still hard to figure out the useful information.

wsnyder, 2021-03-17:
It won't point at a module but please try running with --stats, and see where the time is being spent.  If it's relatively flat times per stage it's just large.  If most all of the time is in one stage, it's more likely a bug.

You can also use --prof-cfuncs and see what filenames end up how often in the generated code, that likely correlates to verilation time.  Perhaps consider making a contribution that takes the resulting information and puts into the --stats report how many instructions came from each of the input files (e.g. indexes instruction counts back to filelines).

zhuzhzh, 2021-03-22:
I am thinking if we could compile some subsystems into library, then we only need compile the volatile parts every time. I am concerned the timing issue. It seems the interface between the different library may have half or one cycle differ.

wsnyder, 2021-03-23:
--hierarchical isn't perfect but does try to line things up properly.

zhuzhzh, 2021-03-26:
The weird thing is that if --debugi 1 is added, the error is hit.

[image]
[image]

wsnyder, 2021-03-26:
--debug enables many internal assertions, including the one that you unfortunately hit.  You didn't attach the backtrace that Verilator suggested, but if it is in V3Combine then try running with -Ob.

zhuzhzh, 2021-03-26:
Thanks, I will try -Ob.
Trace the signal in the first snapshot. There is one signal named last, which is unlinked.

```verilog
assign wo_mux_c_lo.last = wo_mux_o_lo.last;

assign wo_load[0] = wo_push_eff & ~wo_inp_sel;
assign wo_load[1] = wo_push_eff & wo_imp_sel;
```

`wo_mux_c_lo` and `wo_mux_o_lo` are the struct.
Look like it's optimized out.

wsnyder, 2021-08-24:
Think this was answered, if a fix is needed please file a new issue with a test case.

closed 2021-08-24 by wsnyder