verilator/verilator #2842: Internal error: Consumer doesn't match lhs of assign

sjalloq, 2021-03-17:
Hi there,

I'm running v4.106 in `--lint-only` mode and am getting the following internal error.

```
INFO: Building simulation model
verilator -f tigris_system_tigris_0.1.vc -Wall
%Error: Internal Error: ../src/tigris_system_io_0.1/src_rtl/tigris_io.sv:269:21: ../V3Gate.cpp:1099: Consumer doesn't match lhs of assign
  269 |                    .DVDD(VDDD33_io),
      |                     ^~~~
                        ... See the manual and https://verilator.org for more assistance.
```

The line in question is a power rail for an IO cell.  I've created a version of the Verilog cell views with nothing but port declarations for lint use so there's nothing at the end of this net.  It's a top level port that feeds all cells and is declared as an `inout` throughout the hierarchy.

Any pointers as to what to look for?

sjalloq, 2021-03-17:
Same result with v4.200

wsnyder, 2021-03-17:
Can you make a small example showing this?  You might need the "-Oi" flag to turn off inlining to make a small example.

sjalloq, 2021-03-18:
Here's a horribly obfuscated testcase for you.  In `io.sv` there are a number of technology cells instanced.  If I remove one of these, and it doesn't seem to matter which one, then the issue doesn't reproduce.  Alternatively, if I edit the base module (jz5a2dOx8Y) in io_digital.sv to remove one of the two duplicated base cells, the issue also goes away.

[verilator_testcase_2842.tar.gz](https://github.com/verilator/verilator/files/6164387/verilator_testcase_2842.tar.gz)

wsnyder, 2021-03-18:
I get some warnings that need to be suppressed, but your example verilates OK for me.

sjalloq, 2021-03-19:
Looks like it does for me too on an Ubuntu system.  My work machine is RHE7 based.  Verilator was compiled on that machine from source.  Any thoughts?

wsnyder, 2021-03-19:
0. Are you sure "verilator -V" shows same version?  It's odd I got warnings which were fatal and you didn't.

1. You need "*q*" as need to match the ' at the end.  Not the problem this time, but you should read the .vlt file first as some warnings might happen before it has been read.

2. ahh. ok.  Will need some work as seems deep in the gate optimizer, if you can remove the unused inouts or make them inputs you might make progress around it. Ran it through my testcase reducer and down to 60 lines so it's at least manageable to debug now.

wsnyder, 2021-03-19:
[all.v.txt](https://github.com/verilator/verilator/files/6168264/all.v.txt)

sjalloq, 2021-03-19:
Tested with both of the following versions and both behave the same:

Verilator 4.106 2020-12-02 rev v4.106
Verilator 4.200 2021-03-12 rev v4.200

Edgeworth, 2021-06-11:
I also ran into this bug. Can't say for sure if it's exactly the same root cause, but my bisect shows that this error was first introduced for me in de931d6a97930a0ece0294b2b671979e393e489c

This patch 'fixes' the behaviour for me (essentially a revert of de931d6a97930a0ece0294b2b671979e393e489c), although I presume de931d6a97930a0ece0294b2b671979e393e489c fixes a real issue and that removing this line will reintroduce some bug.

```diff
diff --git a/src/V3Tristate.cpp b/src/V3Tristate.cpp
index a4b7515d..c0a4d69c 100644
--- a/src/V3Tristate.cpp
+++ b/src/V3Tristate.cpp
@@ -1166,7 +1166,6 @@ class TristateVisitor final : public TristateBaseVisitor {
                 AstVar* enVarp = new AstVar(nodep->fileline(), AstVarType::MODULETEMP,
                                             nodep->name() + "__en" + cvtToStr(m_unique++),
                                             VFlagBitPacked(), enModVarp->width());
-                enModVarp->direction(VDirection::INPUT);
                 UINFO(9, "       newenv " << enVarp << endl);
                 AstPin* enpinp
                     = new AstPin(nodep->fileline(), nodep->pinNum(),
```

wsnyder, 2021-06-12:
Thanks for the report. First thing is we'd need a test with the change - can you please provide a test (ideally in test_regress format - see the docs) that shows the problem.

Then the bigger issue is removing this line breaks a test, t_altera_lpm_mult_noinl.

Then if possible see if you can find a case that allows both the old test and your new one to pass?

Edgeworth, 2021-06-13:
W.r.t. breaking test case, this test case posted earlier in this thread shows the issue:

> [all.v.txt](https://github.com/verilator/verilator/files/6168264/all.v.txt)

N.B. this bug seems to be affected by the level of optimisation - the above test case works if -O3 but doesn't with no optimisation flags. However, with a complicated enough input this bug does appear with -O3 (above case doesn't do it though).

W.r.t. the change I posted - this isn't intended as a fix, it's intended as a hint/starting point to working out what the underlying bug could be, when someone has time to look at the issue.

ldoolitt, 2022-02-26:
This issue was solved for my use case by commit 4e5f308, Jan 5, 2022, by @yTakatsukasa.
As in, [Edgeworth](https://github.com/Edgeworth)'s workaround (that I had been using) no longer applies, and the fault doesn't come back.

yTakatsukasa, 2022-02-28:
Thanks for the update.
I should have found this issue before filing #3258
I assume this issue can be closed as #3268 is in master now.

closed 2022-02-28 by yTakatsukasa