verilator/verilator #2846: Using an array in a function for parameter computation

ganoam, 2021-03-19:
Hi, 

I have a use case similar, but not exactly the same as what was reported and fixed in #1315:

The following works (example from #1315):
```
module ex24
 (
   input wire i_clk,
   input wire i_rst_n
 );
   parameter int SIZES [3:0] = '{1,2,3,4};
   typedef int calc_sums_t [3:0];
   function calc_sums_t calc_sums;
     begin
       int ii, sum = 0;
       for (ii=0; ii<4; ii++) begin
        sum = sum + SIZES[ii];
        calc_sums[ii] = sum;
       end
     end
   endfunction
   parameter int SUMS[3:0] = calc_sums();
 endmodule
```
The following does not:
```
module ex24
 (
   input wire i_clk,
   input wire i_rst_n
 );
   parameter int SIZES [3:0] = '{1,2,3,4};
   typedef int calc_sums_t [3:0];
   function calc_sums_t calc_sums(int s[], int n);
     begin
       int ii, sum = 0;
       for (ii=0; ii<n; ii++) begin
        sum = sum + n[ii];
        calc_sums[ii] = sum;
       end
     end
   endfunction
   parameter int SUMS[3:0] = calc_sums(SIZES, 4);
 endmodule
```
The Error messages:
```
%Error: ex24.sv:37:40: Expecting expression to be constant, but variable isn't const: 'SIZES'
                    : ... In instance ex24
   37 |    parameter int SUMS[3:0] = calc_sums(SIZES, 4);
      |                                        ^~~~~
%Error: Internal Error: ex24.sv:37:40: ../V3Simulate.h:294: No value found for node.
                                    : ... In instance our
   37 |    parameter int SUMS[3:0] = calc_sums(SIZES, 4);
      |                                        ^~~~~
```
Is this expected behavior? 

I am happy to help, if I can (I am not familiar with the codebase).

wsnyder, 2021-03-19:
First, trivial but confused me for a minute, "n[ii]" should presumably be "s[ii]"

Next, if you change to use standard unpacked arrays this is fixed in git, and added t_dynarray_param test with:

   function calc_sums_t calc_sums(int s[3:0], int n);

note if you go with this workaround using the 3:0 will have backwards index ordering versus your original example with dynamic arrays.  (See t_dynarray_param - that the expected results flipped also confused me for a few minutes - but this is correct based on IEEE endian indexing.)

As to the example as originally provided, currently dynamic arrays aren't supported in constant functions.  If you'd like to support this it would probably be straightforward - currently V3Width makes a METHODHARD "at" meaning access an element.  This should change to be a new AstNode type, then that type should be decoded in V3Simulate to properly get an element from a dynamic array.  Should be straight forward, but a decent task if your first time in the code.

ganoam, 2021-03-22:
Alright, thanks a lot @wsnyder. Your fix already helps, but would be nice to have support for dynamic arrays. I will look into it, but it may take some time!

wsnyder, 2023-08-30:
Passes now, fixed at some point earlier.

closed 2023-08-30 by wsnyder