verilator/verilator #2847: make_top_shell issue

marlonjames, 2021-03-19:
I'm working on a VPI test, and the auto-generated `X__top.v` file is interfering with `cbNextSimTime` callback timing when testing with Icarus Verilog.

I tried this in my .pl file:
```perl
compile(
    make_top_shell => 0,
    ...
```

But the file is still being generated and passed to `iverilog`, e.g. `obj_iv/t_vpi_X/Vt_vpi_X__top.v`.

Is it possible to not have this file generated and compiled as part of the test?

Using version:
Verilator 4.109 devel rev v4.108-42-gf2a7f30b

wsnyder, 2021-03-20:
Looks like a bug, try this:

```
--- a/test_regress/driver.pl
+++ b/test_regress/driver.pl
@@ -969,7 +969,7 @@ sub compile {

     compile_vlt_cmd(%param);

-    if (!$self->{make_top_shell}) {
+    if (!$param{make_top_shell}) {
         $param{top_shell_filename}
         = $self->{top_shell_filename} = $self->{top_filename};
     } else {
```

It also looks like it writes out the _top file even when make_top_shell is clear, which is silly and I will fix it, but should be harmless if no longer it is referred to.

marlonjames, 2021-03-31:
I tried your change but it just overwrites the contents of my test verilog file in `_make_top_v`.

wsnyder, 2021-03-31:
I think everywhere _make_top is called do this:

```
-        $self->_make_top();
+        $self->_make_top() if $param{make_top_shell};
```

marlonjames, 2021-04-01:
That seems to work when I also set `$param{top_shell_filename}` to an empty string. If I don't, `top_filename` is passed twice.

```diff
--- a/test_regress/driver.pl
+++ b/test_regress/driver.pl
@@ -958,9 +958,9 @@ sub compile {

     compile_vlt_cmd(%param);

-    if (!$self->{make_top_shell}) {
+    if (!$param{make_top_shell}) {
         $param{top_shell_filename}
-        = $self->{top_shell_filename} = $self->{top_filename};
+        = $self->{top_shell_filename} = "";
     } else {
         $param{top_shell_filename}
         = $self->{top_shell_filename} = "$self->{obj_dir}/$self->{VM_PREFIX}__top.".$self->v_suffix;
```

Thanks for the help.

closed 2021-04-01 by wsnyder