verilator/verilator #2848: Linking: undefined reference

forember, 2021-03-21:
> Thanks for taking the time to report this.
>
> Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I unfortunately don't know what test_regress format is or if it is applicable, but here is the code with which I am having the issue:

`vga/display_if_sdl.v`:

```verilog
module display_if_sdl (
        input clk,
        input [12:0] mem_waddr,
        input [23:0] mem_wdata,
        input mem_web);

import "DPI-C" function void
    sdl_write(input integer y, input integer x, input integer r, input integer g, input integer b);

always @(posedge clk) begin
    if (mem_web) begin
        sdl_write({26'h0,mem_waddr[12:7]}, {25'h0,mem_waddr[6:0]},
            {24'h0,mem_wdata[23:16]}, {24'h0,mem_wdata[15:8]}, {24'h0,mem_wdata[7:0]});
    end
end

endmodule
```

`vga/display_if_sdl.cpp`:

```c++
#include "Vdisplay_if_sdl.h"
#include "verilated.h"
#include <SDL2/SDL.h>
#include <SDL2/SDL_video.h>
#include <SDL2/SDL_render.h>
#include <SDL2/SDL_events.h>
#include <SDL2/SDL_keyboard.h>
#include <SDL2/SDL_keycode.h>

SDL_Window* window;
SDL_Renderer* renderer;
int pixel_size = 4;

void sdl_write(int y, int x, int r, int g, int b) {
    SDL_Rect rect;
    rect.y = pixel_size * y;
    rect.x = pixel_size * x;
    rect.h = pixel_size;
    rect.w = pixel_size;
    //printf("RGB: 0x%02x%02x%02x\n", r, g, b);
    SDL_SetRenderDrawColor(renderer,
        (Uint8)r, (Uint8)g, (Uint8)b, SDL_ALPHA_OPAQUE);
    SDL_RenderFillRect(renderer, &rect);
    SDL_RenderPresent(renderer);
}

int main(int argc, char** argv, char** env) {
    Verilated::commandArgs(argc, argv);
    Vdisplay_if_sdl* top = new Vdisplay_if_sdl;
    top->clk = 0;
    top->mem_waddr = 0;
    top->mem_wdata = 0;
    top->mem_web = 0;
    bool done = false;

    SDL_Init(SDL_INIT_VIDEO);
    SDL_CreateWindowAndRenderer(128 * pixel_size, 64 * pixel_size, 0,
        &window, &renderer);

    while (!(Verilated::gotFinish() || done)) {
        top->clk = 0;
        top->eval();
        top->clk = 1;
        top->eval();

        if (++top->mem_waddr >= 8192) {
            top->mem_waddr = 0;
        }
        uint32_t r = (top->mem_waddr >> 9) & 0x0f;
        uint32_t g = (top->mem_waddr >> 4) & 0x1f;
        uint32_t b = top->mem_waddr & 0x0f;
        top->mem_wdata = (r << 20) | (g << 11) | (b << 4);
        top->mem_web = 1;

        usleep(20);

        SDL_Event event;
        while (SDL_PollEvent(&event)) {
            switch (event.type) {
                case SDL_KEYDOWN:
                    if (event.key.keysym.sym == SDLK_q) {
                        done = true;
                    }
                    break;
                case SDL_QUIT:
                    done = true;
                    break;
                case SDL_WINDOWEVENT:
                    if (event.window.event == SDL_WINDOWEVENT_CLOSE) {
                        done = true;
                    }
                default:
                    break;
            }
        }
    }
    
    SDL_Quit();
    top->final();
    delete top;
    exit(0);
}
```

Build commands:

```bash
$ verilator -Wall -LDFLAGS -lSDL2 --cc vga/display_if_sdl.v --exe vga/display_if_sdl.cpp
$ make -j -C obj_dir -f Vdisplay_if_sdl.mk Vdisplay_if_sdl
```

Error output:

`Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b`

```text
make: Entering directory '/home/embers/Documents/undergrad-projects/mips_cpu_611/obj_dir'
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o display_if_sdl.o ../vga/display_if_sdl.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated.o /usr/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated_dpi.o /usr/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vdisplay_if_sdl.cpp Vdisplay_if_sdl__Dpi.cpp Vdisplay_if_sdl__Slow.cpp Vdisplay_if_sdl__Syms.cpp > Vdisplay_if_sdl__ALL.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vdisplay_if_sdl__ALL.o Vdisplay_if_sdl__ALL.cpp
Archive ar -cr Vdisplay_if_sdl__ALL.a Vdisplay_if_sdl__ALL.o
g++    display_if_sdl.o verilated.o verilated_dpi.o Vdisplay_if_sdl__ALL.a   -lSDL2    -o Vdisplay_if_sdl
/usr/bin/ld: Vdisplay_if_sdl__ALL.a(Vdisplay_if_sdl__ALL.o): in function `Vdisplay_if_sdl::__Vdpiimwrap_display_if_sdl__DOT__sdl_write_TOP(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)':
Vdisplay_if_sdl__ALL.cpp:(.text+0x69): undefined reference to `sdl_write'
collect2: error: ld returned 1 exit status
make: *** [Vdisplay_if_sdl.mk:64: Vdisplay_if_sdl] Error 1
make: Leaving directory '/home/embers/Documents/undergrad-projects/mips_cpu_611/obj_dir'

```

`Verilator 4.201 devel rev v4.200-37-ga1ab295b (mod)`

```text
make: Entering directory '/home/embers/Documents/undergrad-projects/mips_cpu_611/obj_dir'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o display_if_sdl.o ../vga/display_if_sdl.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o verilated_dpi.o /usr/local/share/verilator/include/verilated_dpi.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vdisplay_if_sdl.cpp Vdisplay_if_sdl__Dpi.cpp Vdisplay_if_sdl__Slow.cpp Vdisplay_if_sdl__Syms.cpp > Vdisplay_if_sdl__ALL.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o Vdisplay_if_sdl__ALL.o Vdisplay_if_sdl__ALL.cpp
Archive ar -rcs Vdisplay_if_sdl__ALL.a Vdisplay_if_sdl__ALL.o
g++    display_if_sdl.o verilated.o verilated_dpi.o Vdisplay_if_sdl__ALL.a   -lSDL2    -o Vdisplay_if_sdl
/usr/bin/ld: Vdisplay_if_sdl__ALL.a(Vdisplay_if_sdl__ALL.o): in function `Vdisplay_if_sdl::__Vdpiimwrap_display_if_sdl__DOT__sdl_write_TOP(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)':
Vdisplay_if_sdl__ALL.cpp:(.text+0x71): undefined reference to `sdl_write'
collect2: error: ld returned 1 exit status
make: *** [Vdisplay_if_sdl.mk:64: Vdisplay_if_sdl] Error 1
make: Leaving directory '/home/embers/Documents/undergrad-projects/mips_cpu_611/obj_dir'
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

Tested initially with `Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b`
Then with git master `Verilator 4.201 devel rev v4.200-37-ga1ab295b (mod)`

> Would you be willing to try to fix Verilator yourself with assistance?

Yes, absolutely.

wsnyder, 2021-03-21:
The DPI for all simulators requires "C" linkage, so try this

    extern "C" void sdl_write(int y, int x, int r, int g, int b) {

Or, add at top

    "Vdisplay_if_sdl__Dpi.h"

If that's not it let us know.

forember, 2021-03-21:
License:

<p xmlns:dct="http://purl.org/dc/terms/" xmlns:vcard="http://www.w3.org/2001/vcard-rdf/3.0#">
  <a rel="license"
     href="http://creativecommons.org/publicdomain/zero/1.0/">
    [image]
  <br />
  To the extent possible under law,
  <a rel="dct:publisher"
     href="https://github.com/forember">
    <span property="dct:title">Emberlynn McKinney</span></a>
  has waived all copyright and related or neighboring rights to
  <span property="dct:title">the code in the above comment</span>.
This work is published from:
<span property="vcard:Country" datatype="dct:ISO3166"
      content="US" about="https://github.com/verilator/verilator/issues/2848">
  United States</span>.
</p>

forember, 2021-03-21:
Ah okay I'll give that a try.

forember, 2021-03-21:
Now I get:

```text
../vga/display_if_sdl.cpp:14:17: error: conflicting declaration of C function ‘void sdl_write(int, int, int, int, int)’
   14 | extern "C" void sdl_write(int y, int x, int r, int g, int b) {
      |                 ^~~~~~~~~
In file included from ./Vdisplay_if_sdl.h:12,
                 from ../vga/display_if_sdl.cpp:1:
./Vdisplay_if_sdl__Dpi.h:17:17: note: previous declaration ‘void sdl_write(const svLogicVecVal*, const svLogicVecVal*, const svLogicVecVal*, const svLogicVecVal*, const svLogicVecVal*)’
   17 |     extern void sdl_write(const svLogicVecVal* y, const svLogicVecVal* x, const svLogicVecVal* r, const svLogicVecVal* g, const svLogicVecVal* b);
      |                 ^~~~~~~~~
```

on both versions tested.

forember, 2021-03-21:
Oh, do I need to make them pointers? Let me try that.

wsnyder, 2021-03-21:
import "DPI-C" function void
    sdl_write(input integer y, input integer x, input integer r, input integer g, input integer b);

You need

     import "DPI-C" function void sdl_write(input int y, input int x, input int r, input int g, input int b);

Hopefully with that you're good!

wsnyder, 2021-03-21:
Or keep the import statement and your C code takes pointers but then it needs to handle the 4-state structures.

forember, 2021-03-21:
Changing to `int` fixed it! Thank you!

closed 2021-03-21 by forember