verilator/verilator #2849: Dubious WIDTHCONCAT warning with 32-bit type cast

rswarbrick, 2021-03-23:
Consider the following example:
```systemverilog
module foo (input logic         a,
            output logic [33:0] b33,
            output logic [32:0] b32,
            output logic [31:0] b31);

  assign b33 = {33'(0), a};
  assign b32 = {32'(0), a};
  assign b31 = {31'(0), a};

endmodule
```
When run with `verilator --lint-only`, you get this:
```
$ /opt/verilator-test/bin/verilator --lint-only concat.sv
%Warning-WIDTHCONCAT: concat.sv:7:21: Unsized numbers/parameters not allowed in concatenations.
                                    : ... In instance foo
    7 |   assign b32 = {32'(0), a};
      |                     ^
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
%Warning-WIDTHCONCAT: concat.sv:7:23: Unsized numbers/parameters not allowed in replications.
                                    : ... In instance foo
    7 |   assign b32 = {32'(0), a};
      |                       ^
%Error: Exiting due to 2 warning(s)

```
I assume that what's happening is that the cast from `0` (a 32-bit int) to `32'(0)` is being optimised away and then some downstream pass is seeing a bare zero in the concatenation? The second warning also looks a bit odd, since this isn't a replication.

I've reproduced this with the tip of master:
```
$ /opt/verilator-test/bin/verilator --version
Verilator 4.201 devel rev v4.200-38-g2e158d88
```

wsnyder, 2021-03-23:
It wasn't supposed to warn in the case given, the cast should have resolved the size, will look into why it didn't.

wsnyder, 2021-03-23:
Fixed in master git, thanks for the small test case.

closed 2021-03-23 by wsnyder