verilator/verilator #2850: CASEOVERLAP warning for a priority case

rswarbrick, 2021-03-23:
With the following code,
```systemverilog
module cases(input logic [1:0]  foo_i,
             output logic [2:0] bar_o);

  always_comb begin
    priority casez (foo_i)
      2'b ?1: bar_o = 3'd0;
      2'b 1?: bar_o = 3'd1;
      default: bar_o = 3'd2;
    endcase
  end

endmodule
```
verilator generates a CASEOVERLAP warning:
```
$ /opt/verilator-test/bin/verilator --lint-only cases.sv
%Warning-CASEOVERLAP: cases.sv:7:7: Case values overlap (example pattern 0x3)
    7 |       2'b 1?: bar_o = 3'd1;
      |       ^~~~~~
                      ... Use "/* verilator lint_off CASEOVERLAP */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

The cases do indeed overlap, but that's kind of the whole point of the `priority` keyword in the spec. Maybe this warning needs disabling for `priority-case` expressions?

Tested with current head of master:
```
$ /opt/verilator-test/bin/verilator --version
Verilator 4.201 devel rev v4.200-38-g2e158d88
```

wsnyder, 2021-03-23:
In the case shown using "2'b10" for the second line and a non-priority case seems much clearer, so I'd argue the warning did help.  However we shouldn't just turn it off for priority case as

      2'b ?1: bar_o = 3'd0;
      2'b ?1: bar_o = 3'd1;

is clearly worthy of warning.

Perhaps you'd be willing to make a pull request to suppress when in a priority case and the values are not complete overlaps?

rswarbrick, 2021-03-23:
That's a good idea. I'll add it to my queue, and should hopefully be able to look at it in the next few days.

PR #2864 (2021-03-28): Allow overlaps in priority case statements
This will still warn if a case item is completely covered by previous
items, but will no longer complain about overlaps like this:

```systemverilog
priority casez (foo_i)
  2'b ?1: bar_o = 3'd0;
  2'b 1?: bar_o = 3'd1;
  default: bar_o = 3'd2;
endcase
```

Before, there was a warning for the second statement because the first
two patterns match 2'b11.

Fixes #2850.
closed 2021-03-28 by wsnyder