verilator/verilator #2851: SUMMARY: Passed 2325  Failed 17  Skipped 160  Unsup 30  Time 180:09 when make test about [verilator@4.200]

Tom-python0121, 2021-03-24:
See:
```console
[root@centos8 test_regress]#make test
……
==SUMMARY: Passed 2325  Failed 17  Skipped 160  Unsup 30  Time 180:09
==SUMMARY: Passed 2325  Failed 17  Skipped 160  Unsup 30  Time 180:09

======================================================================
        #dist/t_a4_examples: %Error: Exec of make -C ../examples/cmake_hello_c failed:
                make && test_regress/t/t_a4_examples.pl  --dist
        #dist/t_dist_manifest: %Error: Files mismatch with manifest: removed_la_files.txt
                make && test_regress/t/t_dist_manifest.pl  --dist
        #vlt/t_savable_open_bad2: %Error: Exec of obj_vlt/t_savable_open_bad2/Vt_savable_open_bad2 failed: %Error: /tmp/root/src/test_regress/t/t_savable_open_bad2.cpp:31: GOT = 1   EXP = 0
                make && test_regress/t/t_savable_open_bad2.pl  --vlt
        #vlt/t_time_vpi_100s10ms: %Error: Line 6 miscompares; obj_vlt/t_time_vpi_100s10ms/vlt_sim.log != t/t_time_vpi_100s10ms.out
                make && test_regress/t/t_time_vpi_100s10ms.pl  --vlt
        #vlt/t_time_vpi_10ms10ns: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_10ms10ns/vlt_sim.log != t/t_time_vpi_10ms10ns.out
                make && test_regress/t/t_time_vpi_10ms10ns.pl  --vlt
        #vlt/t_time_vpi_1fs1fs: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_1fs1fs/vlt_sim.log != t/t_time_vpi_1fs1fs.out
                make && test_regress/t/t_time_vpi_1fs1fs.pl  --vlt
        #vlt/t_time_vpi_1ms10ns: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_1ms10ns/vlt_sim.log != t/t_time_vpi_1ms10ns.out
                make && test_regress/t/t_time_vpi_1ms10ns.pl  --vlt
        #vlt/t_time_vpi_1ns1ns: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_1ns1ns/vlt_sim.log != t/t_time_vpi_1ns1ns.out
                make && test_regress/t/t_time_vpi_1ns1ns.pl  --vlt
        #vlt/t_time_vpi_1ps1fs: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_1ps1fs/vlt_sim.log != t/t_time_vpi_1ps1fs.out
                make && test_regress/t/t_time_vpi_1ps1fs.pl  --vlt
        #vlt/t_time_vpi_1us1ns: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_1us1ns/vlt_sim.log != t/t_time_vpi_1us1ns.out
                make && test_regress/t/t_time_vpi_1us1ns.pl  --vlt
        #vlt/t_wrapper_legacy: %Error: Exec of obj_vlt/t_wrapper_legacy/Vt_wrapper_legacy failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy.pl  --vlt
        #vlt/t_wrapper_legacy_time64: %Error: Exec of obj_vlt/t_wrapper_legacy_time64/Vt_wrapper_legacy_time64 failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_time64.pl  --vlt
        #vlt/t_wrapper_legacy_timed: %Error: Exec of obj_vlt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t0,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_timed.pl  --vlt
        #vltmt/t_gantt: %Error: Exec of obj_vltmt/t_gantt/Vt_gantt failed: %Warning: vltmt/t_gantt: Exec failed with core dump
                make && test_regress/t/t_gantt.pl  --vltmt
        #vltmt/t_wrapper_legacy: %Error: Exec of obj_vltmt/t_wrapper_legacy/Vt_wrapper_legacy failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy.pl  --vltmt
        #vltmt/t_wrapper_legacy_time64: %Error: Exec of obj_vltmt/t_wrapper_legacy_time64/Vt_wrapper_legacy_time64 failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
                make && test_regress/t/t_wrapper_legacy_time64.pl  --vltmt
        #vltmt/t_wrapper_legacy_timed: %Error: Exec of obj_vltmt/t_wrapper_legacy_timed/Vt_wrapper_legacy_timed failed: -V{t1,1}- Verilated::debug is on. Message prefix indicates {<thread>,<sequence_number>}.
               make && test_regress/t/t_wrapper_legacy_timed.pl  --vltmt
……
```
Can anyone fix the bug that occurs during make test?
### Information on your system
* **Python:** 3.6.8
* **Platform:** linux-centos8-aarch64
* **Concretizer:** original

wsnyder, 2021-03-24:
A bit further down then where you cut the log message it will run the tests with verbosity and that will likely show a compiler error message.  You can also get this from running one test by itself e.g.

        test_regress/t/t_time_vpi_100s10ms.pl  --vlt

Look at the output, see what the GCC error message is, and ideally please attempt to fix it and submit a pull request to fix it, portability issues are usually something simple like a missing include, thanks.

Tom-python0121, 2021-03-25:
@wsnyder ok,
see:
```console
[root@centos8 t]# ./t_a4_examples.pl
======================================================================
dist/t_a4_examples: ==================================================
        make -C ../examples/cmake_hello_c
make: Entering directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/cmake_hello_c'

-- Verilator CMake hello world example

-- VERILATE ----------------
mkdir -p build && cd build && cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/cmake_hello_c/build

-- BUILD -------------------
cmake --build build -j
Unknown argument -j
Usage: cmake --build <dir> [options] [-- [native-options]]
Options:
  <dir>          = Project binary directory to be built.
  --target <tgt> = Build <tgt> instead of default targets.
                   May only be specified once.
  --config <cfg> = For multi-configuration tools, choose <cfg>.
  --clean-first  = Build target 'clean' first, then build.
                   (To clean only, use --target 'clean'.)
  --use-stderr   = Ignored.  Behavior is default in CMake >= 3.0.
  --             = Pass remaining options to the native tool.
make: *** [Makefile:55: run] Error 1
make: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/cmake_hello_c'
Use of uninitialized value $param{"logfile"} in concatenation (.) or string at ./driver.pl line 1630.
%Warning: dist/t_a4_examples: Exec of make -C ../examples/cmake_hello_c failed:

        make -C ../examples/cmake_hello_sc
make: Entering directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/cmake_hello_sc'
CMake Error: Problem processing arguments. Aborting.
……
-- DONE --------------------
To see waveforms, open vlt_dump.vcd in a waveform viewer

make: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/make_tracing_c'
        make -C ../examples/make_tracing_sc
make: Entering directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/make_tracing_sc'

%Skip: SYSTEMC_INCLUDE not in environment
(If you have SystemC see the README, and rebuild Verilator)

make: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/make_tracing_sc'
        make -C ../examples/xml_py
make: Entering directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/xml_py'
-- vl_file_copy example
python3 vl_file_copy -odir copied top.v
NOTE: vl_file_copy is only an example starting point for writing your own tool.
-- vl_hier_graph example
python3 vl_hier_graph -o graph.dot top.v
NOTE: vl_hier_graph is only an example starting point for writing your own tool.
Manually run:  dot -Tpdf -o graph.pdf graph.dot
make: Leaving directory '/tmp/root/spack-stage/spack-stage-verilator-4.200-sydtudpvkgw5qd2xykkimbecmv3pe3nt/spack-src/examples/xml_py'
dist/t_a4_examples: %Error: Exec of make -C ../examples/cmake_hello_c failed:
dist/t_a4_examples: FAILED: Exec of make -C ../examples/cmake_hello_c failed:
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:02

[root@centos8 t]# ./t_dist_manifest.pl
======================================================================
dist/t_dist_manifest: ==================================================
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
%Warning: dist/t_dist_manifest: Files mismatch with manifest: removed_la_files.txt
[... 55 lines cut ...]
%Warning: vlt/t_time_vpi_100s10ms: Line 6 miscompares; obj_vlt/t_time_vpi_100s10ms/vlt_sim.log != t/t_time_vpi_100s10ms.out
F1: [18446744073709.551615ns] time%0d=10000  123%0t=12300000000000.000000ns
F2: [0.000000ns] time%0d=10000  123%0t=12300000000000.000000ns
To update reference: HARNESS_UPDATE_GOLDEN=1 {command} or --golden
vlt/t_time_vpi_100s10ms: %Error: Line 6 miscompares; obj_vlt/t_time_vpi_100s10ms/vlt_sim.log != t/t_time_vpi_100s10ms.out
vlt/t_time_vpi_100s10ms: FAILED: Line 6 miscompares; obj_vlt/t_time_vpi_100s10ms/vlt_sim.log != t/t_time_vpi_100s10ms.out
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:03

[root@centos8 t]# ./t_time_vpi_10ms10ns.pl
global vpiSimTime = 0,60000000  vpiScaledRealTime = 6e+07
global vpiTimeUnit = -2  vpiTimePrecision = -8
top.t vpiSimTime = 0,60000000  vpiScaledRealTime = 8.12981e+173
top.t vpiTimeUnit = 254  vpiTimePrecision = -8
*-* All Finished *-*
- t/t_time_vpi.v:46: Verilog $finish
%Warning: vlt/t_time_vpi_10ms10ns: Line 13 miscompares; obj_vlt/t_time_vpi_10ms10ns/vlt_sim.log != t/t_time_vpi_10ms10ns.out
F1: top.t vpiSimTime = 0,60000000  vpiScaledRealTime = 8.12981e+173
F2: top.t vpiSimTime = 0,60000000  vpiScaledRealTime = 60
To update reference: HARNESS_UPDATE_GOLDEN=1 {command} or --golden
vlt/t_time_vpi_10ms10ns: %Error: Line 13 miscompares; obj_vlt/t_time_vpi_10ms10ns/vlt_sim.log != t/t_time_vpi_10ms10ns.out
vlt/t_time_vpi_10ms10ns: FAILED: Line 13 miscompares; obj_vlt/t_time_vpi_10ms10ns/vlt_sim.log != t/t_time_vpi_10ms10ns.out
==SUMMARY: Passed 0  Failed 1  Unsup 0  Time 0:02
……
```
There are too many problems in these 17 big test cases. I will not test them all. There are many problems, including cmake problems and source code test problems. I hope you can test these problems and solve them in the next release.

wsnyder, 2021-03-25:
Sorry you are having trouble.  Given:
* CentOS 8 is end of life this year
* That "cmake -j" doesn't work means you aren't on a recent version of CentOS 8
* You're using Arm
* CentOS 8.2 passes these tests on x86

I'd suggest using a more recent OS. But if you stick with this config, you're probably the only person with this configuration, and furthermore I won't be able to recreate it, therefore it's unlikely these will ever get fixed unless you debug it yourself.  If you are able to figure out the problem we'll of course welcome any fixes that result.

closed 2021-03-25 by wsnyder