verilator/verilator #2852: what's the possible reason for the hang at time 0 with cosim mode?

zhuzhzh, 2021-03-24:
I compile the design sip into library with protect-lib mode. use VCS to compile TB.
It can run but the result is wrong.
So I compile the design with --trace and defined one function sc_time_stamp to return sim time with VPI.

but at runtime, vcs hangs at time 0. with -loop_report, we found the recursive call is in generated library.

```
sip_protectlib_comb_update() ->  Vsip::eval_step() -> _Vsip::_eval_initial_loop -> Vsip::_eval_settle -> Vsip::_combo__TOP__3 -> Vsip::__Vdpiimwrap_sip__DOT_sip_protectlib_combo_pdate_TOP -> sip_protectlib_combo_update -> Vsip::eval_step() -> Vsip::_eval_initial_loop ->...
```

Do you have any suspect on this issue? as DIDNOTCONVERGE error is not reported, I don't have the clue to check.

zhuzhzh, 2021-03-24:
[image]

wsnyder, 2021-03-24:
Looks like the inner isn't reporting DIDNOTCONVERGE as the inner Verilated eval() is completing and returning to the upper wrapper.  The upper wrapper probably then sends an output that activates the outer simulator, which then changes an input and so results in another eval call.  So from Verilator's perspective you keep calling eval just with different inputs, which is strange but legal. To debug I think there's a VCS trace option that shows all zero-time signal changes.  Alternatively perhaps put a printf on all the signals entering and leaving the wrapper's calls into Verilator to see what signal keeps changing

wsnyder, 2021-04-13:
I'm assuming you got through this, otherwise feel free to ask followups.

closed 2021-04-13 by wsnyder