verilator/verilator #2853: Increase signal width of --coverage-toggle

xuejiazidi, 2021-03-25:
Hi,

I want do coverage analysis on a register file which contains 32 32-bit registers.  However, according to Verilator manual, signals that are very wide (>256 bits total storage across all dimensions) are not covered.  

How can I modify Verilator, i.e. change width limit in source code of Verilator, to cover very wide signals?

Looking forward to your reply,

Bing

wsnyder, 2021-03-25:
Surprisingly this is hardcoded, it's in src/V3Coverage.cpp

        if ((nodep->width() * nodep->dtypep()->arrayUnpackedElements()) > 256) {
            return "Wide bus/array > 256 bits";
        }

You can of course change this directly, but perhaps you'd be willing to submit a pull request to add a new "--coverage-max-width" option similar to --trace-max-width"?

xuejiazidi, 2021-03-26:
Sorry it does not work.  I tried to change 256 to bigger value (1024 or 2048) and comment the if sentences.  But Verilator cannot cover the register file.

This is my register file:
logic [31:1][31:0] rf_reg_q;

wsnyder, 2021-03-26:
I just tried this and it worked for me, so please try again.

wsnyder, 2021-03-26:
BTW normally one wouldn't use toggle coverage for something like this, instead e.g. would do a user coverage point on the data bits going in, on every address being written, and every address being read.

xuejiazidi, 2021-03-29:
I figured out why.  After modifying the cpp file I need to rebuild Verilator.  Then it works!  In addition, your suggestion is really useful.  Thank you!

wsnyder, 2021-03-29:
Since that worked for you I added --coverage-max-width, you can use that now.

closed 2021-03-29 by wsnyder