verilator/verilator #2855: Confusing behaviour for int enums in structs

rswarbrick, 2021-03-26:
Consider the following code:
```systemverilog
package test;

  typedef enum int unsigned {
    Mode0 = 0,
    Mode1 = 1
  } mode_e;

  typedef struct packed {
    mode_e a;
    bit    b;
  } foo_t;

  parameter foo_t Foo0 = '{a: Mode0, b: 1'b0};
  parameter foo_t Foo1 = '{a: 0,     b: 1'b0};

endpackage
```

Verilator complains about the definition of `Foo0`:
```
> /opt/verilator-test/bin/verilator --lint-only test.sv
%Warning-WIDTHCONCAT: test.sv:13:38: Unsized numbers/parameters not allowed in concatenations.
   13 |   parameter foo_t Foo0 = '{a: Mode0, b: 1'b0};
      |                                      ^
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

It seems that this is because it's treating `Mode0` as an unsized integer. However, it's fine with an explicit int (as passed to `Foo1`). It's also fine if I change the struct to contain an `int` instead of the enum type.

Changing `mode_t` to be `enum bit [1:0]` is a work-around that silences the warning, but I don't think this behaviour is quite consistent. Is there some "turn an enum value into an int" step missing somewhere?

rswarbrick, 2021-03-26:
Oops, I forgot to say: it also seems wrong that the warning points at `b` rather than `a`: in the actual example where I saw this, pointing at the following field made this rather confusing to track down.

wsnyder, 2021-03-26:
Yes, thanks for the good report.  This is a broken side effect of how Verilator handles packed structs, will give it a look when get a chance.

wsnyder, 2024-01-25:
Got time to look at fixing this. On further examination the warning is believed correct and useful. The intent of IEEE in not allowing integer conversion is to 'force' users to use the symbolic enum name (e.g. "MODE10" instead of a value ('0"). Therefore the warning is correct.

Committed test so we can track this decision.

rswarbrick, 2024-01-25:
It definitely looks like this has been improved since the version we had when I reported this(!). I just tried with version 5.020 and the output for the test in the report looks like:
```
%Error-ENUMVALUE: testme.sv:14:31: Implicit conversion to enum 'MEMBERDTYPE 'a'' from 'logic[31:0]' (IEEE 1800-2017 6.19.3)
                                 : ... Suggest use enum's mnemonic, or static cast
   14 |   parameter foo_t Foo1 = '{a: 0,     b: 1'b0};
      |                               ^
                  ... For error description see https://verilator.org/warn/ENUMVALUE?v=5.020
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Much better!
- The error points at the value we're storing to `a` (rather than the next field)
- It no longer talks about unsized values in concatenations.

Thank you very much for the fix.