verilator/verilator #2856: Verilator 4.2.0 Test Fails After Build

mshaw1990, 2021-03-26:
After building Verilator 4.2.0 using Cygwin 3.1.7 and running `make test` I get the below error, any help would be greatly appreciated.

%Error: Cannot find file containing module: \+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc
        ... Looked in:
             t/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc
             t/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.v
             t/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.sv
             ../include/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc
             ../include/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.v
             ../include/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.sv
             \+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc
             \+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.v
             \+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.sv
             obj_vlt/t_a1_first_cc/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc
             obj_vlt/t_a1_first_cc/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.v
             obj_vlt/t_a1_first_cc/\+define\+TEST_OBJ_DIR\=obj_vlt/t_a1_first_cc.sv
%Error: Exiting due to 1 error(s)
[Thread 7412.0x2ac4 exited with code 1]
[Thread 7412.0x24ac exited with code 1]
[Thread 7412.0x235c exited with code 1]
[Thread 7412.0x2bfc exited with code 1]
[Thread 7412.0x2fe8 exited with code 1]
[Inferior 1 (process 7412) exited with code 01]
No stack.

wsnyder, 2021-03-27:
Debugged down to

    gdb bin/verilator_bin_dbg  -ex "run arg1  \+arg2\=o arg3"

gives different results (unrelated to Verilator itself), the backslash before + gets stripped on Linux under gdb 9.2 and bash 5.0, but does not on Cygwin gdb 9.2 and bash 4.4.12.  Will be fixing this by not escaping the plus and equal to appease tests, but will affect other cases using --gdbbt.

wsnyder, 2021-03-27:
The pushed changes make examples pass for me.  If you do a full "make test" there are other issues that need cleanup, but believe these are all test/test harness related and not problems with the built verilator.

closed 2021-03-27 by wsnyder

mshaw1990, 2021-03-27:
Thank you for looking into this!