verilator/verilator #2857: t_prof failure

toddstrader, 2021-03-26:
I just pulled HEAD (4e41c1350167a5b5c35911125f4755acbaf273fe) and I'm getting a failure on `t_prof`:
```
*-* All Finished *-*
- t/t_prof.v:47: Verilog $finish
        cd obj_vlt/t_prof && gprof Vt_prof gmon.out.30097 > gprof.out
        cd obj_vlt/t_prof && /usr/scratch/local/devs/verilator-hrt/bin/verilator_profcfunc gprof.out > cfuncs.out
%Warning: vlt/t_prof: File_grep: obj_vlt/t_prof/cfuncs.out: Regexp not found: (?^:VLib + VL_POWSS_QQQ)

vlt/t_prof: %Error: File_grep: obj_vlt/t_prof/cfuncs.out: Regexp not found: (?^:VLib + VL_POWSS_QQQ)
vlt/t_prof: FAILED: File_grep: obj_vlt/t_prof/cfuncs.out: Regexp not found: (?^:VLib + VL_POWSS_QQQ)
```
I'm not familiar with how gprof works, so I'm not sure what should be expected here.  I'm running:
```
$ gprof --version
GNU gprof (GNU Binutils for Debian) 2.32
```
and `gprof.out` seems to be mainly just boilerplate:
```
Flat profile:

Each sample counts as 0.01 seconds.
 no time accumulated

  %   cumulative   self              self     total           
 time   seconds   seconds    calls  Ts/call  Ts/call  name    

 %         the percentage of the total running time of the
time       program used by this function.

cumulative a running sum of the number of seconds accounted
 seconds   for by this function and those listed above it.

 self      the number of seconds accounted for by this
seconds    function alone.  This is the major sort for this
           listing.

calls      the number of times this function was invoked, if
           this function is profiled, else blank.

 self      the average number of milliseconds spent in this
ms/call    function per call, if this function is profiled,
	   else blank.

 total     the average number of milliseconds spent in this
ms/call    function and its descendents per call, if this
	   function is profiled, else blank.

name       the name of the function.  This is the minor sort
           for this listing. The index shows the location of
	   the function in the gprof listing. If the index is
	   in parenthesis it shows where it would appear in
	   the gprof listing if it were to be printed.

Copyright (C) 2012-2019 Free Software Foundation, Inc.

Copying and distribution of this file, with or without modification,
are permitted in any medium without royalty provided the copyright
notice and this notice are preserved.

		     Call graph (explanation follows)

granularity: each sample hit covers 2 byte(s) no time propagated

index % time    self  children    called     name

 This table describes the call tree of the program, and was sorted by
 the total amount of time spent in each function and its children.

 Each entry in this table consists of several lines.  The line with the
 index number at the left hand margin lists the current function.
 The lines above it list the functions that called this function,
 and the lines below it list the functions this one called.
 This line lists:
     index	A unique number given to each element of the table.
		Index numbers are sorted numerically.
		The index number is printed next to every function name so
		it is easier to look up where the function is in the table.

     % time	This is the percentage of the `total' time that was spent
		in this function and its children.  Note that due to
		different viewpoints, functions excluded by options, etc,
		these numbers will NOT add up to 100%.

     self	This is the total amount of time spent in this function.

     children	This is the total amount of time propagated into this
		function by its children.

     called	This is the number of times the function was called.
		If the function called itself recursively, the number
		only includes non-recursive calls, and is followed by
		a `+' and the number of recursive calls.

     name	The name of the current function.  The index number is
		printed after it.  If the function is a member of a
		cycle, the cycle number is printed between the
		function's name and the index number.

 For the function's parents, the fields have the following meanings:

     self	This is the amount of time that was propagated directly
		from the function into this parent.

     children	This is the amount of time that was propagated from
		the function's children into this parent.

     called	This is the number of times this parent called the
		function `/' the total number of times the function
		was called.  Recursive calls to the function are not
		included in the number after the `/'.

     name	This is the name of the parent.  The parent's index
		number is printed after it.  If the parent is a
		member of a cycle, the cycle number is printed between
		the name and the index number.

 If the parents of the function cannot be determined, the word
 `<spontaneous>' is printed in the `name' field, and all the other
 fields are blank.

 For the function's children, the fields have the following meanings:

     self	This is the amount of time that was propagated directly
		from the child into the function.

     children	This is the amount of time that was propagated from the
		child's children to the function.

     called	This is the number of times the function called
		this child `/' the total number of times the child
		was called.  Recursive calls by the child are not
		listed in the number after the `/'.

     name	This is the name of the child.  The child's index
		number is printed after it.  If the child is a
		member of a cycle, the cycle number is printed
		between the name and the index number.

 If there are any cycles (circles) in the call graph, there is an
 entry for the cycle-as-a-whole.  This entry shows who called the
 cycle (as parents) and the members of the cycle (as children.)
 The `+' recursive calls entry shows the number of function calls that
 were internal to the cycle, and the calls entry for each member shows,
 for that member, how many times it was called from other members of
 the cycle.

Copyright (C) 2012-2019 Free Software Foundation, Inc.

Copying and distribution of this file, with or without modification,
are permitted in any medium without royalty provided the copyright
notice and this notice are preserved.

Index by function name

```
I see this is passing in CI, so I'm guessing there's some environmental issue here.

wsnyder, 2021-03-26:
Please see if VL_POWSS_QQQ (and what if anything) appears in test_regress/obj_vlt/t_prof/verilator_profcfunc gprof.out
Maybe there's an error or something?

wsnyder, 2021-03-26:
Oh, sorry, missed that's the file you attached.  Maybe that means no trace data came out?  If you run it by hand does it work?

toddstrader, 2021-03-26:
The output of the Verilator script in cfuncs.out is:
```
Overall summary by type:
  % time  type
  100.00  Unaccounted for/rounding error

Overall summary by design:
  % time  design
  100.00  Unaccounted for/rounding error

Overall summary by module:
  % time  module
  100.00  Unaccounted for/rounding error

Verilog code profile:
   These are split into three categories:
      C++:     Time in non-Verilated C++ code
      Prof:    Time in profile overhead
      VBlock:  Time attributable to a block in a Verilog file and line
      VCommon: Time in a Verilated module, due to all parts of the design
      VLib:    Time in Verilated common libraries, called by the Verilated code

  %   cumulative   self              
 time   seconds   seconds    calls   type      filename and line number
```
Vtprof does appear to be producing prof data (different run / PID):
```
$ file gmon.out.25007 
gmon.out.25007: GNU prof performance data - version 1
```
Running `gprof` by hand produces the same results:
```
$ diff -s gprof.out <(gprof Vt_prof gmon.out.25007)
Files gprof.out and /dev/fd/63 are identical
```

wsnyder, 2021-03-26:
Does clang work?  (make distclean ; ./configure CXX=clang++)

toddstrader, 2021-04-05:
> Does clang work? (make distclean ; ./configure CXX=clang++)

This works for me.  I tested with clang 11.0.1.

```
$ grep VL_POWSS_QQQ obj_vlt/t_prof/cfuncs.out
  0.00      0.00     0.00   100000   VLib      VL_POWSS_QQQ(int, int, int, unsigned long, unsigned long, bool, bool)
```

toddstrader, 2021-04-07:
I also tried different versions of gcc.  I was running 6.3 originally, but the problem also happens with 9.2.

wsnyder, 2021-06-19:
@toddstrader did you get this straightened out?  Can you indicate if there anything we still need to do to close it out, or close it?  Thanks.

toddstrader, 2021-06-21:
We can close this.  This one isn't at the top of my list right now.  I do still see the issue, but am running the test via clang and CI to check for regressions.

closed 2021-06-21 by toddstrader