verilator/verilator #2863: NOLATCH warning issued on latches on partial arrays

tsailer, 2021-03-28:
The NOLATCH warning is issued in Verilator 4.108 2021-01-10 rev v4.106-158-g484b76e5b on partial array latches. See test case below.

[t_nolatch.sv](https://github.com/verilator/verilator/files/6217129/t_nolatch.sv.txt)

wsnyder, 2021-03-28:
```
module test
  (input  logic [1:0] a,
   input  logic       e,
   output logic [1:0] z);

   always_latch
     if (e)
       z[0] <= a[0];

   always_latch
     if (e)
       z[1] <= a[1];

endmodule
```

```
%Warning-NOLATCH: /svaha/wsnyder/backpack/frwork/t_nolatch.sv:6:4: No latches detected in always_latch block
    6 |    always_latch
      |    ^~~~~~~~~~~~
%Warning-NOLATCH: /svaha/wsnyder/backpack/frwork/t_nolatch.sv:10:4: No latches detected in always_latch block
   10 |    always_latch
      |    ^~~~~~~~~~~~

```

wsnyder, 2021-03-28:
Agreed, as with #2862 perhaps you could submit a pull request and/or @margej may be interested.

margej, 2021-06-01:
Fix for #2938 merged today, I've confirmed this issue is fixed by the same. I'll raise a PR to add the above test case

closed 2021-06-13 by wsnyder