verilator/verilator #2865: Inconsistent use of enums when checking bit extraction widths

rswarbrick, 2021-03-29:
Consider this example code. Verilator allows the definition of `has_foo2` (presumably because the 32-bit integer literal with value 2 fits in 2 bits) but doesn't allow `has_foo1`, despite the fact that `Foo1` equals 1.

```systemverilog
module enum_idx (input logic [3:0] enable,
                 output logic      has_foo1,
                 output logic      has_foo2);

  typedef enum bit [2:0] {
    Foo0 = 3'd0,
    Foo1 = 3'd1,
    Foo2 = 3'd2,
    Foo3 = 3'd3,
    Bar  = 3'd4
  } opn_t;

  localparam FooOps = Foo3;

  bit [FooOps:0] enabled_foos;
  assign enabled_foos = enable;
  
  assign has_foo1 = enabled_foos[Foo1];
  assign has_foo2 = enabled_foos[2];

endmodule
```

This is a little painful if you want to define some general "operation type" enum but then re-use the literal indices in some module that only supports some operations. Here, the example is supposed to be that the module only supports "Foo" operations.

Is the way Verilator checks widths for enum indices an intentional design decision (I can sort of see an argument for it), or should we allow literal values that fit?

At the moment, we give the following warning:
```
$ /opt/verilator-test/bin/verilator --lint-only enum_idx.sv
%Warning-WIDTH: enum_idx.sv:18:33: Bit extraction of var[3:0] requires 2 bit index, not 3 bits.
                                 : ... In instance enum_idx
   18 |   assign has_foo1 = enabled_foos[Foo1];
      |                                 ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
$ /opt/verilator-test/bin/verilator --version
Verilator 4.201 devel rev v4.200-60-g6d3ec160
```

wsnyder, 2021-03-29:
This is intentional.  It avoids a common bug where you have the wrong size select into an array.  As "1" is unsized it lets this pass, but the enum is sized so it assumes it's supposed to be the right size for the array.  You can easily see how someone could resize the array, but then not realize they had an enum to also resize. 

I don't see any straightforward way to suppress the WIDTH warning in the case you have without loosing the desirable feature mentioned above. I would suggest either suppressing WIDTH, or otherwise.   (If Verilog had enum inheritance that might be the cleanest solution, since it seems what you want is the width of Foo to be 2 bits in one enumeration, then another enumeration with all Foo's plus Bar.)

closed 2021-03-29 by wsnyder

rswarbrick, 2021-03-30:
Thanks for the explanation. Is there a list of features like this somewhere? It might be worth recording them to avoid "bug" reports like this in future?

wsnyder, 2021-03-30:
Good point, I have some doc fixes in a branch I'm working on, I've added a placeholder there to document this better.