verilator/verilator #2866: Verilator generates errors for unreachable generate/if/endgenerate code

udif, 2021-03-30:
Hi, in contiuation with #2625 , #2754 and #2823, I created this MCVE showing the type of problems I'm facing:

```
module a;
localparam A=1;
generate
if (A==0)
begin
b b_inst(.x(1'b0));
end
endgenerate
endmodule

module b;
endmodule
```
It fails with
```
%Error: a.v:6:11: Pin not found: 'x'
    6 | b b_inst(.x(1'b0));
      |           ^
%Error: Exiting due to 1 error(s)
```
Even though the condition in the if/generate block is false.

I'm running it against my branch which is effectively https://github.com/verilator/verilator/commit/2e158d88c106825d43eb1717799e02385bd5caea plus my [synthesis_off](https://github.com/udif/verilator/tree/synthesis_off) branch (adds support for translate_on and translate_off comments).
The code above runs fine on one of the Big3 simulators, which we currently use at work.

Both the error above and my private branch exists due to the fact that I'm trying to get a large legacy codebase to run under Verilator . I have little control over what I can change in this codebase, esp. when most of the errors are in 3rd party IP.

(You wrote in #24 that you are not interested in a translate_on/translate_off PR. Unfortunately that requires me to maintain that branch and merge the master branch to it once in a while, if I ever want to get our legacy code to compile).

wsnyder, 2021-03-31:
The fundamental problem is how elaboration works which isn't going to change soon, however it's probably reasonable to change this to a named error e.g. "PINNOTFOUND" then suppress that error either manually or automatically.  Might you be willing to attempt a pull request to do that?

As to translate on/off, you could write your own pipe filter to implement them if you really want to.

wsnyder, 2021-03-31:
Are there any other errors that similarly result?

udif, 2021-03-31:
I will keep searching and update this issue if I find more problems due to this.

I already have other issues in my codebase, but they are unrelated to if/generate (graceful handling of encrypted models for lint-only mode). As I noticed several styles of encrypted models, I will open an issue on this only when I have a clearer picture (I only expect Verilator to accept a black-box from these files so that lint runs will not fail).

As for the elaboration fix, I assume you mean #1540 .
What about a simpler workaround:
Instead of solving the root problem which is hard and would result in not elaborating "dead" generate sections, would it be possible to collect al errors/warnings that appear within an if/generate section and save them for later processing instead of printing them immediately, and go over this list again after the final stages, when all instance parameter values are known, and then print only the true errors?
The only drawback I can see is that the error printing order will change from previous version because some messages will be printed at a much later stage.

As for your pull request suggestion - are you suggesting I turn this into a named error so it can be turned into a warning, and ignored? I'm not too familiar yet with the Verilator codebase. My only changes so far are in V3PreProc.cpp which is simple. I can try though.

wsnyder, 2021-03-31:
Right, add PINNOTFOUND error to V3Error.h, change where the error is printed to use PINNOTFOUND, update bin/verilator to document it, then add a test case that has a "// verilator lint_off PINNOTFOUND" in the appropriate spot.

With that you can make progress, and we'll think about what the next step is to make this easier.