verilator/verilator #2867: verilator with TestBench.sv that needs TestBench_types.sv

pbreuer, 2021-03-31:
How may we help - what is your question?
I am trying to use verilator on the .sv (or .v) output from Clash, the high level HDL. Clash has successfully generated  netlists and .sv files in Test/TestBench/ but when I run verilator .. -ITest/TestBench TestBench.sv it tells me it needs the types to be "predeclared" (what on earth does that mean!). OK, so I guess it wants the TestBench_type.sv module to be named on the command line first (please use explanatory and whole sentences in a manual, containing no jargon!) before TestBench.sv, so I do that and it generates files in obj_dir, but then hangs forever, seemingly:

% verilator -ITest/TestBench -Wno-fatal -Wno-lint --debug  --cc TestBench_types.sv  TestBench.sv

..
- V3LinkLevel.cpp:45: modSortByLevel()
- V3Ast.cpp:1121:     Dumping obj_dir/VTestBench_types_002_cells.tree
- V3LinkDot.h:37:     linkDotPrimary: 
- V3LinkJump.cpp:304: linkJump: 
- V3Ast.cpp:1121:     Dumping obj_dir/VTestBench_types_007_link.tree
- V3LinkInc.cpp:251:  linkIncrements: 
- V3Ast.cpp:1121:     Dumping obj_dir/VTestBench_types_008_linkInc.tree
- V3Param.cpp:855:    param: 
..[hums away forever without further output of any kind, no syscalls in strace] ...

ls -lrt obj_dir
...
-rw-r--r-- 1 ptb     14442 Mar 31 12:34 VTestBench_types_TestBench_fpu4.vpp
-rw-r--r-- 1 ptb    192510 Mar 31 12:34 VTestBench_types_TestBench_fpu.vpp
-rw-r--r-- 1 ptb     18355 Mar 31 12:34 VTestBench_types_001_linkcells.dot
-rw-r--r-- 1 ptb 433876616 Mar 31 12:34 VTestBench_types_002_cells.tree
-rw-r--r-- 1 ptb 518593847 Mar 31 12:35 VTestBench_types_007_link.tree
-rw-r--r-- 1 ptb        97 Mar 31 12:35 VTestBench_types_008_linkInc.tree

and no more comes out, verilator sits there.

I have tried

verilator -ITest/TestBench -Wno-fatal -Wno-lint --debug -j --l2-name TestBench --bin TestBench --cc TestBench_types.sv --cc TestBench.sv

and the debug output still stops at the point indicated above. IMO it isgenerating TestBench_types stuff, where it should be generating TestBench stuff. HOW do I tell it to use the TestBench.sv file for an entry point, but take the type definitions used there from TestBench_types.sv?

I have tried making an executable with -exe and getting linking to a cc_main.cpp file that I have written working (which references TestBench, not TestBench_types), but it clearly never gets that far. I have tried --Build. I would just appreciate a clearly written explanation of what the flags I am giving even do! Why am I using -I, for example, when those are not (C/C++) "include" files in there but .sv files. What is the "includes" that the manual frustratingly says the flag is for, without defining what it means? Why can't it even find the TestBench.sv file unless I give -I/Test/TestBench ? That's not an "include"! The error message: "%Error: Cannot find file containing module: TestBench.sv" seems to say it is looking for a MODULE named TestBench.sv, whereas the manual says to put SOURCE FILES on the command line (" [source_files.v]"), so should I be specifying the MODULE instead of the SOURCE FILE?
Trying verilator -Wno-fatal -Wno-lint --debug -ITest/TestBench/ --cc TestBench seems to find the MODULE, in the SOURCE FILE, even thouh it says it is looking for INCLUDES, but then it compains about missing all the type definitions, which must be "predeclared"
%Error: Test/TestBench//TestBench.sv:9:14: Package/class 'TestBench_types' not found, and needs to be predeclared (IEEE 1800-2017 26.3)
Well! But if I have guessed what "predeclared" means (put TestBench_types earlier on the command line), it makes stuff called TestBench_types*, which surely lacks for anything to execute and it is no wonder nothing ever happens.

What is the right way to go about all this please? Is verilator just resting forever, or will it eventually produce something?

wsnyder, 2021-03-31:
For predeclared, refer to the IEEE specification section it mentions.  "The compilation of a package shall precede the compilation of scopes in which the package is imported."  One way to predeclare a package is to put it on the command line, this is common across simulators, but we welcome a pull request to improve the documentation.

Verilator is still running if those are the only files, so you .cpp files don't matter.  To debug what's going on, run with --debug --gdb, wait a long time (until hung), hit Ctrl-C then type "bt".  Then resume with "c".  Then hit Ctrl-C and "bt" again.  Do that a few times and it should be obvious where it's stuck.

pbreuer, 2021-03-31:
That only raises the question of what is an "IEEE specification section "? If you have a URL, please give it!  But anyway, one should never refer to anything else in a manual or other document. It must be self contained. How would you like it if your breakfast cereal said "please pre-deslough (see govt edict 4.3.56)"?

So what is "the compilation of a package"? Is what verilator doing "the compilation of a package" ? Come to that, what is a "package"? How do I get verilator to compile whatever it thinks "the package" is before it compiles (if what it does is compile, rather than simulate, generate, analyse, etc) "the scopes" (which perphaps may mean "type definitions", even though "scope" normally means a domain of validity for a definition or declaration). That is no explanation at all!
 
[No, I am not dumb. ]

OK, so verilator is still running but it would indeed be nice if it said so, because how am I supposed to know. It  suggested gdb when I interrupted it the first time, but nothing meaningful seems to come up, certainly nothing to tell me that it is making progress. (I'm so not going to need a guide on how to use gdb). Your reassurance is the only bona fide on offer! I thank you for it, but how do I know? What does verilator think it is doing, and how long does it think it will take?

How many hours should I expect to wait when the intermediate files it has already produced are several 100 MB in size?

-rw-r--r-- 1 ptb 433876616 Mar 31 14:38 VTestBench_types_002_cells.tree
-rw-r--r-- 1 ptb 518593847 Mar 31 14:39 VTestBench_types_007_link.tree
-rw-r--r-- 1 ptb        97 Mar 31 14:39 VTestBench_types_008_linkInc.tree

It didn't seem to take long with those!

AM I DOING RIGHT in naming both TestBench_types.sv and TestBench.sv on the command line, in that order? Is that what "predeclared" means me to do?

I'll assume so, as you are not telling me off about it. 

So I'll let it run a few more hours.

Thanks!

PTB

pbreuer, 2021-03-31:
Hi. Something more has come out now:

...
-rw-r--r-- 1 ptb         97 Mar 31 14:39 VTestBench_types_008_linkInc.tree
-rw-r--r-- 1 ptb 7623816126 Mar 31 16:27 VTestBench_types_010_paramlink.tree

That's after a bit less than 3h wall clock time, 1h cpu time. It's about 30GB into swap at this point. It hasn't yet started on the actual TestBench code yet, has it? Or will everything  come out naned TestBench_types* .foo? I can't imagine what it actually has to generate from mere type definitions in TestBench_types that should cause it to emit near 8GB of a ".tree" so I am afraid that is part of TestBench, not TestBench_types, and the naming is coming out wrong. What is that file for? Why is it there? Is it going to produce more files like this? Why? It should just be translating the system verilog into C++, shouldn't it? What is all this? Is it because of the --debug option, which at least results in reports on what it is doing.

  --dump-tree                 Enable dumping .tree files

I didn't give that option, and what does "enable" mean? Will it or won't it?

--debug                     Enable debugging

Nothing about "tree" files. And what does it mean by "enable"? Or "debugging", come to that? I think it means "emit more reports on what it is doing as it does it" as far as I have seen so far, which would ordinarily be --verbose.  But it would be nice if it told me what it is doing BEFORE it does it, and why. Telling me afterwards and not saying why is not helpful neither to reassure me that it is doing something instead of nothing while it is doing it, nor to remove the mystery of what it is aiming at.

I would have expected a straight translation of verilog wiring into global variables, plus sources of signals to them, sinks for them, and transformers, all managed by a scheduler.

About  package and systemverilog ... I am afraid I have no idea about systemverilog other than that it looks vaguely readable, and I have no interest in it  as it is just an intermediate language to me. The system verilog I am feeding to verilator is automatically generated and I have no part in  it beyond writing the HDL  that gave rise to it via a long sequence of transformations. I would guess that "package" means a collection of modules, and a module probably means a collection or hierarchy of source files, exporting (and importing) certain namespaces. Who knows! 

As to crl-C, etc. I am not brave or mad enough to experiment with a process that is 3h in and 30GB into swap. I know that if started with a --gdb option werilator will attach a gdb and transfer control to it on interrupt, but I can do that anyway, externally. Is the privately attached gdb being talked to by verilator about some DWARF descrption that is not instrinsic to the verilator executable? What do you expect me to see except a printout of the current call stack inside verilator and a tall tale of which subroutine it thinks it is in, none of which would  be  meaningful to me. I don't see any symbol table even on the verilator_bin_dbg executable:

/usr/bin/verilator_bin_dbg: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=c3473461f8e630af7448e9af607a50020eacbcaf, for GNU/Linux 3.2.0, stripped
So I wouldn't expect much info from a debugger. The frame pointer is probably optimized away in the executable, so it won't really know that much about where it is, and if it did, I wouldn't.

PTB

wsnyder, 2021-03-31:
Everything will come out under VTestBench, this is described under --prefix in the manual.

For --debug etc you were only looking a the summary, see later in the manual.

It looks like you're using a version that an OS vendor installed and stripped, you'll need to build from sources to do backtraces.

However I'm wondering if this is just a huge design (e.g. whole chip)?   Or is it something small?   I was assuming it was something very small and should not be taking long, but if it's a few hundred million gates that might be why it's taking so long.

pbreuer, 2021-04-01:
--prefix ! Good idea! But more hours in nothing more has come out yet. WHY is it even producing .tree files and what are they (for)?

It is 40GB into swap now. Yes, this is a whole chip design.

It is examining the TestBench hierarchy, but it seems to be attaching TestBench_types globally as a prefix.

...
-rw-r--r-- 1 ptb      65569 Mar 31 14:38 VTestBench_types_TestBench_alu.vpp
                                                           **********************************
-rw-r--r-- 1 ptb      14442 Mar 31 14:38 VTestBench_types_TestBench_fpu4.vpp
-rw-r--r-- 1 ptb     192510 Mar 31 14:38 VTestBench_types_TestBench_fpu.vpp
-rw-r--r-- 1 ptb      18355 Mar 31 14:38 VTestBench_types_001_linkcells.dot
-rw-r--r-- 1 ptb  433876616 Mar 31 14:38 VTestBench_types_002_cells.tree
-rw-r--r-- 1 ptb  518593847 Mar 31 14:39 VTestBench_types_007_link.tree
-rw-r--r-- 1 ptb         97 Mar 31 14:39 VTestBench_types_008_linkInc.tree
-rw-r--r-- 1 ptb 7623816126 Mar 31 16:27 VTestBench_types_010_paramlink.tree
-rw-r--r-- 1 ptb 7205502143 Mar 31 22:41 VTestBench_types_012_width.tree

Intervals are increasing. It has been 6h since the last output.

--prefix topname
           Specifies the name of the top level class and makefile.  Defaults
           to V prepended to the name of the --top-module switch, or V
           prepended to the first Verilog filename passed on the command line.
                                     ***********************

Yes, but so? What is this FOR? It HAS taken TestBench_types as a prefix to put on the front of the name of every file it emits, but why does that matter, if it does (tidiness?) and what functional effect does it have that I should WANT to control it?

I deduce it is no big deal, or it should have complained about having to guess?

--mod-prefix topname
           Specifies the name to prepend to all lower level classes.  Defaults
           to the same as --prefix.

I infer this is about the C++ translation. What is a "lower level" in terms of classes? Does it mean that C++ classes are generated that are sometimes interior one to another? Or does it mean privacy and visibility? Should I care about this, what difference does it make, and what is this about?

--debug
           Select the debug executable of Verilator (if available), and enable
           more internal assertions (equivalent to "--debug-check"), debugging
           messages (equivalent to "--debugi 4"), and intermediate form dump
           files (equivalent to "--dump-treei 3").

(the manual author really needs to grok that "enable" and DO are different!) So that's why .tree files are coming out! I deduce that I do not want them?? Are they what is holding up the production of the C++ files?

 --no-debug-leak
           In --debug mode, by default Verilator intentionally leaks AstNode
           instances instead of freeing them, so that each node pointer is
           unique in the resulting tree files and dot files.

           This option disables the leak.

[disables -> stops ! ]

That sounds alarming! It could keep a count, instead of the object, if its problem is  reuse of a name with local visibility that has been used in another scope, thus making it globally non-unique. That's hard to imagine! Is this still for real or was it true at some point in development and is no longer so?

I don't know how many gates there are in the generated system verilog description and if there is somewhere special to look for that I would be glad to know of it, with thanks!

Regards and thanks!

PTB

wsnyder, 2021-04-01:
Ok, as your design is large we started down the wrong path, there's no point in --debug.  And, if you did need to debug you'd probably need "--debug --no-debug-leak" and a non-stripped executable.

I would suggest using/learning verilator with a small section of your design first.

It's quite likely it might take a long time to run your whole chip through Verilator, this is normal (though unfortunate).  Typically very large designs are manually split (or use the --hierarchical option) to avoid this.

simdream, 2021-04-01:
to manually split a large design, is there a guide? since some submodules are compiled separately, to call _eval()s of these submodules in the top C++ testbench needs a careful arrangement? or shall the submodule output should be flop flops? Thanks,

pbreuer, 2021-04-01:
Unfortunately there is no --hier* option in the man page of the version I have, nor mention of similar.
% verilator --version
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad
Since it is making a library, it perhaps could make a number of smaller libraries instead, with certain object files within each. That there is a call graph with a hierarchy should make no difference at all, since names would be resolved at link time. The "hierarchical" may refer to an attempt to organise the classes in a hierarchy of name spaces, instead of longer and longer names separated by dots or underlines, though I don't see a theoretical need.

wsnyder, 2021-04-01:
You'll need a newer version, ideally the current stable version, then see the HIERARCHICAL section of the manual. Hierarchical will work much better if it is at flop boundaries.

pbreuer, 2021-04-01:
Thanks - I think the version I launched yesterday is on the verge of outputting some C++, since I have seen a message about a .cpp file. No output yet, however. Maybe it is making one very large one! ?
Because the running version is deep in swap and only at 5% cpu usage, and I have plenty of swap and plenty of cpu, I have launched another in parallel with --debugi 3 instead of --debug, and with --no-debug-leak , and with --prefix TestBench so that stuff does not come out with TestBench_types as a prefix on everything. Unfortunately
(1) it now comes out with TestBench_ as an EXTRA prefix, so one gets TestBench_TestBench_*
(2) it still seems to be dumping .tree files, though that is not part of the --debugi semantics.

-rw-r--r-- 1 ptb      14442 Apr  1 17:54 TestBench_TestBench_fpu4.vpp
-rw-r--r-- 1 ptb     192510 Apr  1 17:54 TestBench_TestBench_fpu.vpp
-rw-r--r-- 1 ptb      18348 Apr  1 17:54 TestBench_001_linkcells.dot
-rw-r--r-- 1 ptb  433876616 Apr  1 17:54 TestBench_002_cells.tree
-rw-r--r-- 1 ptb  518593847 Apr  1 17:55 TestBench_007_link.tree
-rw-r--r-- 1 ptb         97 Apr  1 17:55 TestBench_008_linkInc.tree

But I also had those after a few seconds yesterday too (different prefix):

-rw-r--r-- 1 ptb     192510 Mar 31 14:38 VTestBench_types_TestBench_fpu.vpp
-rw-r--r-- 1 ptb      18355 Mar 31 14:38 VTestBench_types_001_linkcells.dot
-rw-r--r-- 1 ptb  433876616 Mar 31 14:38 VTestBench_types_002_cells.tree
-rw-r--r-- 1 ptb  518593847 Mar 31 14:39 VTestBench_types_007_link.tree
-rw-r--r-- 1 ptb         97 Mar 31 14:39 VTestBench_types_008_linkInc.tree
-rw-r--r-- 1 ptb 7623816126 Mar 31 16:27 VTestBench_types_010_paramlink.tree

So do I expect a "paramlink" tree in 2h time? Why is it producing this stuff and how can I stop it? I used:

% verilator -ITest/TestBench -Wno-fatal -Wno-lint --debugi 3 --no-debug-leak --prefix TestBench -j --l2-name TestBench --MMD -sv --threads 4 --cc TestBench_types.sv --cc TestBench.sv &

Thanks for any light that may be shed on this.

PTB

wsnyder, 2021-04-01:
I wouldn't use debug at all, but otherwise use

--debugi 3 --no-dump-tree --no-debug-leak

I would again encourage working on a small section of your design and get it outputting etc as you want, then go back to the larger design.

pbreuer, 2021-04-01:
Yes, I've gone for precisely  those options already. That still leaves the naming problem, so I am trying just something minimal that stops TestBench_types being used.

% verilator -ITest/TestBench -Wno-fatal -Wno-lint --debugi 3 --no-dump-tree --no-debug-leak --prefix VTB -j --MMD -sv --threads 4 --cc TestBench_types.sv --cc TestBench.sv &

But names now come out as, for example:

-rw-r--r-- 1 ptb      77572 Apr  1 18:21 VTB_TestBench_exec_mix_msg.vpp
-rw-r--r-- 1 ptb      94114 Apr  1 18:21 VTB_TestBench_exec_mix.vpp
-rw-r--r-- 1 ptb      14364 Apr  1 18:21 VTB_TestBench_alu4.vpp
-rw-r--r-- 1 ptb      65569 Apr  1 18:21 VTB_TestBench_alu.vpp
-rw-r--r-- 1 ptb      14442 Apr  1 18:21 VTB_TestBench_fpu4.vpp
-rw-r--r-- 1 ptb     192510 Apr  1 18:21 VTB_TestBench_fpu.vpp

and what I want is just VTestBench_*, and let the TestBench_types module take care of itself, maybe come out as VTestBench_types_* ideally. It is not important, just untidy.

As to working on a small section of the design, nice suggestion, but HOW? I have indirect control over the system verilog code, but resynthesizing it will take 10h, so I do not want to play with it. It is effectively fixed and unalterable and anything I change must be done in/by verilator.

Thanks

PTB

pbreuer, 2021-04-02:
I have debianized the github version of verilator and installed it, with a view to using --hierarchical, but the manual entry seems to say that is not enough?

    ... pass --hierarchical option to verilator command.  There are two
       ways to mark a module:

         a) Write /* verilator hier_block */ metacomment in HDL code
             See L</"LANGUAGE EXTENSIONS"> for more detail.

I'm afraid I don't know a thing about that code, certainly not enough to find anything in it, or locate a place in it where I should add that, and I wouldn't in any case because the code is generated, not written, so whatever I did would vanish next time round.

I need something at the verilator end of things.

         b) add hier_block line in the configuration file.
            See C<hier_block> in L</"CONFIGURATION FILES"> for example.

I am launching verilator from the command line.

So let's grep for what they say to look at:

CONFIGURATION FILES
       In addition to the command line, warnings and other features may be
       controlled by configuration files, typically named with the .vlt
       extension. An example:

Unusually, given the misuse of "enable" to mean "do" elsewhere, here "enable" really does mean enable and not do, because --hierarchical indeed does not DO anything, it merely potentiates a happening. Cute!

So does this configuration file have a .vlt suffix or not? The description is rather vague! Am I to call it ".vlt" or maybe ".verilator.vlt" or "foo.vlt"? WHAT NAME do I give and what option do I give to verilator to have it read it?

hier_block -module "<modulename>"
           Specifies that the module is a unit of hierarchical Verilation.
           Note that the setting is ignored unless --hierachical option is
           specified.  See "HIERARCHICAL VERILATION" for more information.

A grep finds 520 modules. That's fine by me. I don't care. Is there an option like

    hier_block -module ALL

?

I haven't seen any cpp files come out yet, so I am not convinced the .vlt file I made is being read. Is there some special name, or way of making it be read?

Any clues on that?

Thanks

PTB

pbreuer, 2021-04-02:
I am sure the foo.vlt file is being read now, since I placed it on the command line and got errors reported in it at first (which I corrected). Apparently that is what the manual means by it being read "in the normal way"!  But no cpp file has come out yet, 35m of cpu time and several hours later (it is 16.9GB into swap, and running about 5% of a CPU). The last debug message was:
- V3LinkDot.h:42:     linkDotParamed: 
- V3Dead.cpp:465:     deadifyModules: 
and since then nothing (those messages should have times on them!).
Is it cycling round a loop tracing dependencies? That should be a simple static analysis.

What about this option:

       --output-split statements
           Enables splitting the output .cpp files into multiple outputs.
           When a C++ file exceeds the specified number of operations, a new
           file will be created at the next function boundary.  In addition,
           if the total output code size exceeds the specified value,
           VM_PARALLEL_BUILDS will be set to 1 by default in the generated
           make files, making parallel compilation possible. Using
           --output-split should have only a trivial impact on model
           performance. But can greatly improve C++ compilation speed. The use
           of ccache (set for you if present at configure time) is also more
           effective with this option.

           This option is on by default with a value of 20000. To disable,
           pass with a value of 0.

Would that force some output? It is extremely frustrating to have no progress indication.

PTB

wsnyder, 2021-04-02:
You give a configuration file a name ending in ".vlt" e.g. "anything.vlt", but what what makes it a configuration file is the `verilator_config directive. Anyhow looks like you have that figured out.

>Is there an option like hier_block -module ALL

No, you need to understand where you are making your partitions, this is not automatic and if you do it on "everything" it will almost certainly break something.

>But no cpp file has come out yet, The last debug message was: deadifyModules:

You're not even 10% of the way through.

>      --output-split statements

As it says this option defaults on.  You aren't to the .cpp generation process yet.

Please run an experiment on some small example to see the messages for the whole process.  There are examples in github, or any distribution should have installed them somewhere.

While I appreciate your trying Verilator out, it is still at the stage you will need to put together many pieces yourself.  You might consider reducing (your apparent) frustration by getting commercial support, or using a more traditional closed-source simulator with a paid field engineering support organization.

pbreuer, 2021-04-02:
I appreciate and agree with your comments, but I am taking what I think is the fastest route. I should be capable of making this work, by force or persuasion.  Yes, I would like to try it out on a small example first, but I don't have one and don't even know enough about system verilog to split one out from what I have, so my best bet is to help verilator to zone in on something inside what I have. I can tell it to jsut do one particular module of the 520! I will try that soon.

Meanwhile I have started some modification of the verilator source to helpout. I have fitted a signal handler so a HUP signal changes the debug level.

 I can see the sequence of things that verilator does ("iterate(node)" I think it was) and should be able to just debug that - as a last resort! I have a non-stripped version of verilator_bin. I don't know if it linked to a non-stripped library,  shrug.

  bin/verilator_bin:              ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=60e8a096c7a8fe9b14f051b9f91928103a5013d6, for GNU/Linux 3.2.0, with debug_info, not stripped

At worst, debianization has produced a debug package for verilator, which at 500MB is certain to contain symbol tables for everything under the sun. Hopefully I can avoid it.

I'll just wait a few hours more. 10% is good info. Thanks.

Regards

PTB

pbreuer, 2021-04-02:
I forgot to ask about "No, you need to understand where you are making your partitions". Why? Why should anyone care, me or verilator?
What's the problem that is to be avoided? Those are modules produced by a system verilog generator doing synthesis, and IT was happy. It has all wires out of one module going in as wires into another, and so on, and all a simulator has to do is run EACH object corresponding to a module, and connect up the inputs and outputs as per the wires. There should be no "levels". All the components (i.e., modules) are top level, or can be brought to the top by renaming it and exposing its wires and the internal wires of a component it is in, if it happens to be in something.

So why should asking it to do that be problematic?

Do you want me to only name modules for "hierarchical" that are really, truly, naturally top-level? Or am I supposed to confine myself to just ONE hierarchy of component modules? What's the low-down?

Regards

PTB

pbreuer, 2021-04-02:
I've tried it on a single module (chosen at random) and it seems to do things OK.

I lacked instruction on HOW (!!), but I guessed this:
verilator -IKPU/FPUa -Wno-fatal -Wno-lint --debugi 4 --no-dump-tree --no-debug-leak --hierarchical --prefix VKPU_FPUa -j --MMD -sv --threads 4 --output-split 60000 --cc FPUa_types --cc FPUa

-rw-r--r-- 1 ptb    16744 Apr  2 18:03 VKPU_FPUa_FPUa_types.vpp
-rw-r--r-- 1 ptb      900 Apr  2 18:03 VKPU_FPUa_FPUa.vpp
-rw-r--r-- 1 ptb   177102 Apr  2 18:03 VKPU_FPUa_FPUa_fpu.vpp
-rw-r--r-- 1 ptb    70315 Apr  2 18:03 VKPU_FPUa_055_ordermv_hazards.dot
-rw-r--r-- 1 ptb    27699 Apr  2 18:03 VKPU_FPUa_057_ordermv_transitive1.dot
-rw-r--r-- 1 ptb    30818 Apr  2 18:03 VKPU_FPUa_056_ordermv_contraction.dot
-rw-r--r-- 1 ptb      938 Apr  2 18:03 VKPU_FPUa__Syms.h
-rw-r--r-- 1 ptb      851 Apr  2 18:03 VKPU_FPUa__Syms.cpp
-rw-r--r-- 1 ptb    11296 Apr  2 18:03 VKPU_FPUa_084_ordermv_transitive2.dot
-rw-r--r-- 1 ptb    50564 Apr  2 18:03 VKPU_FPUa.h
-rw-r--r-- 1 ptb  1134702 Apr  2 18:03 VKPU_FPUa__Slow.cpp
-rw-r--r-- 1 ptb     1244 Apr  2 18:03 VKPU_FPUa_FPUa_types__Slow.cpp
-rw-r--r-- 1 ptb     1542 Apr  2 18:03 VKPU_FPUa_FPUa_types.h
-rw-r--r-- 1 ptb      219 Apr  2 18:03 VKPU_FPUa_FPUa_types.cpp
-rw-r--r-- 1 ptb  1309617 Apr  2 18:03 VKPU_FPUa.cpp
-rw-r--r-- 1 ptb  9520770 Apr  2 18:03 VKPU_FPUa.xml
-rw-r--r-- 1 ptb     1675 Apr  2 18:03 VKPU_FPUa_classes.mk
-rw-r--r-- 1 ptb    63835 Apr  2 18:03 VKPU_FPUa__stats.txt
-rw-r--r-- 1 ptb     1442 Apr  2 18:03 VKPU_FPUa.mk
-rw-r--r-- 1 ptb     2787 Apr  2 18:03 VKPU_FPUa__verFiles.dat
-rw-r--r-- 1 ptb      735 Apr  2 18:03 VKPU_FPUa__ver.d

I know the .dot files are diagrams of something or other and I can forget them. The Files.dat I think could be a manifest? The classes.mk and hier.mk are clearly make files, but I don't know the difference (what is this "hierarchy" business?). The classes.mk is just defns and is meant to be called by some supdirectory make.
The hier.mk actually has a supdirectory make as a target! What gives there? The .f file contains arguments to be used by a call from the hier.mk, looks like to verilator? The .cpp files are basically just gigantic lists of assignments. 

Essentially, each cpp has an eval_step that calls the huge bunch of assignments. There's also an initial_loop_setup that does some minor fettling and phooey. It looks like the eval_step runs a delta (i.e. infinitesimal) time loop at most 100 times and hopes it stabilises. That could go badly if hooked up to more
that 100 other things, which also hope to stabilise after, say, one delta cycle. That looks a bit hopeful, but never mind.

The question in my mind is CAN I DO THIS FOR EACH MODULE IN TURN AND THEN JOIN THEM UP (i.e., compile .o files, and then link).

The huge bunch of assignments have the form 

    vlTOPp->FPUa__DOT__FPUa_fpu___05Fclash_internal__DOT__c__024vec2_1[0U]    = ..

which looks hopeful because every module I compiled would reference some vlTop (pointer) in this way, so all I would have to do is make sure there is some globally visible/linkable vlTop object created somewhere, somehow, and it has one of the things in this class as a subobject. So I need to pay attention to a constructor for it at most.

That part would work.

Is that what I should be doing?

Regards

PTB

pbreuer, 2021-04-02:
I am getting warnings of a circularity in a signal semantics, even though this has been synthesized already at the HDL level, and the system verilog is merely a rendering of the wiring diagram AND I am perfectly well able to run the model in the HDL at the signal level.

Moreover, the warnings are leading to a SIG 9 and verilator death.

%Warning-UNOPTFLAT: KPU/ReadA/ReadA_read_sync.sv:217:36: Signal unoptimizable: Feedback to clock or circular logic: 'ReadA.ReadA_read_stage_result.ReadA_read_sync_result_6.c$ws3_case_alt_1'
  217 |   ReadA_types::array_of_64_Tup2_34 c$ws3_case_alt_1;
      |                                    ^~~~~~~~~~~~~~~~
                    KPU/ReadA/ReadA_read_stage.sv:500:1: ... note: In file included from ReadA_read_stage.sv
                    KPU/ReadA/ReadA.sv:38:1: ... note: In file included from ReadA.sv
                    KPU/ReadA/ReadA_read_sync.sv:217:36:      Example path: ReadA.ReadA_read_stage_result.ReadA_read_sync_result_6.c$ws3_case_alt_1
                    KPU/ReadA/ReadA_read_sync.sv:1253:16:      Example path: ALWAYS

 %Error: Verilator threw signal 9. Suggest trying --debug --gdbbt
 %Error: Command Failed /usr/bin/verilator_bin -IKPU/ReadA -Wno-fatal -Wno-lint --no-dump-tree --no-debug-leak --prefix VKPU_ReadA -j --MMD -sv --output-split 1000000 --cc ReadA_types --cc ReadA

No files are produced.

The warnings are from

  - V3Order.cpp:1956:     Process Circulars...

The final failure is:

  - V3Expand.cpp:910:   expandAll: 
 %Error: export VERILATOR_ROOT=
 %Error: setarch --addr-no-randomize /usr/bin/verilator_bin_dbg -IKPU/ReadA -Wno-fatal -Wno-lint --debug --no-dump-tree --no-debug-leak --prefix VKPU_ReadA -j --MMD -sv --output-split 1000000 --cc ReadA_types --cc ReadA
 %Error: Command Failed setarch --addr-no-randomize /usr/bin/verilator_bin_dbg -IKPU/ReadA -Wno-fatal -Wno-lint --debug --no-dump-tree --no-debug-leak --prefix VKPU_ReadA -j --MMD -sv --output-split 1000000 --cc ReadA_types --cc ReadA

Umm ... it might have been trying to fall back to gdb.

If this is real, I should be able to make it go away by inserting an extra delay in the loop, IF I can figure out what/where it is. But should I really have to? It may be that an input signal it cannot know anything about is guaranteed to always gate this circuit so that it never loops, as verilator is only getting a small part of the whole picture by examining this small subunit in isolation. In reality it is connected to peer units that are watching it while it is watching them. Static analysis in general should err on the side of overcaution - so has it done that here? 

Does this kind of thing happen often, and what should one do, if anything? I frankly don't believe it.

Thanks for insights and thoughts/guesses

PTB

pbreuer, 2021-04-03:
I'm pretty certain the above error/warning messages fromverilator are mistaken. I have added a register (i.e. time delay component) to the output of one of the subunits (readA_read_sync) named in all the paths given in the messages, so it is impossible to have a looped calculation through it, and the messages persist.

1) verilator reports a problem at this stage of its process:
  - V3Unroll.cpp:511:   unrollAll: 
  - V3Unroll.cpp:64:       Can't Unroll: Unable to simulate loop :WHILE 0x55555b6b1d00 <e11024#> {c674af}

2) then looped calculation warnings here:

 -Info-Loop:   VERTEX=0x55555d7e7350\n   TOP->ReadA__DOT__ReadA_read_stage_result__DOT__ReadA_read_sync_result_8__DOT__c__024ws_app_arg_1 r151 f210 c520
                    KPU/ReadA/ReadA_read_sync.sv:37:43:      Example path:   ReadA.ReadA_read_stage_result.ReadA_read_sync_result_8.c$ws_app_arg_1

3) then verilator hangs forever at this late stage:

  - V3Subst.cpp:382:    substituteAll: 
  - V3Const.cpp:3133:   constifyCpp: 

4) attaching a gdb shows it's nowhere special, just traversing the graph:

#0  0x0000555555799bea in AstNode::checkTreeIter (
    this=this@entry=0x5555bb7422e0, backp=0x5555bb741810) at ../V3Ast.cpp:992
#1  0x0000555555799abe in AstNode::checkTreeIterList (this=0x555566dddab0,     backp=<optimized out>, backp@entry=0x555556583070) at ../V3Ast.cpp:1010
#2  0x0000555555799c3b in AstNode::checkTreeIter (
    this=this@entry=0x555556583070, backp=<optimized out>) at ../V3Ast.cpp:1000 
...
#8  0x000055555579a0ba in AstNode::checkTree (this=0x555555facf80)
    at ../V3Ast.h:1521
#9  AstNode::checkTree (this=0x555555facf80) at ../V3Ast.cpp:1020
#10 AstNode::dumpTreeFile (this=this@entry=0x555555facf80, filename=..., 
    append=append@entry=false, doDump=doDump@entry=false, 
    doCheck=doCheck@entry=true) at ../V3Ast.cpp:1153
#11 0x000055555593b9a1 in V3Global::dumpCheckGlobalTree (stagename=..., 
    newNumber=newNumber@entry=0, doDump=false) at ../V3Global.cpp:96
#12 0x000055555580e334 in V3Const::constifyCpp (nodep=0x7fffffffe1b0)
    at ../V3Const.cpp:3138
#13 0x000055555577149d in process () at ../V3Global.h:128
#14 verilate (argString=...) at ../Verilator.cpp:587
#15 main (argc=<optimized out>, argv=<optimized out>, env=0x7fffffffe2e0)
    at ../Verilator.cpp:732

5) It has had about 21m of cpu tme, over 4h of wall clock time, and has a 42GB memory footprint. 
It is working on a unit that is not very large, so that is impossible. But it does have 4 subunits that communicate to each other in a way that looks superficially like a loop, but is not. One of the subunits (readA_read_sync) is synchronizing the parallel and asynchronous outputs from the others, and that is what I placed an extra delay on the output from. 

6) The HDL in which this is written is certain that it synthesizes to netlists and I can execute a simulation fine in that HDL at signal level. It may be that its fixpoint calculation is inherently lazier than verilator, so verilator finds a different fixpoint that is stricter than it should be and contains a looping-forever calculation where in fact a simple value is forced. That would be a fault in verilator's interpretation of circuit semantics. It could happen if, for example, an if then else were interpreted as requiring both inputs to be resolved in all circumstances, instead of only when required by the gating signal.

That would be weird - it must be more subtle than that and I am experimenting to see if I can pin it down. It is not an artifact of the version of verilator.

wsnyder, 2021-04-03:
>I am getting warnings of a circularity in a signal semantics, even though this has been synthesized already at the HDL level

Please see the documentation on UNOPTFLAT, this isn't related to synthesis but rather the speed of the results. But you can turn off that warning.  I think that addresses many of your other questions.

>Moreover, the warnings are leading to a SIG 9 and verilator death.

If you run under --debug/--debugi it does an abort() after errors to get into gdb.  So this is normal.  There's no flag to disable this behavior at present.  I'll add some notes to the documentation on this.

>- V3Unroll.cpp:64:       Can't Unroll: Unable to simulate loop :WHILE 0x55555b6b1d00 <e11024#> {c674af}

This isn't a problem it's an informational statement for debug, ignore it.

pbreuer, 2021-04-03:
"documentation on UNOPTFLAT" - I can guess! I do actually have all the warnings I can find explicitly turned off (no-lint?), but debugi seems to be partially resupllying them.

"isn't related to synthesis" - I do believe it is a real fault. It  apparently leads to exponential blowup in the calculations, resulting in 40GB of RAM/swap footprint and no output or visible progress for hours.

It has to be circling round a cyclic directed graph forever. The component  is not that complex or big.
.
It is stuck at the 

- V3Const.cpp:3133:   constifyCpp: 

stage (for hours and hours). I have 8GB of RAM and 4 cores.

I do think the circularity warnings are spurious. 

I have looked at one lot, and I see only a potential circularity,  not certainty.

In my opinion it is nonexistent semantically.  Here it is:

    315: assign c$vec2_0 = (ws[1 : $high(ws)]);
                ********    **           **
    324: assign zipWith_in2_0 = c$vec2_0[n_7];
                ************    *******
    326: assign zipWith_out_0 = {zipWith_in1_0,zipWith_in2_0};
                *************                  *************
    329: assign c$ws_app_arg_0[n_7] = zipWith_out_0;
                *************         *************
    341: assign imap_in_0 = c$ws_app_arg_0[n_8];
                *********   **************
    372: assign c$case_alt_5 = c$case_alt_7 ? imap_in_0.Tup2_49_sel1 : {1'b1,i_1};  
                ************                  *********
    368: assign c$case_alt_4 = c$case_alt_4_selection[137:137] ? c$case_alt_5 : {1'b1,i_1};
                ***********                                      ************
    364: assign imap_out_0 = (i1 >= n_3) ? {1'b1,i_1} : c$case_alt_4;
                **********                              ************
    393: assign c$ws_app_arg_1[n_8] = imap_out_0;
                *************         **********
    483: assign ws = ...array_of_9_logic_vector_5_to_lv(c$ws_app_arg_1)}...
                **                                      **************

Lines 372 and 368 can break the loop, and will when their guard conditions are not BOTH true at the same time. 

Since the names indicate they are two different branchess of a switch/case statement back in the HDL, only at most ONE of them would ever be true at one time, and there would be no loop.

I'll look to see if that is derivable from the rest of the system verilog code.

Can some kind of hint be embedded in the (generated) system verilog code that asserts the exclusivity of those two conditions, do you know?

I have examined the original HDL that gave rise to the system verilog, and I see no circularity in it either.

The component in question has 5 subcomponents, one on the input splitting it to provide inputs for three more subcomponents  and there is a one-cycle delay on the outputs of those three, and on the orginal input. Then  a 5th subcomponent   collates, buffers and synchronizes those four signals so they are eventuallly emitted four at a time, all together, at intervals. I see no cycle. If there is one, it would be in nonfunctional stuff like reset signals ... but it's invisible to me and my experiments.

It is very clear that there is no output. I have tried dumping .tree files to get .dot files that might show a circularity (which one should I look at) but I haven't even managed to visualize any of those so far.

-rw-r--r-- 1 ptb    1304 Apr  3 17:19 systemverilog/obj_dir/VKPU_ReadA_001_linkcells.dot
-rw-r--r-- 1 ptb   17869 Apr  3 17:19 systemverilog/obj_dir/VKPU_ReadA_038_task_call.dot
-rw-r--r-- 1 ptb 6428001 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_051_gate_simp.dot
-rw-r--r-- 1 ptb 6407366 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_052_gate_opt.dot
-rw-r--r-- 1 ptb 4820604 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_059_orderg_pre.dot
-rw-r--r-- 1 ptb    2832 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_060_acyc_simp.dot
-rw-r--r-- 1 ptb 4849563 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_061_orderg_acyc.dot
-rw-r--r-- 1 ptb 5025851 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_062_orderg_order.dot
-rw-r--r-- 1 ptb 5025852 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_063_orderg_domain.dot
-rw-r--r-- 1 ptb 5025850 Apr  3 17:20 systemverilog/obj_dir/VKPU_ReadA_065_orderg_done.dot

It s a real concern that the process is cycling forever. I see no reason why it should be having problems if it has a correct sematic understanding of the circuit, and I think that instead it is taking a syntactically guided tour chasing its tail in a chase that never ends. How else can one get a memory footprint that slowly grows to 40GB over hours? This should be a simple circuit to translate. It is unidirectional data flow, with no feedback loops except those inside the subcomponents. If it helps, I will "decohere" those and move their state transform loop up to merge with that of the full component. Then there will be only one top-level loop for verilator to look at. Shall I do that? I'll try it.

Any insight gratefully received.

Regards and thanks

PTB

pbreuer, 2021-04-03:
I've expanded everything in the system verilog code out into a single loop at top level (mostly ... there are still two small external components that I was too impatient to drag in).

I think verlator has now got past  V3Const (evaluation of constant expressions) for the first time and is now sitting in

- V3Subst.cpp:382:    substituteAll: 

with nothing happening. I'm not going to attach a gdb because I am sure it just taken up with traversing a graph, bigly.

PTB

pbreuer, 2021-04-03:
By the way, can you be a little more specific about what is meant by "documentation on UNOPTFLAT"? I looked at every mention in the manual, and it says nothing to me. I presume it  means verilator didn't manage to reduce to a normal form and remove all internal structure just at this point. It should be down to a flat set of parallel left_var = right_exp[input_vars,left_vars] assignments by now. Shrug. It certainly didn't say whether one should be worried about it, or care, or _why_ , which is what a reader needs to know. 

Really the text's got the standard faults expected from an engineer attempting to communicate:  too much irrelevant detail as well as too little and the wrong sort!

Anyway, never mind that, I meant to write that I removed all internal structure from the system verilog and relaunched the translation. It is clear that it gets as far as

- V3Expand.cpp:910:   expandAll: 
- V3Const.cpp:3133:   constifyCpp: 
- V3Subst.cpp:382:    substituteAll: 
(here)

with about 15GB of swap till about half way through that last, then starts increasing swap usage by 100MB per cpu second expended, till reaching about 30GB of swap used. That's where I am. Plainly it has doubled down on some huge graph structure.

You do know about doing graph reduction in place, rather than making a copy? One  rotates the pointers as one goes down a node, and  on coming back up one marks it as done, then goes to the next "child", which will be the true parent after two passes through that node and the node one leaves behind at that point on exit will be again arranged as it was when first entered, modulo changes made. No copy required.

[Jon Fairburn once wrote me a haiku in which he described this, when I frustratedly asked him how, for a higher order functional language implementation of graph reduction. ]

Also ... are you doing garbage collecting on discarded nodes  as you discard each, or are you waiting till the end of their natural duration as provided automatically by C++ scopes? It should really be ongoing because  C++ GC is a syntactically based conservative estimate and the author should always know better than it - or think they do! One reduces a reference count on each node as it is discarded  via an explicit call of the destructor (rather than waiting for control to exit the scope of the object), and one does GC at regular intervals, recycling those nodes to the free node list on which the count has dropped to zero. 

This is a quite small component, completely flat now, and I am not yet getting verilator to produce C++ from it.

 Would you like the system verilog code made available, and see if you can get verilator to process it by applying a standard technique? "no, it didn't work for me either" as an answer would also be good information.

PTB

pbreuer, 2021-04-04:
It is FINISHING! Giving maximally flat system verilog code without any internal structure of its own as input did the trick. The stats are: 14h wall clock computation, 1h cpu time, 8GB RAM + 32GB swap. It has been working through the process() routine in Verilator.cpp,  giving out messaging on about 100 subprocesses/steps with --debugi 3, of which it is now in the last (emitmk).

-rw-r--r-- 1 ptb      58819 Apr  3 22:22 VKPU_ReadA_ReadA_types.vpp
-rw-r--r-- 1 ptb    1580740 Apr  3 22:22 VKPU_ReadA_ReadA.vpp
-rw-r--r-- 1 ptb        865 Apr  4 10:15 VKPU_ReadA__Syms.cpp
-rw-r--r-- 1 ptb        948 Apr  4 10:15 VKPU_ReadA__Syms.h
-rw-r--r-- 1 ptb     214472 Apr  4 10:16 VKPU_ReadA.h
-rw-r--r-- 1 ptb  315877822 Apr  4 10:27 VKPU_ReadA__Slow.cpp
-rw-r--r-- 1 ptb     262364 Apr  4 10:27 VKPU_ReadA__1__Slow.cpp
-rw-r--r-- 1 ptb   70993674 Apr  4 10:32 VKPU_ReadA.cpp
-rw-r--r-- 1 ptb  249136154 Apr  4 10:37 VKPU_ReadA__1.cpp
-rw-r--r-- 1 ptb     276311 Apr  4 10:37 VKPU_ReadA__2.cpp
-rw-r--r-- 1 ptb       2703 Apr  4 10:37 VKPU_ReadA_ReadA_types.h
-rw-r--r-- 1 ptb       2831 Apr  4 10:37 VKPU_ReadA_ReadA_types__Slow.cpp
-rw-r--r-- 1 ptb        223 Apr  4 10:37 VKPU_ReadA_ReadA_types.cpp
-rw-r--r-- 1 ptb 3453871612 Apr  4 11:21 VKPU_ReadA.xml
-rw-r--r-- 1 ptb      72381 Apr  4 11:49 VKPU_ReadA__stats.txt
-rw-r--r-- 1 ptb       1719 Apr  4 11:49 VKPU_ReadA_classes.mk
-rw-r--r-- 1 ptb       1447 Apr  4 11:49 VKPU_ReadA.mk

%/usr/bin/verilator_bin -IKPU/ReadA -Wno-fatal -Wno-lint --debugi3 --no-dump-tree --no-debug-leak --prefix VKPU_ReadA -j --MMD -sv --output-split 1000000 --cc ReadA_types --cc ReadA

The structure of the C++ code ... is that eval_step in ReadA.cpp calls _eval in ReadA__2.cpp which combines _combo__TOP__3 of ReadA, then _combo__TOP__6 of ReadA__1.cpp . I'd better do this again and get it in one file.

It is very encouraging that I now have output for one component, but it is clearly not going to be practical for generating a simulator for the whole structure if I do things this way, since this component is about 1/50th part of that. Perhaps the --hierarchical option would avoid the practical need to flatten the incomng system verilog code out first?  I can do that, but feel bad about that and fear it would increase the computational complexity of the task for verilator more than it would help. 

I do need and would appreciate clear explanation of how to use --hierarchical, even though I can experiment  now that I have a "range estimate" for input code, options and time for which I can expect to get code out. It should not take a week.

Regards

PTB

wsnyder, 2021-04-04:
BTW if you add --no-debug-check and use the optimized executable it'll probably be 2x faster.

wsnyder, 2021-04-04:
As to unoptflat, you read this? https://www.veripool.org/projects/verilator/wiki/Manual-verilator#UNOPTFLAT

pbreuer, 2021-04-04:
"if you add --no-debug-check and use the optimized executable it'll probably be 2x faster" - can you give reasoning behind that, please?

I've already noticed that the makefile generated uses -Os, which the docs seem to think means optimize for speed, whereas it actually means optimize for _size_ (of the object code), which is not going to play well with a monolithic source code a few million lines long. The optmizer will spend all its time trying to _reverse_ the process of substitution in the graph that got verilator to the end, and if the result is faster or slower is a toss-up. I couldn't say. -Os might reduce the cache footprint required, which would be good. Or it might unalign the code and structures via packing, which would be bad. Etc. It's not something I would put in a Makefile that likes a fast result. -O2 should be faster than -Os because -Os is -O2 without the optimizations that (may) increase code size. What went wrong with -Ofast, if anything? (it might). Or -O3?

So I would not be inclined to follow an instruction that did not also say why.

"As to unoptflat, you read this?" - since everything I have read so far appears to be stoically uncommunicative (standard for an engineer's prose - it will be concerned with what and how, not why and because; and I can say that with a phd in ee from good 'ol cantab mesen), I am afraid I would be leary of following a link unless the recommendation came with something like "you will find a great exposition on what a user should understand on p3, and though it is a bit rough in parts, the great light for me shone half way through this passage ..", accompanied by persuasive and enticing quotes from the magnificent volume by the enthusiastic recommender.

Ahem.

I have not got --debug turned on, so --no-debug-check should be needed. Only -debug turns it on if not gven explicitly. (Actually, those flag names break no-surprise SE conventions ...  --debug conventionally means turn on messaging about guff, which is ---debugi here; I would call the --debug flag  --dodebug just to make it look weird and not be used by accident). I see it would have inserted assertions in the code.

[That oughtn't to make a difference since I can vamoosh the assert macro when compiling.]

I'm not sure what to do next. I'll probably see if something about --hierarchical helps in some way. If I reduce the complexity of the code that gets put through this process, it should be faster (I am not concerned about the speed of the resulting object code - it will be fast enough whatever happens, and I can probably massage it as needed), and I can do that by splitting it up into various blackbox units that are connected together ... if I can get that to work.

pbreuer, 2021-04-04:
Out of interest, I have had a look at the indicated wiki. Think I got the right anchor point, but not sure as the page wouldn't stand still for me. I think this was meant:

*UNOPTFLAT*  _Warns that due to some construct, optimization of the specified signal is disabled. The signal reported includes a complete scope to the signal; it may be only one particular usage of a multiply instantiated block. The construct should be cleaned up to improve simulation performance; two times better performance may be possible by fixing these warnings._

That is complete gibberish to me. What do you mean by "optimization", and what do you mean by "signal"?
I suspect you are guilty of confusing code with the concept it is intended to model. Are you optimizing code, or are you optimizing the model? What is the meaning of optimization in either case? Are you trying to make the code you produce run faster (on what platform), are you trying to make it smaller, are you trying to make it parallelizable? If you are talking about a model and concepts related to it, please describe the model and what the sense of optimization means for it. If the model is a bunch of wires and gates, are you trying to produce fewer gates, fewer wires, shorter wires, shallower depth of logic, etc?

As to the prose itself, you need to ask yourself what each word in a sentence does. The start "due to some construct" is deadly to human consciousness. It should not be there, because it says nothing. It's function is to start the sentence off with a question in the reader's mind "what the heck is this about" instead of a certainty, which is precisely what a sentence should not do. A sentence must take a reader on a journey from one point to another, and that cannot happen if the start is indefinite.

It continues with "specified signal" when no signal has been specified, so is not grounded in the reader's perspective. I have not the slightest idea what "a complete scope" is, nor even what an incomplete one is, nor what a "scope" is. I suspect that your usage is the opposite way round from the standard meaning of "scope", which is (I am quoting the dictionary here):

   scope  
     <programming> The scope of an identifier is the region of a
     program source within which it represents a certain thing.
     This usually extends from the place where it is declared to
     the end of the smallest enclosing block (begin/end or
     procedure/function body).

I think that instead you are using "scope" to mean something like "target", but I do not know, nor know in what sense anything could be a target, nor what the sense of aiming would be here that could make even that usage sensible. As to "scope to the signal", I have not the slightest idea how what I think is meant to be a target for something could be "to" a signal, or anything. The sense of motion (or is it belonging?) is the wrong way round. I do have a vague idea as to what you mean by signal (a discrete, timewise stream of values), but what the heck are you talking about concepts of the real world for when discussing code, if you are?

"The signal reported" refers to nothing, because nothing has yet been specified that could be refered back to by the definite article "the". If the intention is to define it here, then you need to tell the reader what you are doing and not try to abbreviate two functions of a part of speech into one: say instead something like "the identifier or assignment to which this annotation is affixed generally maps to a real world signal. That  ...". That is how you (a) provide a concrete anchor point, properly referenced with an indefinite article at the point of definition, and (b) having defined it, proceed to shorten further references to it with the definite article.

In short, every  single possible rule of normal, good, technical writing is broken one zillion times over in every single sentence there, and in the manual. Start by getting the use of "the" right, and much will be cured.

Continuing.

"_Unlike the UNOPT warning, this occurs after flattening the netlist, and indicates a more basic problem, as the less obvious case described under UNOPT does not apply._"

Ditto, in spades. Why start with something irrelevant? Are you trying to send the reader's mental journey with you on a detour to never-land?

_"Often UNOPTFLAT is caused by logic that isn't truly circular as viewed by synthesis which analyzes interconnection per-bit, but is circular to simulation which analyzes per-bus:"_

Truly, staggeringly, amazingly impossible to understand. I am the author of umpteen first-class articles on hardware languages, computer programming languages, specification languages, programming and hardware semantics, compilation, etc., and I can figure out not a single sentence of the above. I have just reviewed for the IEEE on their flagship publication, and I understood every line of the article I refereed.

Don't take it badly. People can't write unless they read, and I don't think the text you have pointed me to has been read by anyone. Not the author - but authors rarely can read their own work, because their head knows what the eyes should see, even when it is not there. That's why one needs editors, or a grandmother, who will question every word.

Have a go at rewriting one sentence so it passes muster by someone who does not start by knowing what it means. You'll be surprised by how much you have to expand it. Maybe ten times. Then carry on ...

(my tuppence-worth of unsought advice!)

pbreuer, 2021-04-04:
Actualy g++ failed on compilation. Nobody's fault, probably:

% make -f VKPU_ReadA.mk
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VKPU_ReadA.o VKPU_ReadA.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VKPU_ReadA__1.o VKPU_ReadA__1.cpp
g++: internal compiler error: Segmentation fault signal terminated program cc1plus

I dont think it was out of memory. Usage was a few hundred MB or a GB or two while I was watching, but I didn't watch all the time. Swap was good for 24GB at least. There is no record that the compilation process was reaped by the kernel, so that's a no, then. The second source file is only three times the size of the first, and the first compiled OK. I will recompile with -O1, which I trust more, but -O2/-Os should have been OK.

Something may have gone over an internal limit in g++, I think. I might translate to C first, then compile. I haven't really seen what the purpose of writing in C++ is here in any case, except that nodes in a graph are good concepts to treat as objects, and that might be helpful to the programmer.

It has not been helpful to this reader of the code, however. I've been finding it more obfuscatory than not, though the code is neatly written and well partitioned on a syntactic level, and I can guess what most things do by looking at the names. What I would like answers to are at a global level:

WHAT IS YOUR PROCESS TRYING TO DO, EXACTLY?

How, I can see for myself once I know what the objective is. And Idon't mean an objective like "run this on a computer", which would be the "too vague" engineer's answer! I mean something like "put the code into this particular normal form", accompanied by a careful description of the normal form.

It is my impression that the normal form aimed at is a single loop in which a bunch of "state" variables are changed via logical operations on their values at the beginning of the loop plus certain "input" variables, to their values at the end of the loop.

I need to know what form the state and input variables can take. Are they booleans, or vectors of booleans, arrays of booleans, or  (fixed point) floats, or (fixed length) integer, etc?

I need to know what the primitive arithmetic operations on them are. Is it boolean logic operators, vectorized versions of same, or generic black-box functions defined in some library?

I need to know if your code is intended to do the state update sequentially, one assignment after another, such that any one final value is unknown until all assignments are completed and each assignement builds on earlier assignments. Or is it intended that all the assignments are parallel, able to be executed in any order, because none of them refer to the value produced by any other? Or some sort of intermediate arrangement?

After that, which describes a "flat" code structure, I would like to know what you mean by a :hierarchical" code structure is, and what the normal form including that is, if it exists.

A full explanation for the world at large would include examples of everything, but I don't much care for examples because I guess good. Still, the above is what I consider minimal  to make anything of what is being done. Can you set me straight on it? No pointers, just explanation.

Regards

PTB

wsnyder, 2021-04-04:
I am getting lost in your maze of general questions, comments, and opinions.

-Os has been benchmarked to generally be faster and so is intentionally the default.  I well understand what the different compiler -O flags do.  Your model may vary, as it says clearly "Generally the answer to which optimization level gives the best user experience depends on the use case and some experimentation can pay dividends."

Obviously g++ should not segfault.  I am unaware of any outstanding issues that cause the compiler to crash on Verilated code, but there certainly have been many in the past. While many of the compilers with those issues are still in use, all of the issues have workarounds made by Verilator itself.  If you track this down please file an issue with the gcc compiler team.  Also try it with clang++ and file a bug on them too if it also crashes.  Finally file an issue here so we can attempt to change the generated code to work around it and to help any others with similar problems.

As to what's going on inside, there's some information in docs/internals.rst, but general users are not expected to not have to look inside the generated code nor to have to (nor want to) understand how generated code works.  We do expect that users understand at a high level the IEEE 1800 standard.

If you want to pay in time, e.g. make a pull request to improve the documentation, or pay in money to e.g. hire a technical writer, that would be greatly appreciated. Otherwise I would appreciate no further "unsought advice" on the writing, or documentation style.  I am in agreement it can be better, but this is known, and find your discussion style non-constructive. Writing time is limited and must be balanced against other tasks in the real world.  Thanks for understanding.

pbreuer, 2021-04-05:
I'm afraid -Os is not "faster", and has not been "benchmarked" to be faster, for one thing because benchmarking is meaningless as a concept as it can be overturned by the slightest changes in source code, compiler semantics and hardware and software platform and infrastructure differences, and secondly because the difference between -Os and -O2 is that -O2 has _more_ "optimizations" than -Os. I quote the gcc manual on this:

    -Os Optimize for size.  -Os enables all -O2 optimizations *except* those
           that often increase code size:

           -falign-functions  -falign-jumps -falign-labels  -falign-loops
           -fprefetch-loop-arrays  -freorder-blocks-algorithm=stc

           It also enables -finline-functions, causes the compiler to tune for
           code size rather than execution speed, and performs further
           optimizations designed to reduce code size.

Making that statement puts many things into question.

I've just run g++ again, this time with -O1, and got a segfault again, after 2h. I'll try -O0, but -O1 may be required for inlining of functions and constant expressions, which sometimes can be needed for correct semantics in some weird macro constructions that I have forgotten ... (sorry for memory lapse).

It's not a big thing.

I'l split the code up if -O0 does not work, no matter that it's obscurantist to do so.

I am afraid I am not interested at all in system verilog code. I haven't written it and have no direct means of influencing it. I generated it automatically.  What I am interested in is the GENERATED C++ code for simulation, and I continue to get no information on that.

I need to know all about it because that is what I get to interact with as a programmer and I may need to change it myself. I am not going to sit around for another 14h to regenerate it. 

I also need to know because I need to know how to get it to appear faster, instead of hanging my server cluster for hours.

Why is it so big? There ought to be one state variable for every boolean in my hardware design, and that would put it at about 4000 bits wide. The underlying complexity of your expressions ought to be limited by the number of operations in my design,  EXCEPT that your substitutions may have led to what is called "complexity explosion", for example where  x & y  | x & ~y  has been generated instead of just "x".

That is what it looks to me, because Verilator needs 40GB to run.

As to why users should be interested in the code they are using,  I personally would not use any piece of code whose working I did not thoroughly understand in every aspect, but apart from that you may well have got the semantics wrong in any number of ways, not merely via complexity explosion, as it looks to be. It can be a delicate thing and needs a solid semantic foundation.

As to helping you see what is wrong with every single sentence in your text, such that it cannot be read by the people who would really, really like to use it, even an experienced reader and EE and computer science professional like me, the usual thing to say is "thank you, I get some of that, I didn't realise that writing was just like coding and the author needs to take account that every word chosen has an effect and a responsibility in changing the state of a reader's mind, just like a piece of code does". Writing is hard. A hundred edits of every sentence is not too many because a single wrong word can lose the reader just like that.

PTB

I have not got an explanation  even of if your generated code is meant to consist of assign statements executed sequentially or in parallel in a big top level loop. You have a switch for the number of threads wanted, so you must know that you may emit assignments that either can be executed in parallel or cannot be!

I really would appreciate a simple answer to the simple, uncomplicated question of what the form of the code you produce is. There should be a normal form that you are aiming at. 

Regards
PTB

I'll ask again what  the normal form is of the code that you are aiming to produce. 

If you know, you say, end of. Don't know, no say. It's not a hard thing. No mystery should be involved!

pbreuer, 2021-04-05:
I specified --output-split 10000000 (10,000,000) but it still split:

-rw-r--r-- 1 ptb  320129631 Apr  5 02:17 VKPU_ReadA.cpp
-rw-r--r-- 1 ptb     276311 Apr  5 02:17 VKPU_ReadA__1.cpp

There are 4,597,455 lines in the bigger file. 4,211 in the smaller. 

It's disappointing, as I wanted a less obfuscatory code to use as a starting point. Can one safely turn off some or most of the functions in the process() subroutine? [explaining the form of your output would help decide]

Regards

wsnyder, 2021-04-05:
As to assignments it's not a simple question.  That's like asking what instruction a C "=" assignment compiles into.  It might optimize to nothing, or it might be inlined, or any number of other possibilities.  But if a Verilog-process's assignment hypothetically remains relatively unchanged through to the output, then, as you aren't using --threads AFAIK, then yes the simple answer is they will be put into some series order, as they obviously have to be given there's only a single thread of execution. 

If using multithreaded then verilog-processes within a mtask are in series, and across mtasks, depending on how mtasks are assigned to threads, might be in parallel.  For what a mtask is see the internals documentation.

Use --output-split 0 to turn off splitting, as it says in the manual you don't like ;)

Reading between the lines it seems like the Verilog input code is in some unusual form. Can you describe it?   While in theory any valid input code should work, of course it is more likely to be bugs if the input code looks different from any style of code it has seen before.  (i.e. greatly differs from "normal" human-generated RTL code).

pbreuer, 2021-04-05:
(1) An assignment has the form x = y. I am expecting to get (a) the names and types of the x (e.g. foofoo_gum[n][m], and "bool") and (b) the abstract grammar for y (e.g. y OR y | y AND y | y PLUS y | NOT y | VAR x | CONST {true|false} | ...).

(2) For the normal form of the RHS y of an assignment, I am expecting something like "all ORs of all ANDs of (VAR or NOT VAR)".

As a first step, (1) would be great, especially (1b).

I am expecting further info for parallelization and normal form as follows:

(3) either (a) the RHS y of each assignment x=y may contain a reference to an x that is the LHS of an assignment earlier in the sequence of assignments, or (b) it may not, and the only references allowed in each assignment are to input variables (named what?) and to the initial values of the variables x before any of these assignments happened (which have been saved as what names?), or (c) something between the two of (a) and (b).

[Saying what machine code instruction a C assignment compiles into is an everyday task. It would be one of add x a b, or mul x a b, etc, where x is the register in which the value of the LHS of x=y is being deposited, and a, b are registers in which the immediate subexpressions of the RHS y have been deposited, and the add, mul etc. correspond to the top level operator in the expression y. Far from being an example of something difficult to explain, it is something easy, and would be expected of about a second year undergraduate in computer science. But never mind.]

Though I thank you for the answer above, I am afraid it has turned out pretty much gibberish. Assignments do not "remain unchanged", because as code, they do not change. I think you mean "if the TARGET of an assignment IS NOT FURTHER ASSIGNED, then it remains ... unchanged", which would be tautologous, so not something to say. It cannot remain "relatively" unchanged, since changing is a 1/0 thing, and so on ...

As to whether or not using --threads does anything, it would cause either normal form (3a) or something approaching (3b) to be used, as not or as parallelism is required, respectively.  Specifying the number of threads say as 4 would cause assignments to be grouped into 4 sets each of which contains assignments that do not reference the final values of variables on the LHS  of the assignments in the other 3 sets.
 
But that could be done in two ways: either (a) passively by making the grouping on an existing set of assignments without mucking about with them individually, or (b) actively, by substituting through for a variable from the "wrong" set that appears on the RHS in an assignment with the expression assigned to it in that wrong set, and simplifying. That would be done to balance the computational load among the four (hypothetical) threads.

That's the kind of thing I expect to be explained.

What is an "mtask" and what is a "verilog-process" and why are they involved in an explanation about C/C++ code? It is really not acceptable to point vaguely elsewhere in an explanation.

I have noticed that several times Verilator has announced that it cannot simplify a loop, and that is strange because there should be no loops in the system verilog I supplied. Everything should have been in the standard form of just one loop (perhaps that is what you mean by verilog-process; i.e., there is just one verilog process) with pure logic inside it, resulting in a simple state machine that converts input to outputs. There will orignally have been memory elements inside the original HDL descripton, and they would have been mapped to elements of the one, monolithic  state changed by the one loop.

Nevertheless, the C++ code produced by Verilator contains some short loops:

   vlTOPp->ReadA__DOT__xs1[0xfU] = vlTOPp->ReadA__DOT__c__024vec_0[0xfU];
    __Vilp = 0U;
    while ((__Vilp <= 0x7fU)) {
        vlTOPp->ReadA__DOT__xs1_2[__Vilp] = vlTOPp->ReadA__DOT__c__024vec_9[__Vilp];
        __Vilp = ((IData)(1U) + __Vilp);
    }
    vlTOPp->ReadA__DOT__c__024xs1_app_arg_0[0U] = 1ULL+vlTOPp->ReadA__DOT__c__024vec_1[0U];

There is also some question about that last assignment. Is it really doing a 64-bit increment  ("LL") by adding 1? That's what ++ is for in C++. But that's minor. The questionable thing is the short loop. What is that all about? Aah .. it's copying a 128-element array from one place to another. That's the slow way!

If that array is in memory not registers, which surely any 128-element array would be if its elements are (probably) long long ints, it should be instead:

memcpy(vlTOPp->ReadA__DOT__xs1_2, vlTOPp->ReadA__DOT__c__024vec_9, 128*sizeof(long long));

and then the compiler will substitute fast machine code for a block move, or use one of the hardware's block move instructions, if there is one.

That loop's OK, just not a good idea. I'm not going to search through the generated code for what might have been the loops it could not optmize away, but I am anxious about them. There should not have been any to optimize, and more to the point, Verilator should ALWAYS be able to optimize them away by pushing them up to top level, since there are no delta-delay feedback loops in the original circuit. Data flow is unidirectional. It should also point me at whatever difficulty I had explicitly, so I can go look at it!

[... cut ...]
Imbetween there are approximately 12900 lines of automatically generated system verilog code, including blank lines in that count. I have offered to send it to you if you would like a look and that offer stands. If you like, I will run such code analysers as you suggest over it, but it looks straghtforward to me, and is really quite readable.  Your analysis would doubtless be of interest to the Clash development group at Twente and they would be glad to hear from you.

I would appreciate it if "I don't know about this end of things" is the answer, that it be said. It's a simple matter of being truthful. You wouldn't be a hardware engineer, perchance?

wsnyder, 2021-04-05:
Do you mean normal form in the sense of disjunctive normal form?  Verilator isn't a synthesizer, it keeps operations generally at a much higher level, e.g. a multiply stays a multiply unless it optimizes it otherwise (e.g. x*2 -> x<<1).

Or do you mean normal form in terms of A-normal form, no Verilator doesn't do that.

For the definition of "Verilog process" refer to IEEE 1800 chapter 9.

There are many additional optimizations that could be implemented including using e.g. memcpy.  There are bugs open describing many possiblities, and these already include optimizing the loops you mentioned.  If you would like to improve the optimizations that would be appreciated.

Verilator isn't usually used on Synthesized netlists, so that might be part of the reason you are hitting Verilation-time performance bugs. 

I don't know how soon it can be looked at, but no harm in filing an issue with the clash *output* attached and saying it's slow.  12900 normal-length lines seems tiny to take so long to Verilate (e.g. Swerv EH2 is about same number of statement lines and takes a few seconds).

pbreuer, 2021-04-06:
1) I checked the gcc translation of verilator's tight loop for copying an 128-element array of  64-bit ints (I commented that it looked odd above), and it is 4x slower than if verilator had emitted a memcpy call. The loop is recognized by gcc (yes, amazingly!) and reverse engineered to 256 repeats of a 32-bit copy (movls) while the memcpy is translated to 64 repeats of a 128-bit copy (movqs).

I checked a few more things too, and verilator's productions are always slow. Just the -Os makes that particular verilator loop 6.25% (average) to 12.5% (max) slower than if it had been 16-word aligned (which happens under -O2 at the cost of up to 63 extra bytes of padding in the program space, 64B/16 words being the cache line length). But it was already 4x slower than need be, so that's moot.

I have to conclude that verilator, though written neatly enough by somebody who evidently is above averagely good at C++ as a software engineering development task, appears NOT to be written by somebody who is familiar with how C/C++ is translated or how the machine code that it is translated to is executed on computers. The mistakes in that area are so ... glaring, that the inference seems forced. I guess that is par for the course - a software engineer generally has little more than a vague idea about how computer hardware  works and conversely, a hardware engineer has little idea about coding and program language semantics and computer sciency things in general. You're really lucky if you find someone who knows both areas.

2) I bothered to check what system verilog uses the keyword "package" for. Earlier I guessed here that it meant "collection of modules" (whatever "module" means!), and was not corrected, instead being  directed to some language standard or other uninformative location on the web. No. It means "header file", containing typedefs and other header-ish info, to be included in another system verilog file via an "import" line before the module start.

What on Earth was the point of not saying so? It is completely unacceptable.  It costs one sentence, as above, which is far less energy than it took to look up a url for a place that might define it in some officiously unintelligible way and plonk that down instead of the one-liner! 

One uses referencing not to _replace_ saying something, but to back it up! One can't say "see encyclopedia britannica", instead of "an elephant is a large animal", in reply to "what is an elephant".

There are only two conclusions possible to explain that: (a) obstructiveness, (b) doesn't know.

I am afraid I have to lean towards (b). It has been fantastically impossible to get straight, plain, clear answers here. When something has actually been said, it has been nonsense. The texts that have been pointed to instead of explanations are nonsensical in themselves, sentence by sentence. My diagnosis, as someone who has encountered this kind of thing several times in a professional career, is deliberate bluffing. Be warned. Look out for answers never given, evidence never given, reasoning never given, information always traveling towards the supposed source of information and expertise,  never away. People tend to suppose honesty and blame themselves, which is why bluffing works.

I've spent a week on verilator now. It took days to get it to produce anything. As per the header to this question, I had to guess that I had to name TWO files on the command line, one the header file and one the real source code file. I couldn't get even that straightforward answer. Has anybody suggested I add "import ReadA_types::*;" to the second file instead? No.

Then I asked how to get the output files named right. Did anybody tell me to use --prefix second_file_name ? No. Is that easy? Yes. Did I have to offer that solution? Yes.

Have I got any explanation of what a package is? Or of what a module is? No. Yet it takes a couple of lines. 

Have I got any explanation of what the actual code transformation that verilator does is? No. Only condemnation for wanting - and needing! - to know. Yet that is what ANY programmer needs to know! I have seen that before too, and it has been covering behavior, in my experience. 

Has an explanation appeared of what the form is that verilator transforms to? No. Yet it is evident at a glance that it is a simple loop ("eval_step" as body), containing assignments.

The standard, quick, informative answer would be to list the types of the assignees, and list the types of the assignment expressions, and state any conditions. I have had a look inside the AST (abstract syntax tree) node class definition now, and it looks like the assignees can be arrays, or arrays of arrays, or ..., up to any number of dimensions, but I have not noticed what the primitive types in the array can be yet.

Is that simple to say, if one knows? Yes. If one knows where to look, is it simple to say that? Yes.

As to the assignment expressions, is it simple to say if they are compound, or simple opertions? Yes. I have only seen single operators used in the huge C++ file I got out, but the verilator code itself says they can be compound. Does it cost anything to say that? No. Are they "all allowed operations of system verilog" (what they?). I don't know. Would it be hard to say so? No. It's just yes or no as an answer.  A quick look on the web tells me "it's all the standard C operators, plus 4 or 5 more weird things". [Really? How is one supposed to implement the the extra << operator in ha [... cut ...] . It looks like a duck.

I'm not going to waste more time with verilator. I take the lesson that it takes 14h to turn 29000 lines of already simplified, clearly organised, single mealy-machine, tested, system verilog into 4.5 million lines of obfuscatory and inefficient C++ and nobody will give an explanationof its organisation or structure.
That is not effective open source software engineering practice.

Perhaps the aim is to help hardware engineers!

Be warned, best wishes, etc.

Regards

PTB

wsnyder, 2021-04-06:
Thanks for considering Verilator.  As mentioned I still believe what you are experiencing is a bug, but given your strong opinions many which I've found insulting, I am glad to permanently end our exchanges.

For the thousands of others that use it, I'm glad it's suiting your needs, and thanks for your continued support.

closed 2021-04-06 by wsnyder