verilator/verilator #2871: How to build multiple file ? and multiple design ? how to build on .a ?

jalcim, 2021-04-03:
Hi,

I don't find in the documentation who explain how to build multiple design or multiple file...
My english is not perfect... its maybe because of that i have missed the explication.

1) With a file containing multiple design i can select only one with the "--top-module <name>" argument but i cant choice more of one.

2) And for compiling multiple file (containing multiple design) i cant just add verilog file on the command...

3) And finaly i do not find any documentation who explain how to build on library (file.a or file.so)

exemple of command i use to build
verilator -Wall --cc -Wno-style --top-module gate_buf gate_buf.v --exe --build main.cpp

Its possible to link me here where its explained in the documentation ? i don't find...
Or maybe i have not understand...
with icarus verilog it was simply to do that...

Thanks for helping <3

jalcim, 2021-04-03:
here the solution who i have find, but its only for multiple file with only one design, other problem its make a lib.a for every design.

`cmake_minimum_required(VERSION 3.20)
project(cmake_cosim)
find_package(verilator HINTS $ENV{VERILATOR_ROOT})
add_executable(Vgate_buf main.cpp)
add_library(gate_not STATIC)
add_library(gate_and STATIC)
target_link_libraries(Vgate_buf PRIVATE gate_not gate_and)
verilator_generate_key(KEY_INIT)
set(PROTECT_KEY ${KEY_INIT} CACHE STRING "Random seed for protection")

# Add the Verilated modules to the targets                                                                                                            
verilate(gate_not
  VERILATOR_ARGS --protect-lib gate_not
  --protect-key ${PROTECT_KEY}
  DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/gate_not
  SOURCES gate_not.v)

verilate(gate_and
  VERILATOR_ARGS --protect-lib gate_and
  --protect-key ${PROTECT_KEY}
  DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/gate_and
  SOURCES gate_and.v)

verilate(Vgate_buf
  VERILATOR_ARGS "-I${CMAKE_CURRENT_BINARY_DIR}/gate_buf"
  SOURCES gate_buf.v)
`

wsnyder, 2021-04-03:
As you figured out, if you want multiple "tops" to verilator this is just running verilator multiple times then linking against the appropriate executable.    You can also suppress the MULTITOP warning and all the designs will be verilated together, but in this case any I/O will be merged across the modules.

wsnyder, 2021-04-03:
Also as you figured out I think (but if others see this issue), if you omit the --exe the default behavior for Verilator is to build an archive you can link against.

BTW we're working on documentation improvements, but perhaps you'd like to contribute a pull request to improve the documentation?  

Closing as I think you got your answer, but feel free to followup if needed.

closed 2021-04-03 by wsnyder

jalcim, 2021-04-03:
> As you figured out, if you want multiple "tops" to verilator this is just running verilator multiple times then linking against the appropriate executable. You can also suppress the MULTITOP warning [...]

In first, thanks four your response,
Can i ask you for precise the subject of merging I/O for multitop ?
Why can't i build multiple design ? /*AUTOINST*/ still works !

Yes its an good idea, when i resolve my question about the building multiple design i would contribute on the doc about that.

wsnyder, 2021-04-03:
Multiple tops means e.g.

     module top1; endmodule
     module top2; endmodule

That is one isn't under the other.  AUTOINST connects modules so isn't relevant.  If you have

     module top1; input a; endmodule
     module top2; input b; endmodule

Then your C++ output will have both a and b as inputs.  There's no equivalent in other simulators as there's nothing "above" the top as there is (potentially) with Verilator.

jalcim, 2021-04-03:
Thanks for your explain !