verilator/verilator #2874: Verilator fails to ignore tgrailing comma in non-ANSI module declaration

udif, 2021-04-05:
Using the head version of Verilator, the following module fails:
```
module t5(a,);
input a;
endmodule
```
This yields:
```
$ ~/git/verilator/bin/verilator --lint-only t5.v 
%Error: t5.v:1:13: syntax error, unexpected ')', expecting '['
    1 | module t5(a,);
      |             ^
%Error: Exiting due to 1 error(s)
```
While not particularly pretty, this is allowed as seen both [here](https://www.accellera.org/images/eda/sv-bc/10162.html) as well as here #321 (which implies this is expected to work)

This is yet another fight against 3rd party IP (this time from a giant CPU vendor codebase).

udif, 2021-04-05:
Just to make it clear, `verilator --version` reports `Verilator 4.201 devel rev v4.200-70-g961a2fef`

simdream, 2021-04-05:
the last , is ok in instanciation but not in the module definition?

wsnyder, 2021-04-05:
Ending commas are NOT ignored, they are null ports.  Using your example you'd connect to it like this:

     t5 t5(a,b);

See IEEE 1800-2017 "The port expression is optional because ports can be defined that do not connect to anything internal to the
module. "

If you did not supply the "b" you should get a PINMISSING warning (similar to if you had "module t5(, b)" with a missing.   Most trailing commas/null ports are mistakes and people don't understand that they are not ignored; I suspect your example is one of them.  But if someone wants to submit a patch to support this, I'll take it provided there is a new NULLPORT warning added describing what this does and throws the PINMISSING unless the missing ports are supplied.  Note a 5 minute attempt at supporting this in the parser caused bison shift/reduce conflicts, so be prepared to spend some effort.

udif, 2021-04-06:
Yes, I've read about null port in Shalom Bresticker's comment, but as he said, I've yet to see a SINGLE design using them or even a SINGLE RTL designer who actually knows they are allowed.

I've had my share of Yacc/Bison coding in the past (mostly Yosys), so I'll try taking a look at this myself. I'll leave the leading comma case out as it is clearly way less useful than solving the trailing comma case (which comes from blindly using copy/paste when creating port lists).

udif, 2021-04-06:
Added pull request #2875

wsnyder, 2021-06-13:
This was fixed with  #2875.

closed 2021-06-13 by wsnyder