verilator/verilator #2876: Unsized expressions disallowed in concatenations, even with casts

rswarbrick, 2021-04-07:
This code:
```systemverilog
module cast (input clk_i, output [63:0] out);
  assign out = {32'(0), 32'd0};
endmodule
```
produces the following width warnings when passed to Verilator:
```
$ verilator --lint-only cast.sv
%Warning-WIDTHCONCAT: cast.sv:2:21: Unsized numbers/parameters not allowed in concatenations.
                                  : ... In instance cast
    2 |   assign out = {32'(0), 32'd0};
      |                     ^
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
%Warning-WIDTHCONCAT: cast.sv:2:23: Unsized numbers/parameters not allowed in replications.
                                  : ... In instance cast
    2 |   assign out = {32'(0), 32'd0};
      |                       ^
%Error: Exiting due to 2 warning(s)
```

I think the "no unsized values in a concatenation" rule is probably too strict. In the IEEE 1800 spec, it appears in section 11.4.12, where they say:
> Unsized constant numbers shall not be allowed in concatenations. This is because the size of each operand in the concatenation is needed to calculate the complete size of the concatenation.

That's perfectly reasonable, but I don't think it applies here: the expression `32'(0)` has width 32. (I looked at the "Expression bit lengths" section but, in keeping with its general vagueness, it doesn't say a static cast has a self-determined width. I think it probably should though!)

Note that Verilator *does* (correctly) allow things like `assign out = 64'(0);`.

wsnyder, 2021-04-07:
I believe this is a duplicate of #2849, did you try using git master?

rswarbrick, 2021-04-07:
Ah! You're completely right! I think messed up my paths so was running the wrong version of Verilator. Slightly embarrassing that I didn't recognise the bug, since I reported that one too :-) In my defence, the original version of this had a typecast which hid the "32", but I probably should have noticed once I had the cut-down version!

closed 2021-04-07 by rswarbrick