verilator/verilator #2877: XML missing instance port connection (Verilog 1995, ordered list)

errare33, 2021-04-07:
The information needed to determine port connections of an instance is missing in the XML when the module is defined using the Verilog 1995 standard and the instance is declared with the ports as an ordered list.

Verilator (Verilator 4.018 2019-08-29 rev UNKNOWN_REV) was used with the --xml-only option to generate XML for the following example:

```verilog
module Test_Bottom(B,C,D);
  input B;
  input C;
  output D;
  wire B;
  wire C;
  wire D;
  wire not_b;
  not g1(not_b, B);
  and g2(D,not_b, C);
endmodule

module Test_Top(in0,in1, out);
  input in0;
  input in1;
  output out;
  wire in0;
  wire in1;
  wire out;
  Test_Bottom t1(in0,in1,out);
endmodule
```
Running again with the order of the ports changed, Test_Bottom(C,B,D), the XML is exactly the same. So, the information needed to correctly map the ports is not in the XML. However, the final .tree file dumped using --debug shows that Verilator is assigning the ports correclty. So, the verilation is being done correclty, just the information is not reflected in the XML.

```verilog
module Test_Bottom(C,B,D);
  input B;
  input C;
  output D;
  wire B;
  wire C;
  wire D;
  wire not_b;
  not g1(not_b, B);
  and g2(D,not_b, C);
endmodule

module Test_Top(in0,in1, out);
  input in0;
  input in1;
  output out;
  wire in0;
  wire in1;
  wire out;
  Test_Bottom t1(in0,in1,out);
endmodule
```

wsnyder, 2021-04-07:
The XML is post elaboration, so the pin number information is no longer present; all the connections have been converted to name based in the output XML, hence it is correctly the same (or I'm not understanding).  Hopefully you don't need this information.

If you do need the pin numbers for some reason I'd accept a pull request to do that.  You'd need to take the AstPin/pinNum() value and put it into the AstVar that identifies the pin, then put into the XML in V3EmitXml.

errare33, 2021-04-07:
For example, I'm looking for a way to extract whether __pinNumber1 connects to B or C in the two cases. That doesn't appear to be in the XML based on the fact that the two XML outputs are identical.

```XML
<cells>
<cell fl="c17" name="Test_Top" submodname="Test_Top" hier="Test_Top">
<cell fl="c24" name="t1" submodname="Test_Bottom" hier="Test_Top.t1"/>
</cell>
</cells>
<netlist>
<module fl="c17" name="Test_Top" origName="Test_Top" topModule="1">
<var fl="c18" name="in0" dtype_id="1" dir="input" vartype="logic" origName="in0"/>
<var fl="c19" name="in1" dtype_id="1" dir="input" vartype="logic" origName="in1"/>
<var fl="c20" name="out" dtype_id="1" dir="output" vartype="logic" origName="out"/>
<instance fl="c24" name="t1" defName="Test_Bottom" origName="t1">
<port fl="c24" name="__pinNumber1" direction="in" portIndex="1">
<varref fl="c24" name="in0" dtype_id="1"/>
</port>
<port fl="c24" name="__pinNumber2" direction="in" portIndex="2">
<varref fl="c24" name="in1" dtype_id="1"/>
</port>
<port fl="c24" name="__pinNumber3" direction="out" portIndex="3">
<varref fl="c24" name="out" dtype_id="1"/>
</port>
</instance>
</module>
<module fl="c5" name="Test_Bottom" origName="Test_Bottom">
<var fl="c6" name="B" dtype_id="1" dir="input" vartype="logic" origName="B"/>
<var fl="c7" name="C" dtype_id="1" dir="input" vartype="logic" origName="C"/>
<var fl="c8" name="D" dtype_id="1" dir="output" vartype="logic" origName="D"/>
<var fl="c12" name="not_b" dtype_id="1" vartype="logic" origName="not_b"/>
<contassign fl="c13" dtype_id="1">
<not fl="c13" dtype_id="1">
<varref fl="c13" name="B" dtype_id="1"/>
</not>
<varref fl="c13" name="not_b" dtype_id="1"/>
</contassign>
<contassign fl="c14" dtype_id="1">
<and fl="c14" dtype_id="1">
<varref fl="c14" name="not_b" dtype_id="1"/>
<varref fl="c14" name="C" dtype_id="1"/>
</and>
<varref fl="c14" name="D" dtype_id="1"/>
</contassign>
</module>
<typetable fl="a0">
<basicdtype fl="c21" id="1" name="logic"/>
</typetable>
</netlist>
```

wsnyder, 2021-04-08:
Agreed, it was in the pinNum of the module's port, but that got removed.  If you'd like this information can you make a pull that adds pinNum to AstVar?  Then where AstPort is resolved in V3LinkDot.cpp, set the Var's pinNum to the Port's pinNum.  Then in V3EmitXml print it out.

closed 2021-06-19 by wsnyder