verilator/verilator #2879: classes and randomization constraints support

agrobman, 2021-04-07:
What are verilator team plans for SV clases and randomization constraints support? when (if) these will be supported, at  least partially?  

Thanks
Alex

wsnyder, 2021-04-07:
Depends how soon you start submitting pull requests to support whatever you need ;)

But seriously, without that probably a year to UVM at current rate, then randomization would be next.

agrobman, 2021-04-07:
can you add something simple for now? Like this:

```
class ready_gen;
int prob;
rand bit value;

constraint c_prob{
    value dist {1:= prob, 0:= 100-prob};
}

function new(int wt);
    prob = wt;
endfunction

function get();
    assert(this.randomize);
    return value;
endfunction
endclass

ready_gen aready_gen;
initial begin
    aready_gen = new(90);
end

```

wsnyder, 2021-04-08:
I suspect it will be a while before I personally have time to work on this, especially given general class features I suspect are higher priority, but if you or @kbieganski or anyone wants to make a pull request please do.  Note the parser already has most the appropriate rules in place, so the next step is to add appropriate AstNodes for the given new constructs.

wsnyder, 2024-08-11:
While still much to do, basic randomization works.

closed 2024-08-11 by wsnyder