verilator/verilator #2883: Basic interfaces support for RTL

agrobman, 2021-04-11:
Any plans to support interfaces in near future?
( at least in very basic form as just bunch of wires for modules connectivity) something like this:

```

// basic interface 
interface tl_if;
logic a_ready;
logic a_valid;
endinterface

// multibus parameterizable interface
interface multi_tl #(CORES=4);
tl_if bus[CORES]();
endinterface

//inner module, connected by the basic interface
module tl_bus_proc(interface tl);
endmodule

// parent  module aka interconnect, containing parameterizable number of inner modules
module llc # (
    parameter CORES = 1,
    parameter SIZE = 3
)
( 
input clk,
interface tls
);

wire aready2 = tls.bus[2].a_ready; 

for(genvar i=0; i<CORES; i++) begin : BP
    tl_bus_proc bp(.tl(tls.bus[i]));
end

tl_if b[2]();
endmodule

//top module, containing N cores with standard interfaces
module top;

wire clk;
for( genvar i = 0, i<=3, i++) begin : CPU
    core core(.tl(tls.bus[i]));
end
multi_tl #3 tls();

llc  #(.CORES(2)) llc0 (.clk(clk), .tls(tls));
llc  #(.CORES(3)) llc1 (.*);
endmodule
```

It is very annoying to type multiple signals standard interfaces and their connections, especially when there are multiple similar interfaces like in the interconnect designs . Also it's kind of hard to make these designs to be parameterized with number of buses...

wsnyder, 2021-04-11:
Verilator already supports interfaces.  What more specific feature is missing?

agrobman, 2021-04-12:
Getting with 4.200:

%Error-UNSUPPORTED: interfaces.sv:15:20: Unsupported: virtual or generic interfaces
   15 | module tl_bus_proc(interface tl);
      |                    ^~~~~~~~~
%Error-UNSUPPORTED: interfaces.sv:24:1: Unsupported: virtual or generic interfaces
   24 | interface tls
      | ^~~~~~~~~
%Error: Exiting due to 2 error(s)

Not sure how to use parameterized interface in module portlist otherwise ... ( I need to define a module with variable number of buses/interfaces

wsnyder, 2021-04-12:
I suspect you didn't intend to use a generic interface.

     module tl_bus_proc(interface tl);

should be

     module tl_bus_proc(multi_tl tl);

wsnyder, 2021-04-12:
If you need parameters you would pass them from the module into the interface:

    module #(CORES = 4) tl_bus_proc(multi_tl #(.CORES(CORES)) tl);

agrobman, 2021-04-12:
No the tl_bus_proc gets one interface/bus, but llc needs to be parameterized to support N buses/interfaces.

llc instantiates N "bus processors"

But I think, I've found better solution - using array of interfaces as port of llc :

```
module llc # (
    parameter CORES = 1,
    parameter SIZE = 3
)
( 
input clk,
tl_if tl[0:CORES-1]
);

...
endmodule
```
thanks again. Error message was confusing me ..

wsnyder, 2021-04-12:
Glad you found a workaround.

closed 2021-04-12 by wsnyder