verilator/verilator #2890: Compilation with dependent packages requires file list to be ordered based on dependencies

dhorton-imc, 2021-04-15:
Verilator fails when processing a design using a file list, if a package earlier in the files list references a package later in the file list.  The error generated references IEEE 1800-2017, section 26.3.

`Package/class 'foo_pkg' not found, and needs to be predeclared (IEEE 1800-2017 26.3)`

Reading section 26.3 of the spec the only text I saw that appeared relevant to this error is the following:

`"The compilation of a package shall precede the compilation of scopes in which the package is imported."`

I interpret this as meaning that a package has to be compiled before any users of that package are compiled.  Which makes perfect sense.  But I don't interpret it as enforcing any limits/constraints on tools that haven't yet seen/processed the package.  They can simply defer compilation until such time as they process the referenced package.  Of course if no such package exists at all then it is an error.

The restriction in verilator means that file lists need to take into account all the dependencies between files.  If definitions are moved or refactored, this often necessitates a change to the file list, if dependency orders have changed.  In a large project with many packages this can be time consuming.

Other tools seem to act differently (Vivado for example) and are happy with file lists with files in any order.  This restriction in verilator seems contrary to the spec (unless I've missed or misunderstood something) and at odds with other tools.

Trivial Example:
```SystemVerilog
package a;
    import b::*;

    function automatic integer a();
        return 0;
    endfunction
endpackage
```

```SystemVerilog
package b;
    function automatic integer b();
        return 1;
    endfunction
endpackage
```

```SystemVerilog
module c(
        input  wire         clk,
        input  wire         res_n,

        output logic      result
    );

    import a::*;
    import b::*;

    logic a_b;
    assign a_b = a() || b();

    always_ff @(posedge clk)
    begin
        if (!res_n)
        begin
            result <= '0;
        end
        else
        begin
            result <= {a_b ? 1'b1 : 1'b0};
        end
    end
endmodule
```

```
# Packages
# 'a' depends on 'b' (unnecessarily in this trivial example)
# In this order, verilator fails.
# Reverse the order of lines 5/6 and it is fine.
a.sv
b.sv

# Module
c.sv
```

Reversing the order of `a.sv` and `b.sv` in `files.lst` resolves the issue.

wsnyder, 2021-04-15:
"Compilation" is defined in IEEE 3.12 as "the process of reading in SystemVerilog source code".  Also as SystemVerilog is a context sensitive grammar, it makes sense that packages must be predeclared.   So I believe Verilator is compliant.

Regardless I appreciate this might deviate from other simulators and cause pain, however this is currently a strong assumption in the parser that isn't likely to get fixed.  There is a project to replace the parser, if/when that is finished this would be resolved.

As to determining what files, the typical package coding practice if followed avoids these problems.  Put every package in a header file e.g. my_pkg.vh, and inside use header guards

     `ifndef MY_PKG_VH_  // guard
     `define MY_PKG_VH_
     package ....;
     `endif   // guard

Then in any modules where a package is needed include the file

    `include "my_pkg.vh'

If followed appropriately, and your module names match the filenames, you should be able to pass only the top module's filename and Verilator will find all other packages, modules, etc, automatically.  Similarly for several (all?) other commercial tools.

wsnyder, 2021-04-15:
As above although agree is annoying won't be fixed soon, so not worth keeping bug open.  Thanks for your report though!

closed 2021-04-15 by wsnyder

Radhe93, 2023-09-05:
hi can you help me with this 

Error-PKGNODECL: wrapper.sv:8:9: Package/class 'bp_common_pkg' not found, and needs to be predeclared (IEEE 1800-2017 26.3)
    8 |  import bp_common_pkg::*;
      |         ^~~~~~~~~~~~~
                  ... For error description see https://verilator.org/warn/PKGNODECL?v=5.015
%Error: wrapper.sv:8:9: Importing from missing package 'bp_common_pkg'
    8 |  import bp_common_pkg::*;
      |         ^~~~~~~~~~~~~
%Error: wrapper.sv:9:9: Importing from missing package 'bp_be_pkg'
    9 |  import bp_be_pkg::*;
      |         ^~~~~~~~~
%Error: wrapper.sv:10:9: Importing from missing package 'bp_me_pkg'
   10 |  import bp_me_pkg::*;
      |         ^~~~~~~~~
%Error: wrapper.sv:11:26: syntax error, unexpected IDENTIFIER
   11 |  #(parameter bp_params_e bp_params_p = e_bp_default_cfg
      |                          ^~~~~~~~~~~
%Error: wrapper.sv:12:4: Define or directive not defined: '`declare_bp_proc_params'
   12 |    `declare_bp_proc_params(bp_params_p)
      |    ^~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:14:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   14 |    , parameter num_caches_p = 1
      |      ^~~~~~~~~
%Error: wrapper.sv:15:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   15 |    , parameter wt_p = 1
      |      ^~~~~~~~~
%Error: wrapper.sv:16:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   16 |    , parameter sets_p = dcache_sets_p
      |      ^~~~~~~~~
%Error: wrapper.sv:17:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   17 |    , parameter assoc_p = dcache_assoc_p
      |      ^~~~~~~~~
%Error: wrapper.sv:18:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   18 |    , parameter block_width_p = dcache_block_width_p
      |      ^~~~~~~~~
%Error: wrapper.sv:19:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   19 |    , parameter fill_width_p = dcache_fill_width_p
      |      ^~~~~~~~~
%Error: wrapper.sv:20:4: Define or directive not defined: '`declare_bp_bedrock_lce_if_widths'
   20 |    `declare_bp_bedrock_lce_if_widths(paddr_width_p, lce_id_width_p, cce_id_width_p, lce_assoc_p)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:21:4: Define or directive not defined: '`declare_bp_bedrock_mem_if_widths'
   21 |    `declare_bp_bedrock_mem_if_widths(paddr_width_p, did_width_p, lce_id_width_p, lce_assoc_p)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:21:37: syntax error, unexpected '(', expecting ';'
   21 |    `declare_bp_bedrock_mem_if_widths(paddr_width_p, did_width_p, lce_id_width_p, lce_assoc_p)
      |                                     ^
%Error: wrapper.sv:22:4: Define or directive not defined: '`declare_bp_cache_engine_if_widths'
   22 |    `declare_bp_cache_engine_if_widths(paddr_width_p, dcache_ctag_width_p, sets_p, assoc_p, dword_width_gp, block_width_p, fill_width_p, dcache)
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:25:6: syntax error, unexpected parameter, expecting IDENTIFIER or randomize
   25 |    , parameter lock_max_limit_p=8
      |      ^~~~~~~~~
%Error: wrapper.sv:27:6: syntax error, unexpected localparam, expecting IDENTIFIER or randomize
   27 |    , localparam cfg_bus_width_lp= `bp_cfg_bus_width(vaddr_width_p, hio_width_p, core_id_width_p, cce_id_width_p, lce_id_width_p)
      |      ^~~~~~~~~~
%Error: wrapper.sv:27:35: Define or directive not defined: '`bp_cfg_bus_width'
   27 |    , localparam cfg_bus_width_lp= `bp_cfg_bus_width(vaddr_width_p, hio_width_p, core_id_width_p, cce_id_width_p, lce_id_width_p)
      |                                   ^~~~~~~~~~~~~~~~~
%Error: wrapper.sv:27:66: syntax error, unexpected ','
   27 |    , localparam cfg_bus_width_lp= `bp_cfg_bus_width(vaddr_width_p, hio_width_p, core_id_width_p, cce_id_width_p, lce_id_width_p)
      |                                                                  ^
%Error: wrapper.sv:29:39: Define or directive not defined: '`bp_be_dcache_pkt_width'
   29 |    , localparam dcache_pkt_width_lp = `bp_be_dcache_pkt_width(vaddr_width_p)
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:31:6: syntax error, unexpected localparam, expecting IDENTIFIER or randomize
   31 |    , localparam lg_num_lce_lp = `BSG_SAFE_CLOG2(num_lce_p)
      |      ^~~~~~~~~~
%Error: wrapper.sv:31:33: Define or directive not defined: '`BSG_SAFE_CLOG2'
[... 47 lines cut ...]
%Error: wrapper.sv:118:3: syntax error, unexpected IDENTIFIER
  118 |   bp_bedrock_lce_fill_header_s [num_caches_p-1:0] lce_fill_header_lo;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:123:3: Define or directive not defined: '`declare_bp_cfg_bus_s'
  123 |   `declare_bp_cfg_bus_s(vaddr_width_p, hio_width_p, core_id_width_p, cce_id_width_p, lce_id_width_p);
      |   ^~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:123:24: syntax error, unexpected '('
  123 |   `declare_bp_cfg_bus_s(vaddr_width_p, hio_width_p, core_id_width_p, cce_id_width_p, lce_id_width_p);
      |                        ^
%Error: wrapper.sv:124:3: Define or directive not defined: '`bp_cast_i'
  124 |   `bp_cast_i(bp_cfg_bus_s, cfg_bus);
      |   ^~~~~~~~~~
%Error: wrapper.sv:382:7: syntax error, unexpected IDENTIFIER
  382 |       bp_bedrock_lce_resp_header_s cce_lce_resp_header_li;
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:387:7: syntax error, unexpected IDENTIFIER
[... cut ...]
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: wrapper.sv:393:7: syntax error, unexpected IDENTIFIER
  393 |       bp_me_xbar_stream
      |       ^~~~~~~~~~~~~~~~~
%Error: Exiting due to 45 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.

i have gone through the manual and it says that ""An error that a package/class appears to have been referenced that has not yet been declared. According to IEEE 1800-2017 26.3, all packages must be declared before being used."".

wsnyder, 2023-09-05:
@Radhe93 I don't understand your question as you appeared to correctly understood what the error means.  Did you declare the package before that line of code or not?  If not, you need to.  If you did please open a new issue rather than attaching to an old one, so we can debug, thanks.