verilator/verilator #2891: verilator4200 version coverage error

RaynardQiao, 2021-04-15:
verilator version： 4200 

error message：
%Error: Internal Error: twocounter.v:31:32: ../V3Const.cpp:116: Only true is set in Xor tree
   31 |     .data_i   ({1'b0, bank_rdata_synd_vec_m3[0][31:0]}),
        |  
              ... See the manual and https"//verilator.org from more assistance.

Code  twocounter.v:

module  ecc_check_pipe (
    input logic        clk,
    input logic        bank_rd_m3,
    input logic [32:0]  data_i,
    output logic     ecc_err_o
);
logic [3:0]   check_group_6_0;
logic check_group_6_0_q;

always_comb check_group_6_0 = {data_i[0], data_i[2], data_i[4], data_i[7] };
always_ff @(posedge clk) if (bank_rd_m3)  check_group_6_0_q <=^check_group_6_0;
endmodule

module twocounter
(
    input logic clk,
    input logic bank_rd_vec_m3,
    input logic [1:0][38:0] bank_rdata_synd_vec_m3
);
logic bank_ecc_err_vec_m4;

ecc_check_pipe u_bank_data_ecc_check(
    .clk       (clk),
    .bank_rd_m3  (bank_rd_vec_m3),
    .data_i  ({ 1'b0, bank_rdata_synd_vec_m3[0][31:0] } ),
    .ecc_err_o  (bank_ecc_err_vec_m4)
);
endmodule

cmd is :
verilator_bin  --sc --top-module twocounter   twocounter.v  --coverage

wsnyder, 2021-04-15:
You can use "-Oo" to workaround this issue.  @yTakatsukasa could you please take a look?

yTakatsukasa, 2021-04-16:
I expect #2892 fixes this issue.
If you could try the latest master, that would be appreciated.

Thank you for reporting with small testcase.

RaynardQiao, 2021-04-17:
testcase passed with latest master.

There's a similar problem:

verilator version： latest master

error message：
%Error: Internal Error: twocounter.v:31:32: ../V3Const.cpp:117: Only true is set in Xor tree
31 | .data_i ({1'b0, bank_rdata_synd_vec_m3[i][31:0]}),
|
... See the manual and https"//verilator.org from more assistance.

Code twocounter.v:

module  ecc_check_pipe (
    input logic        clk,
    input logic        bank_rd_m3,
    input logic [32:0]  data_i,
    input logic [6:0] ecc_check_i,
    output logic     ecc_err_o
);
logic [3:0]   check_group_6_0;
logic [3:0]   check_group_6_3;
logic  ecc_err_tmp;
logic one_bit_ecc_err_tmp;
logic [6:0] ecc_syndrome;

logic check_group_6_0_q;
logic check_group_6_1_q;
logic check_group_6_2_q;
logic check_group_6_3_q;

always_comb check_group_6_0 = {data_i[0], data_i[2], data_i[4], data_i[7] };
always_comb check_group_6_3 = {data_i[28], data_i[31], data_i[32], ecc_check_i[6] };
always_ff @(posedge clk) if (bank_rd_m3)  check_group_6_0_q <=^check_group_6_0;
always_ff @(posedge clk) if (bank_rd_m3)  check_group_6_3_q <=^check_group_6_3;
always_comb ecc_syndrome[6] = ^{check_group_6_0_q, check_group_6_1_q, check_group_6_2_q, check_group_6_3_q};

always_comb ecc_err_tmp = |ecc_syndrome;
always_comb one_bit_ecc_err_tmp = ^ecc_syndrome;
endmodule

module twocounter
(
    input logic clk,
    input logic [7:0]bank_rd_vec_m3,
    input logic [7:0][38:0] bank_rdata_synd_vec_m3
);
logic [7:0]  bank_ecc_err_vec_m4;

generate
    for (genvar i = 0; i < 8; i=i+1) begin: data_ecc_check
        ecc_check_pipe u_bank_data_ecc_check(
            .clk       (clk),
            .bank_rd_m3  (bank_rd_vec_m3[i]),
            .data_i  ({ 1'b0, bank_rdata_synd_vec_m3[i][31:0] } ),
            .ecc_check_i ( bank_rdata_synd_vec_m3[i][38:32]),
            .ecc_err_o  (bank_ecc_err_vec_m4[i])
        );
    end
endgenerate
endmodule

cmd is :
verilator_bin --sc --top-module twocounter twocounter.v --trace

yTakatsukasa, 2021-04-17:
Oh, thanks.
I will check again.

yTakatsukasa, 2021-04-25:
The second trial (#2899) is merged to master.
Please let me know the result if you have a chance to try the latest master.

yTakatsukasa, 2021-04-26:
Closing this ref: https://github.com/verilator/verilator/pull/2899#issuecomment-826558678

PR #2892 (2021-04-16): Fix assertion failure in bitOpTree
Fixes #2891 

Thanks for the simple testcase.
closed 2021-04-26 by yTakatsukasa