verilator/verilator #2895: Internal Error in `V3EmitC.cpp` related to a literal for an array of strings

jeras, 2021-04-17:
To reproduce:
```
git clone https://github.com/jeras/rp32.git -b 4f078eae04583c7c765e1cb8569c6cec4cfdb1db
cd rp32
cd sim
make -f Makefile.verilator lint
```
The output is:
```
/home/ijeras/VLSI/verilator/bin/verilator --lint-only ../rtl/riscv_isa_pkg.sv ../rtl/rp_br.sv ../rtl/rp_gpr.sv ../rtl/rp_alu.sv ../rtl/rp_core.sv ../tbn/riscv_asm_pkg.sv ../tbn/mem.sv ../tbn/rp_tb.sv --top rp_tb
%Error: Internal Error: ../tbn/riscv_asm_pkg.sv:8:34: ../V3EmitC.cpp:1268: Unknown node type reached emitter: INITITEM
    8 | localparam string REG_X [0:31] = '{"zero", "ra", "sp", "gp", "tp", "t0", "t1", "t2", "s0/fp", "s1", "a0", "a1", "a2", "a3", "a4", "a5",
      |                                  ^~
                        ... See the manual and https://verilator.org for more assistance.
make: *** [Makefile.verilator:98: lint] Error 1
```

Tested using Verilator `stable` and the almost latest commit `f579e557`.

I tried to create a short test to trigger this issue, but I was not able to, the simple test did not trigger a warning, and it run the example with proper printouts as expected. the example tarball is attached.
[test.zip](https://github.com/verilator/verilator/files/6329534/test.zip)
From this I suppose the problem is not in the C emitter itself.

Sorry I cant help fixing this, I would spend days before I could even read C++ code with confidence.

jeras, 2021-04-17:
Please ignore this issue for now, there are too many parser issues with the RTL to focus on this detail.

wsnyder, 2021-04-24:
Committing partial fix just to ignore this for lint - larger fix for C++ emit use needs some work.

jeras, 2021-06-06:
I spent a bit more time on this issue and was able to get to something like a minimal example (attached).

To trigger the issue, the path between the string array and the $display must go through two functions, and the second one must have a case statement. I tried to just use 2 functions, but it did not fail if the second one did not contain a case statement. I also tried to remove the first function, just keeping the function with the case statement and again it did not fail. If the `5'd1` line in the case statement (2-nd function call) is commented out the test does not fail to compile.

The example code was tested in a couple of professional simulators and it worked as expected.

The testcase is in the regress format, but I could not think of a good name, since it is not a single feature or a specific combination of features.

```SystemVerilog
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2018 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

module t (/*AUTOARG*/);

   localparam string REG_X [0:31] = '{"zero", "ra", "sp", "gp", "tp", "t0", "t1", "t2", "s0/fp", "s1", "a0", "a1", "a2", "a3", "a4", "a5",
                                      "a6", "a7", "s2", "s3", "s4", "s5", "s6", "s7", "s8", "s9", "s10", "s11", "t3", "t4", "t5", "t6"};

   function string reg_x (logic [5-1:0] r, bit abi=1'b0);
     reg_x = abi ? REG_X[r] : $sformatf("x%0d", r);
   endfunction: reg_x

   // the issue is triggered by a second function containing a case statement
   function string f2 (logic [5-1:0] r, bit abi=0);
      case (r)
         5'd0:    f2 = $sformatf("nop");
         5'd1:    f2 = $sformatf("reg %s", reg_x(r[5-1:0], abi));
         default: f2 = $sformatf("ILLEGAL");
      endcase
   endfunction: f2

   initial begin
      for (int unsigned i=0; i<32; i++) begin
         $display("REGX: %s", reg_x(i[4:0], 1'b1));
      end
      $display("OP: %s", f2(5'd7));
      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

gezalore, 2021-07-23:
This should now be fixed on master, please open a new issue if you find further problems.

closed 2021-07-23 by gezalore

jeras, 2021-09-05:
I just updated my code, the issue seems to be fixed and the my code is working correctly.

commit 90c917ee (2021-07-23): Enable now supported tests for string array initializers

Fixes #2895.