verilator/verilator #2897: perl/verilator_includer not run. No "__ALL.o" object file

emohohen-that-spells, 2021-04-18:
Good day,

I discovered a problem recently where running make (see commands below) on verilated SV doesn't create the "__ALL.o" object file. For example, the attached SV aims to create a 16 bit, 64 word RAM based on the NAND2TETRIS project and the problem was discovered with the following commands:
```
verilator -GDATA_WIDTH=16 -Wall -Wpedantic -cc -top RAM64 RAM64.sv
make -C obj_dir/ -f VRAM64.mk
```
I was expecting there to be the object file obj_dir/VRAM64__ALL.o to link with when compiling a test bench. It does create VRAM64__ALL.a but static linking with this fails.

After poking around a bit, I found that up to RAM32 can be verilated without issue. And looking through the make log, I picked out and ran manually this command:
```
/usr/bin/perl $PATH_TO/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VRAM64.cpp  VRAM64_Register__D10.cpp  VRAM64_Register__D10__1__Slow.cpp  VRAM64_Register__D10__Slow.cpp  VRAM64__Slow.cpp  VRAM64__Syms.cpp > VRAM64__ALL.cpp
```
Followed by
```
ccache g++  -I.  -MMD -I/home/fried/.local/share/verilator/include -I/home/fried/.local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++14 -Os -c -o VRAM64__ALL.o VRAM64__ALL.cpp
```
and
```
ar -rcs VRAM64__ALL.a VRAM64__ALL.o
```
Now I have the object to link with when compiling the testbench C++.

Verilator 4.201 devel rev v4.200-81-g0d6099b2
[RAM64.sv.txt](https://github.com/verilator/verilator/files/6331165/RAM64.sv.txt)

Best regards,
emohohen

wsnyder, 2021-04-18:
The problem should be unrelated to the design - it seems to work fine for me - so is probably something related to the operating system or installation.

When you run the make command what did it print?  There should have been some error printed which might suggest the problem.

emohohen-that-spells, 2021-04-18:
No errors. I rerun make with -d to generate the attached logs of RAM64 and RAM32. I've also set RAMn as top module and use -GRAM_DEPTH when launching verilator. It's around line 2067 where things go different for RAM64:

```
  Considering target file 'VRAMn__ALL.a'.
   File 'VRAMn__ALL.a' does not exist.
   Looking for an implicit rule for 'VRAMn__ALL.a'.
   Trying pattern rule with stem 'VRAMn__ALL'.
   Found an implicit rule for 'VRAMn__ALL.a'.
    Considering target file 'VRAMn.o'.
     File 'VRAMn.o' does not exist.
     Looking for an implicit rule for 'VRAMn.o'.
     Trying pattern rule with stem 'VRAMn'.
     Trying implicit prerequisite 'VRAMn.cpp'.
     Found an implicit rule for 'VRAMn.o'.
      Considering target file 'VRAMn.cpp'.
```
Whereas, RAM32 goes ahead with VRAMn__ALL.o

```
  Considering target file 'VRAMn__ALL.a'.
   File 'VRAMn__ALL.a' does not exist.
   Looking for an implicit rule for 'VRAMn__ALL.a'.
   Trying pattern rule with stem 'VRAMn__ALL'.
   Found an implicit rule for 'VRAMn__ALL.a'.
    Considering target file 'VRAMn__ALL.o'.
     File 'VRAMn__ALL.o' does not exist.
     Looking for an implicit rule for 'VRAMn__ALL.o'.
     Trying pattern rule with stem 'VRAMn__ALL'.
     Trying implicit prerequisite 'VRAMn__ALL.cpp'.
     Found an implicit rule for 'VRAMn__ALL.o'.
      Considering target file 'VRAMn__ALL.cpp'.
```

[makeRAM32.log](https://github.com/verilator/verilator/files/6331381/makeRAM32.log)
[makeRAM64.log](https://github.com/verilator/verilator/files/6331382/makeRAM64.log)

wsnyder, 2021-04-18:
I'm confused by your mentioning RAM32 and RAM64.  What exactly are you running to get both versions?  And what do you expect to happen that is not happening?  (You may want to try removing obj_dir between runs.)

emohohen-that-spells, 2021-04-18:
The RAMn module has parameter RAM_DEPTH, where 32 and 64 refer to the RAM_DEPTH specific to the respective RAM modules. RAM32 is just the deepest RAM I can create that doesn't have this problem. For convenience of generating the log files for different depths, I set RAMn as top module and set the parameter at command line. The make logs were generated by these commands:
```
rm -r obj_dir/
verilator -GRAM_DEPTH=32 -GDATA_WIDTH=16 -Wall -Wpedantic -cc -top RAMn RAM64.sv
make -dC obj_dir/ -f VRAMn.mk > make32.log
rm -r obj_dir/
verilator -GRAM_DEPTH=64 -GDATA_WIDTH=16 -Wall -Wpedantic -cc -top RAMn RAM64.sv
make -dC obj_dir/ -f VRAMn.mk > make64.log
```
With RAM_DEPTH=32, make successfully builds the object file VRAMn__ALL.o which I would then link to when building a testbench, for example,
```
g++ -I$MUSR/share/verilator/include/ -Iobj_dir/ $MUSR/share/verilator/include/verilated.cpp obj_dir/VRAMn__ALL.o tb_RAM32.cpp -o tb_RAM32
```
But with RAM_DEPTH=64, this object file is not created. I've also noticed that, for RAM64, VRAMn_Register__D10__Slow.cpp is split or duplicated, I don't what this second file is for, into VRAMn_Register__D10__1__Slow.cpp. This is doesn't happen for RAM32.

And finally, as I mention in the initial post, if I run /usr/bin/perl $PATH_TO/share/verilator/bin/verilator_includer etc. manually on the cpp files inside obj_dir/ for RAM64, followed by g++ and ar, VRAMn__ALL.o is built and linked to the testbench without error.

Thus, it appears to me that make is having difficulty when the Register Slow.cpp gets split up.

wsnyder, 2021-04-18:
I get

    /svaha/wsnyder/SandBox/homecvs/v4/verilator/bin/verilator -GRAM_DEPTH=32 -GDATA_WIDTH=16 -Wall -Wpedantic -cc -top RAMn RAM64.sv
    %Error: Specified --top-module 'RAMn' isn't at the top level, it's under another instance 'RAM64'

emohohen-that-spells, 2021-04-18:
Begging your pardon, I forgot to mention that I commented out the RAM64 module.

wsnyder, 2021-04-18:
Both logfiles you attached make a 'VRAMn__ALL.a' file.  I see no problems trying any of these.

emohohen-that-spells, 2021-04-18:
I tried static linking against this file also for RAM64 with I think was `-Lobj_dir/ -l:VRAMn__ALL.a` when compiling the testbench but the linker failed.

wsnyder, 2021-04-18:
What would be helpful is a compete self-contained test that when run shows exactly what is wrong.  E.g. tell me to download a .zip file then type "make" and you expected it to show "X" but it show "Y".

emohohen-that-spells, 2021-04-19:
Part of the difficulty is that neither verilator nor make throws an error, even with debug switches enables. I don't really know what to expect but presumed creation of an "_ALL.cpp" and  "__ALL.o" was standard. The SV and commands I gave in the original post demonstrate that these files are not created, and I state that the "__ALL.a" that is generated causes a link failure. I don't believe that the problem lies in the design, verilator, or in building the testbench because I can generate these __ALL files manually by taking perl, g++, & ar commands from a successful run with RAM_DEPTH=32 but adjusted for RAM_DEPTH=64. That leaves make and the makefiles as the source of the problem. Reading through the portion of the make logs posted, for RAM_DEPTH=64 make neglects the stage "Considering target file 'VRAMn__ALL.o'." after "Found an implicit rule for 'VRAMn__ALL.a'."

Building a testbench with RAMn as the top module, RAM_DEPTH=64 and static linking with VRAMn__ALL.a, the error I get is:
```
/usr/bin/ld: /tmp/ccwSZtCu.o: in function `VRAMn::VRAMn(char const*)':
tb_RAMn.cpp:(.text._ZN5VRAMnC2EPKc[_ZN5VRAMnC5EPKc]+0x21): undefined reference to `VRAMn::VRAMn(VerilatedContext*, char const*)'
/usr/bin/ld: /tmp/ccwSZtCu.o: in function `VRAMn::eval()':
tb_RAMn.cpp:(.text._ZN5VRAMn4evalEv[_ZN5VRAMn4evalEv]+0x14): undefined reference to `VRAMn::eval_step()'
collect2: error: ld returned 1 exit status
```

wsnyder, 2021-04-19:
Can you please provide a complete standalone test that shows this working with 32 and not working with 64?

wsnyder, 2021-04-19:
(that is links something and works in one case, and shows link errors in the other)

emohohen-that-spells, 2021-04-22:
I'm getting the impression that my assumption about __ALL.cpp/.o is incorrect and after playing around a bit more I've found that the order in which __ALL.a and the testbench.cpp appear makes the difference when compiling. For RAM_DEPTH=32 or less, the order doesn't matter but 64 and up, testbench.cpp must appear first.

wsnyder, 2021-04-22:
Ah, that makes sense then, you might want to see https://stackoverflow.com/questions/45135/why-does-the-order-in-which-libraries-are-linked-sometimes-cause-errors-in-gcc

Note you can also put the .cpp file on the Verilator command line.

I'm closing as assumes answers your problem, but feel free to followup.

wsnyder, 2021-04-22:
If you put the .cpp on the command line note you'll also probably want the --exe, and also perhaps the --build option.

closed 2021-04-22 by wsnyder