verilator/verilator #2903: Multiple declarations of net and variable names are not detected

albert-magyar, 2021-04-21:
I've tried the following circuit with `Verilator 4.201 devel rev v4.032-1160-gc443e229`:

```
module multidecl(
   input  i,
   output o
);

   wire x = i;
   wire x = ~i;
   reg x = x;
   reg x = ~x;
   assign o = x;
endmodule // multidecl
```

Though the circuit does violate Section 3.13 of the 1880 standard, there are no warnings or errors, even with both `-Wall` and `-Wpedantic` enabled.

wsnyder, 2021-04-21:
Indeed, perhaps you'd be willing to look at a pull request to check for this.

A likely complication is not breaking when the declarations are under generate blocks.

albert-magyar, 2021-04-21:
> Indeed, perhaps you'd be willing to look at a pull request to check for this.

I'd certainly be happy to try anything out and follow up.

> A likely complication is not breaking when the declarations are under generate blocks.

Indeed; I've just been looking at flat module-level scopes so far.

In the cases where Verilator does currently report an error, it is due to conflicting continuous assignment and procedural assignment creating a `BLKANDNBLK` situation.

wsnyder, 2021-04-21:
It wouldn't cover everything, but first order V3LinkDot when it resolves symbols should be able to see it was already declared in the same symbol table (symtab).  This would miss cases where one was under a generate and another wasn't but hits the common case.

wsnyder, 2025-09-08:
Linting was improved in the last release, this now reports a duplicate error.

closed 2025-09-08 by wsnyder