verilator/verilator #2904: FYI:  potential codecov security issue

abalib, 2021-04-23:
https://about.codecov.io/security-update/
"On Thursday, April 1, 2021, we learned that someone had gained unauthorized access to our Bash Uploader script and modified it without our permission..."

Security scans are flagging certain verilator files because of references to codecov

wsnyder, 2021-04-23:
Known and doesn't affect Verilator itself.  If there's something we need to do to "acknowledge and clear" the warning let me know, thanks.

closed 2021-04-23 by wsnyder

abalib, 2021-04-23:
My IT guy asked me to remove my local repo all together which I wasn't willing to do.  I removed the scripts or lines that contain the string codecov instead to pass the security scan.  You could do the same as what I did in your master branch although not sure if it breaks anything.   Just a suggestion.  Thank you for responding to the issue

wsnyder, 2021-04-24:
The IT rule is of course pointless as the script itself has no problem, it's the destination site which was fixed.  But I know it's pointless to try to argue with security folks.  I manually downloaded the script and committed that, hopefully that will avoid the red flag.

abalib, 2021-04-24:
Agreeing with you on all of your comments and observations. Thank you.