verilator/verilator #2907: using unpacked array as a module parameter

jeras, 2021-04-27:
Using Git version of Verilator.

I will try to describe various errors I encountered, but will not provide code to reproduce each one, since they are probably related. At the end are instructions to reproduce one error.

https://github.com/jeras/rp32/blob/b632d8139262eaa1a9f8ccd090921de89ae6b30e/rtl/r5p_bus_dec.sv
The decoder is also using *binary wildcard equality operator* `==?` operator and *array reduction method* `and()`.
I first assumed they are responsible for some issues, but I still have issues with parameter array without this extra constructs.

### Array as parameter

I defined an array parameter inside `#()` of a module.
```SystemVerilog
parameter logic [AW-1:0] AS [BN-1:0] = '{BN{'x}}
```
I tried to instantiate the module with the next configuration.
```SystemVerilog
  .AS  ('{DAW'('h0_xxxx),   // 0x0_0000 ~ 0x0_ffff - data memory
          DAW'('h1_xxxx)})  // 0x1_0000 ~ 0x1_ffff - controller
```
I get the next error:
```
%Error-UNSUPPORTED: ../tbn/r5p_tb.sv:114:9: Unsupported/Illegal: Assignment pattern member not underneath a supported construct: PIN 'AS'
                                          : ... In instance r5p_tb
  114 |   .AS  ('{DAW'('h0_xxxx),    
      |         ^~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=4.202
%Error: ../tbn/r5p_tb.sv:114:9: Expecting expression to be constant, but can't convert a PATTERN to constant.
                              : ... In instance r5p_tb
  114 |   .AS  ('{DAW'('h0_xxxx),    
      |         ^~
%Error: ../tbn/r5p_tb.sv:114:4: Can't convert defparam value to constant: Param 'AS' of 'ls_dec'
                              : ... In instance r5p_tb
  114 |   .AS  ('{DAW'('h0_xxxx),    
      |    ^~
%Warning-WIDTH: ../rtl/r5p_bus_dec.sv:12:18: Operator VAR 'AS' expects 17 bits on the Initial value, but Initial value's CONST '?1?h0' generates 1 bits.
                                           : ... In instance r5p_tb.ls_dec
   12 |   logic [AW-1:0] AS [BN-1:0] = '{BN{'x}}
      |                  ^~
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
```

### Using parameter array in generate loop

I cant reproduce this one right now, so I will just write down what I remember.
Verilator complained about the expression `AS[i]` not being a constant, but is should be, as `AS` is a parameter.

```
generate
for (genvar i=0; i<BN; i++) begin
  assign s_dec[i] = s_adr ==? AS[i];
end
endgenerate
```

### Compilation error

I get an error while compiling the generated C code.
```
Vr5p_tb.cpp: In static member function ‘static void Vr5p_tb::_sequent__TOP__33(Vr5p_tb__Syms*)’:
Vr5p_tb.cpp:918:37: error: no match for ‘operator&’ (operand types are ‘IData’ {aka ‘unsigned int’} and ‘VlUnpacked<unsigned char, 2>’)
  917 |                             [0U] = ((IData)(vlTOPp->r5p_tb__DOT__ls_dec__DOT__m_dec)
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                     |
      |                                     IData {aka unsigned int}
  918 |                                     & vlTOPp->r5p_tb__DOT____Vcellinp__ls_dec__m_ack);
      |                                     ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               VlUnpacked<unsigned char, 2>
In file included from /opt/systemc-2.3.2/include/sysc/communication/sc_port.h:35,
                 from /opt/systemc-2.3.2/include/sysc/communication/sc_signal.h:32,
                 from /opt/systemc-2.3.2/include/sysc/communication/sc_buffer.h:34,
                 from /opt/systemc-2.3.2/include/systemc:79,
                 from /opt/systemc-2.3.2/include/systemc.h:219,
                 from Vr5p_tb.h:11,
                 from Vr5p_tb.cpp:5:
/opt/systemc-2.3.2/include/sysc/kernel/sc_event.h:817:1: note: candidate: ‘sc_core::sc_event_and_expr sc_core::operator&(sc_core::sc_event_and_expr, const sc_core::sc_event&)’
  817 | operator & ( sc_event_and_expr expr, sc_event const & e )
      | ^~~~~~~~
/opt/systemc-2.3.2/include/sysc/kernel/sc_event.h:817:32: note:   no known conversion for argument 1 from ‘IData’ {aka ‘unsigned int’} to ‘sc_core::sc_event_and_expr’ {aka ‘sc_core::sc_event_expr<sc_core::sc_event_and_list>’}
  817 | operator & ( sc_event_and_expr expr, sc_event const & e )
      |              ~~~~~~~~~~~~~~~~~~^~~~
/opt/systemc-2.3.2/include/sysc/kernel/sc_event.h:825:1: note: candidate: ‘sc_core::sc_event_and_expr sc_core::operator&(sc_core::sc_event_and_expr, const sc_core::sc_event_and_list&)’
  825 | operator & ( sc_event_and_expr expr, sc_event_and_list const & el )
      | ^~~~~~~~
/opt/systemc-2.3.2/include/sysc/kernel/sc_event.h:825:32: note:   no known conversion for argument 1 from ‘IData’ {aka ‘unsigned int’} to ‘sc_core::sc_event_and_expr’ {aka ‘sc_core::sc_event_expr<sc_core::sc_event_and_list>’}
  825 | operator & ( sc_event_and_expr expr, sc_event_and_list const & el )
      |              ~~~~~~~~~~~~~~~~~~^~~~
Vr5p_tb.cpp:921:45: error: no match for ‘operator&’ (operand types are ‘IData’ {aka ‘unsigned int’} and ‘VlUnpacked<unsigned char, 2>’)
  920 |                             [1U] = (((IData)(vlTOPp->r5p_tb__DOT__ls_dec__DOT__m_dec)
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  921 |                                      >> 1U) & vlTOPp->r5p_tb__DOT____Vcellinp__ls_dec__m_ack);
      |                                      ~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      |                |
      |                                      |                VlUnpacked<unsigned char, 2>
      |                                      IData {aka unsigned int}
```

To reproduce:

```bash
git clone https://github.com/jeras/rp32.git -b b632d81
cd rp32
cd sim
make -f Makefile.verilator
```

jeras, 2021-04-30:
I tried to rewrite the code so that it would compile.

1. The error while compiling generated code seems to be due to the use of unpacked arrays as module ports.

2. The *binary wildcard equality operator* `==?` seems to work just fine, great.

3. The *array reduction method* `and()` reports a syntax error, I would assume it is not implemented.

wsnyder, 2021-06-04:
Sorry, forgot to comment.  Yes, the items you indicated are unsupported, if you or someone would like to work on these I can provide pointers.  Otherwise, adding this to the work stack!

jeras, 2021-06-04:
I do not feel confident enough to write C++ code, but I could write testcases, as I wrote Verilator testcases before.
In any case I have to go back to the original code, and figure what exactly triggered the issue.
In any case this is the second SystemVerilog issue in unusual code I fount recently, so I will prepare both testcases at the same time in about 2 weeks.
The second issue is using struct literals inside enumerations, when the enumeration itself is of the correct struct type.